enesbcs/rpieasy #111: PCA9685 issue

scargill, 2019-12-17:
I have rpieasy running - no problem - got the Dallas chip running and serial 5050 leds - but when I checked the dependency to use PCA9685, rpieasy, instead of loading up the dependency - complained...

pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
VLC is not supposed to be run as root. Sorry.
If you need to use real-time priorities and/or privileged TCP ports
you can use /usr/bin/vlc-wrapper (make sure it is Set-UID root and
cannot be run by non-trusted users first).

enesbcs, 2019-12-17:
> I have rpieasy running - no problem - got the Dallas chip running and serial 5050 leds - but when I checked the dependency to use PCA9685, rpieasy, instead of loading up the dependency - complained...
Clicking on the underlined text to install dependencies may take several minutes (10-15) if several thing needs to be installed. In the meantime the webGUI is not responding and no status messages will be written to the console except the regular Time rules. 
In case you run out of patience, you can run the following command to install PCA9685 support directly from Linux command line: (or you can see an error message if any, such as no file found, no package found, dependency failure, net unreachable, etc...)
`sudo pip3 install PCA9685-driver`
Last time i've checked PCA9685-driver package it was installed without problems to Buster.

> pygame 1.9.4.post1
> Hello from the pygame community. https://www.pygame.org/contribute.html
> VLC is not supposed to be run as root. Sorry.
> If you need to use real-time priorities and/or privileged TCP [...]
It is not a real error just checking if vlc binary can be started/existing. Python based RPIEasy plugins are using the first non-root user to run vlc. The dependency checker on the other hand starts everything as root, as it is the only way to install dependencies. Consider it as a debug message.

scargill, 2019-12-17:
I tried sudo pip3 install pca9685-driver

pi@ukpi3:~:22:39[0]> sudo pip3 install pca9685-driver
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pca9685-driver
  Using cached https://www.piwheels.org/simple/pca9685-driver/PCA9685_driver-1.2.0-py3-none-any.whl
Collecting smbus-cffi==0.5.1
  Using cached https://files.pythonhosted.org/packages/29/3d/a50bd997979c7554c9c571753d34e976eb88ebf41d3a66accb2468bd3c69/smbus-cffi-0.5.1.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7o1vdbrl/smbus-cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7o1vdbrl/smbus-cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-7o1vdbrl/smbus-cffi/pip-egg-info
         cwd: /tmp/pip-install-7o1vdbrl/smbus-cffi/
    Complete output (161 lines):
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
    c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
              ^~~~~~~
    compilation terminated.
    Traceback (most recent call last):
      File "/usr/lib/python3.7/distutils/unixccompiler.py", line 118, in _compile
        extra_postargs)
      File "/usr/lib/python3.7/distutils/ccompiler.py", line 909, in spawn
        spawn(cmd, dry_run=self.dry_run)
      File "/usr/lib/python3.7/distutils/spawn.py", line 36, in spawn
        _spawn_posix(cmd, search_path, dry_run=dry_run)
      File "/usr/lib/python3.7/distutils/spawn.py", line 159, in _spawn_posix
        % (cmd, exit_status))
    distutils.errors.DistutilsExecError: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 172, in run
        cmd = self.call_command('install_lib', warn_dir=0)
      File "/usr/lib/python3/dist-packages/setuptools/command/bdist_egg.py", line 158, in call_command
        self.run_command(cmdname)
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 24, in run
        self.build()
      File "/usr/lib/python3.7/distutils/command/install_lib.py", line 109, in build
        self.run_command('build_ext')
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
[... 81 lines cut ...]
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 724, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 791, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 704, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 730, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 915, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1183, in build_and_install
[... cut ...] up_base, args)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1171, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
pi@ukpi3:~:22:41[1]>

enesbcs, 2019-12-18:
It looks like a libffi dependency is missing on you installation, i guess it is provided by another package on mine test device. The solution is:
`sudo apt-get install libffi-dev`
I've added this dependency for tthe PCA9685 at commit https://github.com/enesbcs/rpieasy/commit/e77dd41e88aa176e925a06676f1a759db1edf7aa
Tested and yes, it takes a lot of time on my RPI Zero. :)

enesbcs, 2019-12-27:
Issue closed as no further informations received.

closed 2019-12-27 by enesbcs