enesbcs/rpieasy #113: !Your GPIO can not be identified!   PI 4 with Buster full install.

Budman1758, 2019-12-24:
I get that message when RPIEasy first starts up.  Also cannot access either the "Hardware" or "Sysinfo" pages in the UI. All other pages seem to load fine.  

If I try to set up a GPIO device I get a "Device not supported message" when trying to access the "pin configured" link in the device setup page.

Other than the GPIO not identified message there is nothing more in the logs that I can see.....

Let me know what else I can do to help....

Merry Christmas!!!!!

enesbcs, 2019-12-24:
> I get that message when RPIEasy first starts up. Also cannot access either the "Hardware" or "Sysinfo" pages in the UI. All other pages seem to load fine.
> 
> If I try to set up a GPIO device I get a "Device not supported message" when trying to access the "pin configured" link in the device setup page. Also the GPIO drop down list does nothing.
> 
> Other than the GPIO not identified message there is nothing more in the logs that I can see.....
> 
> Let me know what else I can do to help....
> 
> Merry Christmas!!!!!

Merry Christmas!
It looks like your Pi4 hardware revision is brand new. Until i add it, most functions will not available. Please show me the result of this command:

cat /proc/cpuinfo

Budman1758, 2019-12-24:
Here is the output of that command.

`pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : c03112
Serial          : 100000009e4a8dbe
Model           : Raspberry Pi 4 Model B Rev 1.2`

enesbcs, 2019-12-24:
As i see it's a really new stuff. :) https://bugs.launchpad.net/ubuntu/+source/flash-kernel/+bug/1854487
Revision code added at commit https://github.com/enesbcs/rpieasy/commit/6e376bf9002f2af999751c9e2b4b7f0adbdafb10

Budman1758, 2019-12-24:
That seems to have done the trick!!  Thanks for the fast work!!

closed 2019-12-24 by Budman1758