enesbcs/rpieasy #114: Feature Request - MPU9250 or MPU6050

happytm, 2019-12-25:
Can you please implement either MPU9250 or MPU 6050 over I2C ? There are many use cases but in my case I am going to experiment  with using it as digital joystick to control menu system.

https://github.com/tuupola/micropython-mpu9250
https://pypi.org/project/mpu6050-raspberrypi/
https://github.com/nickcoutsos/MPU-6050-Python/blob/master/MPU6050.py

Thanks.

enesbcs, 2019-12-25:
It is possible, as i remember i have some mpu6050 hardware laying somewhere.

enesbcs, 2019-12-27:
MPU6050 support added at commit https://github.com/enesbcs/rpieasy/commit/9bf3aae39515ec6834b2d1a258dbb8ea8c8277b1
I am still curious how would you implement menu usage based on Gyro x,y,z values?

happytm, 2019-12-27:
As I said earlier I am going to experiment with it. It may not be successful or practical. I have not read or seen anywhere it being experimented.

Hardware wise my crazy idea is to mount MPU-9250 OR MPU6050 (protected by plastic around it) on top of something like following but in plastic or some other material ( may be 3d printed? ) :

https://www.ebay.com/itm/RC-Boat-Metal-Cardan-Joint-Gimbal-Couplings-Universal-Joint-AccessoriescjaFDBB/192903253831?var=493069420907&hash=item2ce9ee0f47:m:mLWJ1Pk75zw78azmrc-EfZA

In software I was thinking of doing it like we do it with rotary encoder, joystick, MPR121 or APDS-9960 using rules. Benefit of this sensor is it does not require extra gpio pin or does not require analog pin which raspberry pi does not have. Also it gives us 3 variables as opposed to 2 variables in case of rotary encoder and joystick. Until now I was using rotary encoder, MPR121 or APDS9960 for my inputs but this sensor could be cheaper and better.

Unfortunately I have only MPU-9250 on hand right now so I have to order MPU-6050 from china. 

Thanks.

enesbcs, 2019-12-27:
As i see MPU-9250 is basically an MPU6500 + an AK8963 module.
Could you try that MPU6050 code works on MPU6500?
I've ordered an MPU9250 for testing.

happytm, 2019-12-27:
I was curious to know what are the output for x,y,z ? are they 0 to 360  degrees or 0 to 100 percentage?

Thanks.

enesbcs, 2019-12-27:
According to the library ( https://pypi.org/project/mpu6050-raspberrypi/ ) Acceleration in m/s^2, Gyroscope data returned by 250DEG range, whatever it means...

enesbcs, 2019-12-27:
```
Three-Axis MEMS Gyroscope with 16-bit ADCs and Signal Conditioning
The MPU-60X0 consists of three independent vibratory MEMS rate gyroscopes, which detect rotation about
the X-, Y-, and Z- Axes. When the gyros are rotated about any of the sense axes, the Coriolis Effect causes
a vibration that is detected by a capacitive pickoff. The resulting signal is amplified, demodulated, and filtered
to produce a voltage that is proportional to the angular rate. This voltage is digitized using individual on-chip
16-bit Analog-to-Digital Converters (ADCs) to sample each axis. The full-scale range of the gyro sensors
may be digitally programmed to ±250, ±500, ±1000, or ±2000 degrees per second (dps). The ADC sample
rate is programmable from 8,000 samples per second, down to 3.9 samples per second, and user-selectable
low-pass filters enable a wide range of cut-off frequencies.
```

happytm, 2019-12-27:
This device is even more complex (that means better?) then I initially realized. I did not know about 16 bit ADCs which is lot better than some external ADCs.

Thanks.

happytm, 2019-12-29:
I am linking following 2 videos for you reference.

https://www.youtube.com/watch?v=4BoIE8YQwM8
https://www.youtube.com/watch?v=j-kE0AMEWy4

Thanks.

enesbcs, 2020-02-20:
MPU9250 support added at commit https://github.com/enesbcs/rpieasy/commit/76fbfc65c9b82fee4a7eda28caf49be5561052e6
Values from RTIMULib: Roll,Pitch,Yaw

closed 2020-02-29 by enesbcs