enesbcs/rpieasy #118: Feature Request: Fingerprint sensor support

Budman1758, 2020-01-03:
This would be nice for access control system.  I currently have a keypad\RFID setup and adding fingerprint support would be cool!

Most of the Arduino style sensors are similar to [this ](https://www.adafruit.com/product/751)one in the Adafruit shop. It uses the AS608 chip. I think most use the same chip or ones very similar.  There are many [examples ](https://www.ebay.com/itm/DY50-Optical-Fingerprint-reader-Sensor-Module-All-in-one-For-Arduino-Locks/112070491878?hash=item1a17ebfee6:g:bpgAAOSwrhRdnv~-)on EBay for much less $$ than Adafruit.

Support in RPIEasy could be as simple as triggering a GPIO when it recognizes a fingerprint that is already enrolled. Or if you want the entire process supported that would be great.

Some links to software examples.....
[https://learn.adafruit.com/adafruit-optical-fingerprint-sensor/downloads](url)
[https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library](url)
[https://github.com/brianrho/FPM](url)

Happy New Year!!!

enesbcs, 2020-01-03:
Thanks for the idea, it looks interesting. I've ordered a DY50 from ebay just now. :)
I see that it needs DC 3.6 ~ 6.0V power supply so i guess that RX-TX needs to be level-shifted to be connected with a Raspberry Pi. I will see what can i use for its functions.
Happy New Year! :)

enesbcs, 2020-02-29:
Support added at commit https://github.com/enesbcs/rpieasy/commit/cf42850319e445bbbcd21b9dfc6b96b77c4bb042

enesbcs, 2020-03-13:
Support added, issue is now closed. Feel free to reopen in case of related problems.

Budman1758, 2020-03-19:
Finally got a chance to test this.  I installed both dependencies ant they are checked in the dependencies  page but when I try to add the device I get this error message. I also rebooted the device after each dependency install.

`Importing failed, please double check dependencies! libopenjp2.so.7: cannot open shared object file: No such file or directory`

This is a fresh install and updated version of Buster Lite on an RPI Zero W

enesbcs, 2020-03-19:
Interesting, libopenjp2-7 is in the dependency list... could you try to install manually with:

`sudo apt-get install libopenjp2-7`

Budman1758, 2020-03-19:
After installing libopenjp2-7 (it was not installed) i get this message.

`Importing failed, please double check dependencies! libtiff.so.5: cannot open shared object file: No such file or directory`

enesbcs, 2020-03-19:
I am testing it now an a new Raspbian installation. I think i left out accidentally a bond in requirements.

enesbcs, 2020-03-19:
dependency list fixed in commit https://github.com/enesbcs/rpieasy/commit/0ffa0932e10699167628e274276f625e5e090821

Budman1758, 2020-03-19:
Got the dependencies all fixed.  But the plugin seems to not like my particular module.  I get this message in the web log page....
`14:44:25 FPM init error: The received packet do not begin with a valid header!`

enesbcs, 2020-03-20:
Do you use USB-serial adapter in between the RPI and the FPM module? I saw "received packet do not begin with a valid header!" errors when serial connection failed.

Budman1758, 2020-03-20:
I hooked up the module direct to the Pi.  Should I use the usb adapter?

enesbcs, 2020-03-20:
Please try with an USB adapter first. 

It also can be done with direct serial on the Pi, but it is difficult to setup.
First you have to disable onboard Bluetooth, to release the default serial line and of course you have to enable UART. (Hardware/Pinout&Ports)

Than use command line 'raspi-config' tool: 5- Interfacing Options-P6 Serial- (Kernel logging disabled + serial port hardware enabled)

For connection see:
https://www.instructables.com/id/Connect-an-ESP8266-to-your-RaspberryPi/

Budman1758, 2020-03-20:
I now have it working!  Connected directly to the Pi too.  Using a PI zero (not "W")  for my testing. Some observations and weirdness....
For test purposes I am running RPIEasy from the console  so as to be able to see output messages. Here is the console message when it starts up...

 > sudo ./RPIEasy.py
Init hardware...
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
14:12:41: !FPM init error: write failed: [Errno 9] Bad file descriptor
Exception ignored in: <function PyFingerprint.__del__ at 0xb5d476f0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pyfingerprint/pyfingerprint.py", line 177, in __del__
    self.__serial.close()
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 451, in close
    os.close(self.fd)
OSError: [Errno 9] Bad file descriptor
14:12:47: FPM initialized
Load notifiers from file
14:12:47: Event: System#Boot
14:12:47: Webserver starting at port 80

The fingerprint module seems to work ok after those error messages.  Not too sure what to make of it.  Here is the console output when I terminate the program using  Ctrl+C

> ^C14:17:50: Event: System#Shutdown
Program exiting gracefully
Exception ignored in: <function PyFingerprint.__del__ at 0xb5d476f0>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pyfingerprint/pyfingerprint.py", line 177, in __del__
  File "/usr/local/lib/python3.7/dist-packages/serial/serialposix.py", line 451, in close
TypeError: 'NoneType' object is not callable

Not too sure what to make of that either. :)

EDIT:  Suddenly I am getting the valid header message and it no longer works.

`14:38:08: !FPM error: The received packet do not begin with a valid header!`

I am going to start over using a Pi 3 and the USB connection.  Stay tuned.....

enesbcs, 2020-03-20:
> The fingerprint module seems to work ok after those error messages.

Nothing to do with that. At startup RPIEasy tries to restore objects from tasks.json, but serial object refers to a non-occupied memory region, so the plugin deletes and recreates it after some seconds delay.

> Not too sure what to make of that either. :)

Serial object do not likes CTRL-C much. :) Also nothing to do with that.

> EDIT: Suddenly I am getting the valid header message and it no longer works.
> `14:38:08: !FPM error: The received packet do not begin with a valid header!`
Invalid header means that serial communication goes off-sync. It works similarly on my x86 PC with Ubuntu and USB-serial. Pressing submit button too much will drive serial communication offsync easily. Restarting RPIEasy solves that usually.

> I am going to start over using a Pi 3 and the USB connection. Stay tuned.....
Results will be very similar.

enesbcs, 2020-04-08:
Closed as no further informations provided. Feel free to reopen in case of errors.

closed 2020-04-08 by enesbcs