enesbcs/rpieasy #120: New plugin: MH-ET Live e-paper display

enesbcs, 2020-01-05:
1.54" monochrome model used for implementing plugin.

enesbcs, 2020-01-28:
support added at commit https://github.com/enesbcs/rpieasy/commit/423973f07589caa007ff6b7c9da05ec5a3bd52ca Only tested with mono-color 1.54" version.

Budman1758, 2020-01-29:
Just ordered one of these displays. Should have it next week.  Looking forward to this!

enesbcs, 2020-01-29:
Interesting devices. I've just ordered because of its very low consumption. It is a bit slow refresh rate, but works ok. I've only mono-color version, not sure if the bi-color version works ok as that is using two separated layer to display. :)

TD-er, 2020-01-30:
Another very nice feature of these displays is that they are perfectly readable in full sunlight.

enesbcs, 2020-03-13:
Support added, issue is now closed. Feel free to reopen in case of related problems.

Budman1758, 2020-03-18:
Hello. Finally got around to testing this.  I have a Waveshare 4.2 inch model.  When attempting to pick that size in the dropdown list I cannot select anything but the 1.54 inch (200x200) size. No matter what size you try to select it bounces back to the 1.54 size.  I know the development was with 1.54 size. Not sure if this was intentional or not.

Also would it be possible to be able to hook up to SPI-1 and select other pins for the DC, RES and BUSY pins?  This test project is on a PI Zero and I have an SPI ethernet module connected to SPI-0. I know this is an unusual setup but the ability to use either SPI lines would be nice for other setups I would think.

So possibly select boxes for SPI 0 or 1 and for the other pins also?

As always, great job on this software!!

enesbcs, 2020-03-19:
Did you see any message in the console when submitting display type? Like: "EPD type unknown?"

Pinout is not really a chance in EPD library (without rewriting it), it is harwired:
https://pypi.org/project/epd-library/

Budman1758, 2020-03-19:
> Did you see any message in the console when submitting display type? Like: "EPD type unknown?"

I will have to look for that.  Was connected to the web UI from a different machine.  As for pinout,  oh well. Didn't hurt to ask.  :)

enesbcs, 2020-03-19:
> Hello. Finally got around to testing this. I have a Waveshare 4.2 inch model. When attempting to pick that size in the dropdown list I cannot select anything but the 1.54 inch (200x200) size. No matter what size you try to select it bounces back to the 1.54 size. I know the development was with 

It is really strange. The first entry in the select box is selected when the incoming value is 0 or empty string. Did you try it with another browser?
I'll try to remove reloading based on select box change, (commit https://github.com/enesbcs/rpieasy/commit/0ffa0932e10699167628e274276f625e5e090821)  but honestly, with Firefox/Linux i can select any display from the select and click to submit button, the setting is stored. (Even without connecting the display to SPI)

Budman1758, 2020-03-19:
The size select problem is resolved it seems.  Pretty sure I have the display connected properly but it is not showing anything at all.  Tried connections described in the plugin page and also the ones at Waveshare's wiki
[https://www.waveshare.com/wiki/4.2inch_e-Paper_Module](url)

enesbcs, 2020-03-20:
The EDP library in fact contains 13 different libraries for each display, and some of them differs each other.. i will check out "epd4in2" for differences.

Budman1758, 2020-03-20:
> The EDP library (for some reason which is dont know) in fact contains 13 different libraries, one for each display, and some of them differs each other.. i will check out "epd4in2" for differences.

Of course they are all different!  Why would they make it easy??  :)   LOL

enesbcs, 2020-03-20:
partialupdate and setframe is now only used if available at commit https://github.com/enesbcs/rpieasy/commit/39bc5376a639167e77560290130380ec63a94218

Budman1758, 2020-03-20:
I think we are making some progress.  The display flashes a couple times and then displays some "noise" but then the program crashes and becomes unresponsive.  I see 2 different error messages in the console when trying to stop the program. (using ctrl+C)

` !EPD init error: 'Perver' object has no attribute '_server`

` !EPD write error! Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)'`

Budman1758, 2020-03-21:
> Did you enable SPI-0 and setup output/input pins at pinout settings? dc-reset-busy?

Yes

> Did you try to run the test application? ( ImageFont.truetype() path needs to be modified for example to img/UbuntuMono-R.ttf attached in RPIEasy)

I have no idea how to do that....  Sorry.  :(

I am getting this message when  exiting the program

`21:51:12: !EPD write error! Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
`

The display is flashing and sometimes the text is displaying but when it does it is extremely slow and the web interface is completely unresponsive. I have 8 lines set with test text and sometimes it will show all eight lines but takes almost a minute to fill it in. All during this time the web interface is non-responsive.  If need be I could probably make a video and sent it to you.

enesbcs, 2020-03-21:
> > Did you try to run the test application? ( ImageFont.truetype() path needs to be modified for example to img/UbuntuMono-R.ttf attached in RPIEasy)
> 
> I have no idea how to do that.... Sorry. :(

You have to download the linked python source 'main.py' into the RPIEasy directory, modify the font line to
`ImageFont.truetype('img/UbuntuMono-R.ttf')`
and run it with 'python3 main.py'.

> The display is flashing and sometimes the text is displaying but when it does it is extremely slow and the web interface is completely unresponsive. I have 8 lines set with test text and sometimes [...]

About slowness: yes E-paper is slow. Mine supports partial_update which means it draws only the changed pixels, and it is slow. If yours do not even support partial_update, than 400x400=160000 pixel update one-by-one through spi will be very very slow. 
Flashing: every time (if you do not checked clear only used lines), the plugin will delete the full image, and rewrite the whole content.
As is see only the 1.54, 2.13 and 2.9 displays supports partial update feature. (at this time)

Did you try enable "Clear only used lines" option and set interval for example to 10 seconds with one line only?

If it's not better i will order a 2.7" model for testing which works similarly as the 4.2".

Budman1758, 2020-03-21:
The example main.py works.  It takes almost 10 seconds for the image to display.

I have the plugin set for 8 lines with basic text and a 120 second update interval.

When I start ESPEasy from command line all 8 lines display.  It takes 65 seconds to do so.  After that 65 seconds the program starts the web server and the UI becomes available. When attempting to access the device settings page the display turns to noise and the UI is then completely unresponsive. RPIEasy appears to crash.  Ctrl+C ends the program and gives this message in the console.

` !EPD write error! Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)`

I cannot get to the device settings page at all to change settings.

enesbcs, 2020-03-21:
> When I start ESPEasy from command line all 8 lines display. It takes 65 seconds to do so. After that 65 seconds the program starts the web server and the UI becomes available. When attempting to [...]

Yes, EPD driver has a blocking nature and RPIEasy plugin_init() sections not using threading, it is waiting reply from the plugin if it able to start succesfully. I will try to implement a background process to handle EPD init, that way RPIEasy will startup normally, but EPD settings will still be unavailable until display init section finish. (~60seconds)
 
> ` !EPD write error! Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)`

It can be because plugin_exit() calling display sleep mode. I will remove that.

Budman1758, 2020-03-21:
Its actually crashing the program.  UI becomes completely unresponsive.  You could wait for hours and it wont respond. (I did!)

enesbcs, 2020-03-21:
> Its actually crashing the program. UI becomes completely unresponsive. You could wait for hours and it wont respond. (I did!)

Sure, RPIEasy will wait forever for the response from EPD library. And as init() runs on the main thread, UI will not respond in the meantime.

Threading init() added to EPD plugin at commit https://github.com/enesbcs/rpieasy/commit/0e62e3c070a6f6f2a6b8e4cb0dffba0451ab113a In case of strange behaviour try to delete device and recreate it.
Tested with my 1.54" model, noise on screen appears when Resetting display module. After that 2 lines (ip+systime) written in 3 seconds in partial update mode.
RPIEasy will now works even if epd not responding, but display will not work until init returns...

Budman1758, 2020-03-21:
Well, this is still pretty much unusable.  Crashes, Does not display anything, displays noise and UI still unresponsive.  It is obviously not working anyway near the same as your 1.54" module.  I'm not a programmer at all but if these things take an hour for the thing to "init" then something just isn't right.

I'm getting this error even after a reboot.
`16:52:47: !EPD init already in progress try again later!`

And this error constantly
`!EPD write error! 'dict' object has no attribute 'getsize'`

Nothing on the display.

enesbcs, 2020-03-22:
> Well, this is still pretty much unusable. Crashes, Does not display anything, displays noise and UI still unresponsive. It is obviously not working anyway near the same as your 1.54" module. I'm not [...]
 
> I'm getting this error even after a reboot.
> `16:52:47: !EPD init already in progress try again later!`

Yes it is by design. An ongoing init is already in progress internally, and it will not interrupt that.

> And this error constantly
> `!EPD write error! 'dict' object has no attribute 'getsize'`
> Nothing on the display.

It is very likely, that the init has never finished, so the epd object in memory is still invalid, which has no functions to write. This is why nothing on the display.
Did you try to delete the device and recreate?

In commit https://github.com/enesbcs/rpieasy/commit/162c8f5575397af05d922d5010367fc25e6e563f the init is now forced even if it is reported to be done.
If you enable Debug level logging at RPIEasy, you have to see something similar:

```
08:09:54: Start EPD background init
08:09:54: Start EPD init
08:09:54: EPD device preinit
Load notifiers from file
Loading rules...
08:09:54: Event: System#Boot
08:09:54: Webserver starting at port 80
08:09:54: Event: Clock#Time=Sun,08:09
08:09:54: Event: Network#Connected
08:09:57: EPD background drawed
08:09:57: EPD device ready
08:09:57: Starting first EPD data display
08:10:03: Event: Clock#Time=Sun,08:10
```

enesbcs, 2020-03-22:
I've ordered a 2.7" epaper model for further testing.

Budman1758, 2020-03-22:
Updated to latest commit. Deleted device and recreated it. Upon submit to add the device nothing happens but getting this error in the console...

`!EPD write error! 'dict' object has no attribute 'getsize'`

Display is not working at all now.  Interestingly, when I click "Run" at the devises list page program becomes completely unresponsive. Have to halt or reboot at that point.

enesbcs, 2020-03-22:
> Upon submit to add the device nothing happens but getting this error in the console...
> 
> `!EPD write error! 'dict' object has no attribute 'getsize'`

Did you changed log level to debug? It is not so much information. If init was unsuccesful an "EPD init error" should have been printed. So I guess that the "EPD write error" may be after 'Starting first EPD data display" line, and i believe it lays somewhere at display_frame() or get_frame_buffer() function. I've added several further debug lines for debugging.

When you are clicking Run before the EPD succesfully finished the previous write, that may cause SPI concurrency problems and "Run" from webserver button is running in the same thread as the webserver does, effectively making it unresponsive. This now avoided in commit https://github.com/enesbcs/rpieasy/commit/9ddf9aef3f4174477e9a69a11d7e5b2078d63f3f

Budman1758, 2020-03-23:
> 18:40:42: !EPD display error! Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
18:40:42: !EPD init error: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)

Got the above messages when exiting the program

>  !EPD write error! 'dict' object has no attribute 'getsize'

Got the above message when restarting the program and at every interval whatever it is set to.

> Init hardware...
18:48:37: Pi 2 Model B 40 pins
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
18:48:38: EPD init already in progress!
Load notifiers from file
18:48:38: Event: System#Boot
18:48:38: Webserver starting at port 80
18:48:38: Event: Clock#Time=Sun,18:48
18:48:38: New P2P unit discovered: 1 192.168.1.124 B8:27:EB:C5:B3:3A
18:48:48: New P2P unit discovered: 11 192.168.1.61 60:01:94:5D:86:67
18:48:49: New P2P unit discovered: 10 192.168.1.60 5C:CF:7F:3D:C8:CD
18:48:57: !EPD write error! 'dict' object has no attribute 'getsize'

Above message after a reboot.  Log level set to "debug developer"  Display is blank.

> 18:56:15: Pi 2 Model B 40 pins
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
18:56:15: EPD init already in progress!
Load notifiers from file
18:56:15: Event: System#Boot
18:56:16: Webserver starting at port 80
18:56:16: Event: Clock#Time=Sun,18:56
18:56:16: New P2P unit discovered: 1 192.168.1.124 B8:27:EB:C5:B3:3A
18:56:17: New P2P unit discovered: 12 192.168.1.62 B8:27:EB:F8:66:B7
18:56:18: New P2P unit discovered: 11 192.168.1.61 60:01:94:5D:86:67
18:56:19: New P2P unit discovered: 10 192.168.1.60 5C:CF:7F:3D:C8:CD
18:56:35: !EPD write error! 'dict' object has no attribute 'getsize'
18:56:35: New P2P unit discovered: 63 192.168.1.63 5C:CF:7F:55:F2:2D
18:56:38: Unit alive: 11

Above message after a shutdown, power off (about 10 minutes) and then restart.  Display is blank.

enesbcs, 2020-03-23:
> 18:40:42: !EPD display error! Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
> 18:40:42: !EPD init error: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)

Did you setup Pinout&Ports according to the proposed settings?

GPIO25 as Output
GPIO17 as Output
GPIO24 as Input

Although it is done by the epd library automatically, but if is different in the RPIEasy settings, it may cause collisions. I think this error came from epdif.py which is not able to change pin state by RPi.GPIO for some reason.

Budman1758, 2020-03-23:
> Did you setup Pinout&Ports according to the proposed settings?

Yessir!

enesbcs, 2020-03-23:
Then sorry, i have no other idea at this time. The 2.7" display which API is similar than the 4.2" model will arrive to me in 40 days. :(

Budman1758, 2020-03-23:
That's fine.  I am in no particular hurry to use it.  No big deal.  Was thinking about upgrading my irrigation controller but there is absolutely no hurry.  The one I am using now is doing fine.

I have tons of stuff waiting for me to get around to it.  :)

enesbcs, 2020-05-11:
Purchase protection of this item is running out, and package did not arrived. :( I have to search another source...

enesbcs, 2020-06-16:
The display i've ordered second time is just arrived in my country. I hope it will appear soon at the local post office.

enesbcs, 2020-07-16:
I've bought a 2.7" modell without partial update and red+black colors.. i've to say that these multi-color version is really slow, and i can not recommend it to anybody. For some reason it is flashing when updating the image shown. :(
Plugin updated in commit https://github.com/enesbcs/rpieasy/commit/c59cf0dee73d8a508e180b62059fd50c19a841a3 and it now should have work without partial update feature.

enesbcs, 2020-10-17:
Closed as no further informations. Feel free to reopen in case of related errors.

closed 2020-10-17 by enesbcs