enesbcs/rpieasy #124: Tools Load fails to reinstall Controllers and Notifications

alan-rpi, 2020-02-18:
Restored setting using Tools Load. Hardware Pinouts and Devices restored OK but Controllers and Notifications are empty. Opened zip file and it does contain Controllers and Notifications.
Redefined 1 x Controllers and 1 x Notifications plus set Config IP settings. Saved setting then Reset Device Settings then Loaded saved settings:  1x Controller restored, 0 Notifications restored. Config IP settings the Primary and Secondary are swapped round.  Clicking on Hardware / Pinouts gives error "This hardware is currently not supported!" Using Putty / CLI Stopped and Restarted RPi Easy. Now can access Hardware / Pinouts  and the missing 1 Notifications appears! Phew!

enesbcs, 2020-02-18:
Did you restart the RPIEasy after restoring settings? Main structures created in memory after startup once. If the device type is not setted at startup time, and you import anything than it can not be used until restart. (Pinout)
Did you import it to another RPI? Plugins and controllers can not be imported before dependencies installed if it is an other RPI.

alan-rpi, 2020-02-19:
I did restart RPIEasy but by Ctrl-C on the CLI. I did not import it. Today I started RPIEasy and I had no Devices. Controllers, Notifications and Piouts were fine. Stopped and started via CLI. No change. I then reLoaded my backup settings and now everything is as normal. Luckily the backup was recent. 

I sense this problem is going to be difficult to resolve. Are there any log files that I can send/upload?

enesbcs, 2020-02-21:
> I sense this problem is going to be difficult to resolve. Are there any log files that I can send/upload?
Not hard. Impossible.
Console logs can help at the time of stopping and starting RPIEasy.

enesbcs, 2020-02-21:
> Restored setting using Tools Load. Hardware Pinouts and Devices restored OK but Controllers >and Notifications are empty. Opened zip file and it does contain Controllers and Notifications.

It was restored, but this point you should have restart the system.

> Redefined 1 x Controllers and 1 x Notifications plus set Config IP settings. 

At this point you have succesfully overwrite the restored ones on the disk with the new ones you've created in the memory.

>Hardware / Pinouts gives error "This hardware is currently not supported!" Using Putty / CLI 

This reply is normal, as you manually called Reset, which destroyed GPIO structures. That is only recreated at startup.

>Stopped and Restarted RPi Easy.

This had to be the first step after restoring.

enesbcs, 2020-02-22:
Load json enhanced at commit https://github.com/enesbcs/rpieasy/commit/2aeed3731a3b58a5a45bad4d04b18bed86b5a81d
Immediate restart still required to operate normally.

closed 2020-02-22 by enesbcs