enesbcs/rpieasy #127: Tools Load is failing; Tools Reset Device Settings is Failing

alan-rpi, 2020-02-19:
No Rules were working, even simple ones, although the event triggers were on the log. So I reloaded a copy of my setting Tools / Load from yesterday and got this error on the CLI:

I2C smbus not installed!
Init hardware...
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
..........
Load notifiers from file
Loading rules...
23:01:49: Event: System#Boot
23:01:49: Webserver starting at port 8008
23:01:50: Event: Clock#Time=Wed,23:01
23:02:00: Event: Clock#Time=Wed,23:02
malloc_consolidate(): invalid chunk size
Aborted

Then I tried to clear my configuration with Tools Reset Device Settings and got on the CLI:
......... as above but last lines are:
23:05:00: Event: Clock#Time=Wed,23:05
23:05:28: CMD: reset
free(): double free detected in tcache 2
Aborted

enesbcs, 2020-02-20:
It is very interesting, maybe some file access permission errors occured. Did you run it as root? If not, try "chmod -R a+rw *" in rpieasy directory.
malloc and free is not in python language, this error must be throwed by a lower-level library. 
Try to delete every file from you data/ directory and recreated files.

alan-rpi, 2020-02-21:
I deleted the RPIEasy folder and reran git clone https://github.com/enesbcs/rpieasy.git
Started RPiEasy and went to Tools/Load and loaded file data.zip

> pi@raspi4hms:~/bin/hms/rpieasy $ sudo ./RPIEasy.py
I2C smbus not installed!
Init hardware...
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
Load notifiers from file
21:16:35: Event: System#Boot
21:16:35: Webserver starting at port 8008
21:16:35: Event: Clock#Time=Fri,21:16
21:16:41: CMD: reset
rm: cannot remove 'data/*.json': No such file or directory
21:17:00: Event: Clock#Time=Fri,21:17

I did not install dependency paho-mqtt  as it was ticked (not X) but with a fresh install I would expect to have to install again.

Config OK, No Controllers, No Hardware & Pinouts: "This hardware is currently not supported!". Devices OK, No Notifications

data.zip has all the items above. Tools/Json has System, Wifi, Sensors 0-8 (each with 0-3 DataAcquisition), and TTL

Tools/Halt then restart from the command line. Now OK

Looks like Load of settings needs to "Halt"/Pause system, then Load setting, then Restart system.

Suggest that Tools / Halt comment is changed from "Halts device" to "Halts System" as it is not halting as device as per the Devices tab.

closed 2020-02-21 by alan-rpi