enesbcs/rpieasy #128: Domoticz : Mijia BLE Battery

rrraf, 2020-02-20:
Hi,

I set up Xiaomi Mijia V1 (MJ_HT_V1) & V3 (LYWSD03) and the battery is correctly read (100% in this example) but 255 is sent to Domoticz (via MQTT or HTTP) (please note that "Add Battery value for non-Domoticz system" doesn't change the issue) :

_Domoticz log :_
> 2020-02-20 17:39:43.459 MQTT: Topic: domoticz/in, Message: { "idx": XX, "nvalue": 0.00, "svalue": "20.8;44.5;**_100.0_**", "RSSI": 10, "Battery": **_255_** }

_RPIEasy log :_

> 17:39:43 Event: MijiaV1#Temperature=20.8
> 17:39:43 Event: MijiaV1#Humidity=44.5
> 17:39:43 Event: MijiaV1#Battery=**_100.0_**

enesbcs, 2020-02-20:
Could you show me your log after "Add Batter value.." is unchecked, submitted, and rpieasy restarted?

enesbcs, 2020-02-20:
You are right, i've isolated this bug and fixed in commit https://github.com/enesbcs/rpieasy/commit/0554294cd6ada2b93c4abec445c84a88acf8c930

rrraf, 2020-02-20:
you're a star! it works for both V1 and V3 ;-)

sorry i couldn't find the problem `usebattery = float(str(usebattery).strip())`

closed 2020-02-20 by rrraf