enesbcs/rpieasy #129: Domoticz : RSSI for Mijia BLE (and other BLE)

rrraf, 2020-02-20:
Hi again,
Great follow up for the battery issues. I see that the Bluetooth radio signal quality (-dBm) is available in the tool `Scan Bluetooth LE` (in hardware), any chance you could add this info to be sent to domoticz? For now it just sends `10`
It seems the code is ready to receive the value ;-)
Thanks

enesbcs, 2020-02-20:
The code is ready, but BLE devices RSSI can be gathered by changing active mode to scan mode. Until Bluepy implements an easier way to do this, it can not be added.

rrraf, 2020-02-20:
makes sense. i thought it was something the devices were sending. anyway, it's not amazingly usefully and maybe extends a little bit the batteries.

thanks for the answer

closed 2020-02-22 by enesbcs