enesbcs/rpieasy #130: Debugging Rules

alan-rpi, 2020-02-21:
I added a rule and that worked, then another that failed. Removed comments and it worked. Swapped rules around and they stopped working. Deleted and pasted in and they work. The Events always showed on the CLI but the actions seem very temperamental. This is the problem I had the other day. 

For example a simple rule
on  PanelAlarmSet#State do
        gpio, 13, [PanelAlarmSet#State]
endon

When the electronic switch is activated the Event is recognised
23:30:36: Event: PanelAlarmSet#State=1
and the Device page for PanelAlarmSet shows a state of 1 but the rule does not execute.
Pressing Run on the Device page triggers the Event but not the Rule.

I don't know if the device type is pertinent. Rules usually work for a PushButton (defined as Push Button and Rising) but not always. In the example above the PanelAlarmSet is defined as GPIO23(Input), Normal Switch, 0 debounce, Both with Value of State and it is Enabled,

I note under the Event detection type is says "Note: Only valid if event detection activated". I see nowhere to say event detection is enabled. I presumed it was enabled if "Force 10/sec periodic checking of pin:" was off.

I have over 20 more complex rules to define. Looks like debugging is going to be a major struggle. Is there a way to turn on a debug trace as the rules are parsed?

enesbcs, 2020-02-22:
As in RPIEasy, every condition evaluated once, so you can not define multiple event handler for the same event, only the first one will be processed. This is by design, maybe this is your problem.

P001 plugin tries to setup Event detection when possible, if it is succesful, than there were no extra output. If failed, it will write than backup 1/10s checking activated.

Debug: Tools/Advanced/Log levels...

TD-er, 2020-02-22:
You can probably still make it work by just triggering on the `PanelAlarmSet` and then in the rule context test the variable?
Not sure if rpieasy is literally the same as in ESPEasy, and to be honest also not 100% sure the event value is accessible in ESPEasy. But I think if it isn't, then it should also be a feature in ESPEasy.

enesbcs, 2020-02-29:
Closed as no further informations added

closed 2020-02-29 by enesbcs