enesbcs/rpieasy #131: Tools Reboot and Halt not as expected

alan-rpi, 2020-02-22:
Tools/ Reboot  gives prompt "Do you really want to reboot device". Click OK and the Rasp Pi shuts down has has to be powered off and on to restart

Tools / Halt gives the prompt "Do you really want to shutdown machine". Click OK does shutdown the machine but the word Halt implies stop. Maybe rename to Shutdown. 

Rasp Pi 4 running Buster

enesbcs, 2020-02-22:
Reboot executes "sudo reboot" If reboot command not works as expected, than it is out of the scope of this application.
Halt executes: "sudo shutdown -h now"

TD-er, 2020-02-22:
@enesbcs I think he refers to the label in the software (Tools / Halt) to be renamed to "Tools / Shutdown"

In Unix/Linux terms it is indeed a "halt" command, but people not used to that may get confused to what "Halt" means.

I believe `reboot` has been an alias for `shutdown -r now` for years now (maybe over 15 years already?)
I am not even sure anymore what the default parameter for shutdown is when you omit it, maybe it is `-h` ? So then the omission of it may be quite nasty if forgotten as you may need to actually go to machine and switch it on again, so that's probably why they made an alias for it named `reboot`.  (been there done that....)

enesbcs, 2020-02-22:
> In Unix/Linux terms it is indeed a "halt" command, but people not used to that may get confused to what "Halt" means.

I find it inconceivable for someone who does not have the necessary Linux knowledge to operate a Raspberry Pi with Linux. Or at least very urgently fill this gap... :) RPIEasy makes easier to rapidly create some sort of iot device, but it does not replace neither Linux, nor Raspberry PI GPIO/Hardware knowledge.
 
> I believe `reboot` has been an alias for `shutdown -r now` for years now (maybe over 15 years already?)

Yes, it is an alias surely, and as i wrote earlier, it is working fine on my hardware. Although i have not got a new RPI4 which may or may not have some sort of hardware flow. Time and testers will tell that.

> I am not even sure anymore what the default parameter for shutdown is when you omit it, maybe it is `-h` ? 

I am not sure, i am using parameters all the time.

enesbcs, 2020-02-22:
Reboot works as expected. Shutdown added to command description at Tools page at commit https://github.com/enesbcs/rpieasy/commit/2aeed3731a3b58a5a45bad4d04b18bed86b5a81d

closed 2020-02-22 by enesbcs