enesbcs/rpieasy #132: EQ3 termostat plugin

michapr, 2020-03-01:
Thank you for providing the eq3 plugin!
https://github.com/enesbcs/rpieasy/issues/100

I have tested some commands and it is working fine.

The only question I have, how to use it via http call?
If I have multiple devices, how I can select the right one?

Have checked the code - maybe it is missing in this time?
But have no idea in this time, where to add it...

Thanks!!
Michael

enesbcs, 2020-03-01:
Commands can be executed the same way as with ESPEasy:
http://<rpieasyip>/control?cmd=<command>

Selection between devices is not implemented yet.

michapr, 2020-03-01:
> Commands can be executed the same way as with ESPEasy:
http://[rpieasyip]/control?cmd=[command]

Yes, I know,.. Question was more about the different devices.

Is there anyway A sample where it is done, so that I could understand how to make it?
Then I could help maybe.

enesbcs, 2020-03-02:
Command structure in EQ3 plugin changed in commit https://github.com/enesbcs/rpieasy/commit/0b15729ed0ae1da1e6ccbdef4f67e418252848f1

Available commands now looks like this:
```
 eq3,taskname,sync
 eq3,taskname,mode,closed
 eq3,taskname,mode,open
 eq3,taskname,mode,auto
 eq3,taskname,mode,manual
 eq3,taskname,temp,20.5
```

michapr, 2020-03-03:
Thank you vey much for checking this!
Made update (via tools-update) , checked, new _P516_BLE_EQ3 is here.

But I have now any problems,... something is going wrong:
```
18:04:43: CMD: eq3,sync,eq3CC-RT-BLE
18:04:43: !Unknown command: eq3,sync,eq3CC-RT-BLE
```
Task name is eq3CC-RT-BLE.
(maybe better would be to use the number here? Not sure, but if user will use any UTF-8 names....)

Same for other commands like 
```
17:59:22 : CMD: eq3,eq3CC-RT-BLE,temp,20.5
17:59:22 : Unknown command: eq3,eq3CC-RT-BLE,temp,20.5
```

Thanks!
Michael

michapr, 2020-03-03:
I must correct a bit, sorry.
The sync comand is working if repeat it (there seems to be a timeout sometimes - but device is near the Pi)
```
{
"target temperature": 20.0,
"mode": "auto dst"
}
18:30:27 : Event: eq3BLE#TempOffset=1.0
18:30:36 : CMD: eq3,eq3BLE,sync
```
But the set command is not working:
```
18:31:46 : Unknown command: eq3,eq3BLE,temp,20.5
18:31:52 : CMD: eq3,eq3BLE,temp,20.5
18:31:52 : Unknown command: eq3,eq3BLE,temp,20.5
18:31:57 : CMD: eq3,eq3BLE,temp,20.5
18:31:57 : Unknown command: eq3,eq3BLE,temp,20.5
```
Mode auto/manual/open/closed are working fine

Maybe it could be possible to add a "second try" if connection is not success?

Thanks,
Michael

enesbcs, 2020-03-03:
> Maybe it could be possible to add a "second try" if connection is not success?
Yes. BLE's weakest point is the low range.

michapr, 2020-03-03:
easy... :)
     temp = misc.str2num(cmdarr[2].strip())
should be
     temp = misc.str2num(cmdarr[3].strip())

Michael

enesbcs, 2020-03-04:
Thanx for debugging, fixed in commit https://github.com/enesbcs/rpieasy/commit/2912583d060b1b4a1e63640ce504177d79f0a9c1

michapr, 2020-03-07:
Thank you, is working fine (if no timeout ;) )

closed 2020-03-07 by michapr