enesbcs/rpieasy #133: New BLE Temp. Hum - Cleargrass

michapr, 2020-03-07:
Hi,
maybe you can include it in plugin?
I have tried - but always get any errors...  Something I do wrong :(

This code is working fine - so maybe it is easy for you to integrate?

```
from bluepy import btle

mac_addr = '58:2d:34:10:54:50'

# service for temp/hum
uuid = '00000100-0000-1000-8000-00805f9b34fb'

class MyDelegate(btle.DefaultDelegate):
    def handleNotification(self, cHandle, data):
        hum=0
        temp=0
        hum_bytes = data[4:]
        temp_bytes = data[:4][2:]
        hum = int.from_bytes(hum_bytes, byteorder='little')/10.0
        temp = int.from_bytes(temp_bytes, byteorder='little')/10.0
        print('Temp=', temp, 'Hum=', hum)

per = btle.Peripheral(mac_addr)
per.setDelegate(MyDelegate())
char = per.getCharacteristics(uuid=uuid)[0]
descr = char.getDescriptors(0x2902)[0]

descr.write(0x01.to_bytes(2, byteorder="little"), withResponse=True)

per.waitForNotifications(5.0)
```

(in this time it is working with dummy and cron job here ;)

Thanks!

enesbcs, 2020-03-07:
what is this device? link + device model.

michapr, 2020-03-07:
Sorry,.... forgot of course...

Manufacturer Name: Qingping Technology (Beijing) Co., Ltd.
Model Number: CGG1

distributed by Xiaomi 
Xiaomi Cleargrass E-Link INK Screen Bluetooth Temperature Smart Humidity Sensor 
https://www.aliexpress.com/i/32994401600.html

enesbcs, 2020-03-07:
> I have tried - but always get any errors... Something I do wrong :(
> This code is working fine - so maybe it is easy for you to integrate?

Integrating a device without a physical device is never easy. The (blocking) code you provided may work on its own, but RPIEasy is a multithreading application.
I've replaced main parts of your code and LYWSD03 plugin into the new _P518 plugin, it has to be tested... https://github.com/enesbcs/rpieasy/commit/4da748af367923033676288ee9a631f80087a2f1

michapr, 2020-03-08:
> Integrating a device without a physical device is never easy

Of course...
I'm testing it - and something is really working ! Great job.
I will try to optimize a bit (reduce the number of exceptions) and/or ask you if I need help.

If I think that finished will attach the result (with or without the rest of problems ;) )

michapr, 2020-03-10:
Hi,
the curent version can be used for the BLE device.
Maybe you can remove the not needed battery parameter for Domoticz? (if possible)
`21:07:53: 192.168.178.200:8080/json.htm?type=command&param=udevice&idx=9998&nvalue=0&svalue=22.9;36.6;0&rssi=10&battery=255`

Anyway the "request_temp_hum_value" function will be called sometimes twice, that's why there will occur an error there:

```
21:07:05: BLE connection failed 58:2d:34:10:54:50
21:07:13: BLE connection initiated to 58:2d:34:10:54:50
21:07:19: BLE connection failed 58:2d:34:10:54:50
21:07:28: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (wr)
Unexpected response (desc)
21:07:35: BLE connection failed 58:2d:34:10:54:50
21:07:45: BLE connection initiated to 58:2d:34:10:54:50
21:07:53: BLE connected to 58:2d:34:10:54:50
21:07:53: Event: ClearGrassTemp&RH#Temperature=22.9
21:07:53: Event: ClearGrassTemp&RH#Humidity=36.6
21:07:53: 192.168.178.200:8080/json.htm?type=command&param=udevice&idx=9998&nvalue=0&svalue=22.9;36.6;0&rssi=10&battery=255
'NoneType' object has no attribute 'poll'
21:08:00: Event: Clock#Time=Tue,21:08
```
```
21:10:00: Event: Clock#Time=Tue,21:10
Unexpected response (wr)
Unexpected response (desc)
21:10:05: BLE connection failed 58:2d:34:10:54:50
21:10:13: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (wr)
21:10:23: BLE connection failed 58:2d:34:10:54:50
Device disconnected
21:10:33: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (find)
21:10:40: BLE connection failed 58:2d:34:10:54:50
'NoneType' object has no attribute 'poll'
21:10:46: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (wr)
21:10:53: BLE connection failed 58:2d:34:10:54:50
Device disconnected
21:11:00: Event: Clock#Time=Tue,21:11
21:11:02: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (wr)
Unexpected response (desc)
21:11:12: BLE connection failed 58:2d:34:10:54:50
21:11:20: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (wr)
21:11:29: BLE connection failed 58:2d:34:10:54:50
Device disconnected
21:11:36: BLE connection initiated to 58:2d:34:10:54:50
Unexpected response (find)
21:11:47: BLE connection failed 58:2d:34:10:54:50
'NoneType' object has no attribute 'poll'
21:11:54: BLE connection initiated to 58:2d:34:10:54:50
21:12:00: Event: Clock#Time=Tue,21:12
Unexpected response (wr)
Unexpected response (desc)
```
I have tried to fix it with any status parameters, but no luck in this time ;)
It seems not to be very critical, only some values cannot be requested.

**Other question**
```
self._nextdataservetime = rpieTime.millis()-self.preread
sample:
self._nextdataservetime = 100.000 - 4.000  => = 96000

```
is this right?
Later:
```
if self._nextdataservetime-rpieTime.millis()<=self.preread
same sample (one second later):
if 96.000 - 101.000 <= 4.000   => -5000 (always negative)

```
Maybe it should be 
`self._nextdataservetime = rpieTime.millis() **+** self.preread`  ?

I made everywhere log output and have seen the duplicate calls, maybe you have an idea?

BTW: Battery seems to be reported only in advertising data...

Thanks!
Michael

enesbcs, 2020-03-13:
> Maybe you can remove the not needed battery parameter for Domoticz? (if possible)
> `21:07:53: 192.168.178.200:8080/json.htm?type=command&param=udevice&idx=9998&nvalue=0&svalue=22.9;36.6;0&rssi=10&battery=255`

In theory it should not be seen. 255 means no support for battery.
A possible solution added at commit https://github.com/enesbcs/rpieasy/commit/153b7ad49a7f5f07e5b2eeb12fedb6001f4362a6

> Anyway the "request_temp_hum_value" function will be called sometimes twice, that's why there will occur an error there:

Indeed. I do not know any other reliable method to check if BLE connection is alive.

> **Other question**
> self._nextdataservetime = rpieTime.millis()-self.preread
> is this right?

If you refers to the line in plugin_init then yes, it is by design. It is forcing the first readout earlier. Normally init executes one time only.

> Later:
> 
> ```
> if self._nextdataservetime-rpieTime.millis()<=self.preread
> same sample (one second later):
> if 96.000 - 101.000 <= 4.000   => -5000 (always negative)

Yes, it is by design, to make the first report ASAP. After the first readout the nextdataservetime will be greater.

> `self._nextdataservetime = rpieTime.millis() **+** self.preread` ?

Nextdataservetime handled at plugin_read() based on lastdataservetime and interval.

> BTW: Battery seems to be reported only in advertising data...

It is unfortunately.

enesbcs, 2020-03-23:
If you have any enhancement to the plugin, please open a PR. Issue is closed as implemented.

closed 2020-03-23 by enesbcs