enesbcs/rpieasy #135: Ping plugin timeout

enesbcs, 2020-03-20:
enhancement: add an option to modify timeout value

enesbcs, 2020-03-22:
Timeout option added at commit https://github.com/enesbcs/rpieasy/commit/9ddf9aef3f4174477e9a69a11d7e5b2078d63f3f

angelo1967, 2020-03-23:
> Timeout option added at commit 9ddf9ae
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url": "

Hi Alex
I installed update directly from the device (tools / update)
before i saved everything (rules and json), the raspberry rebooted
but the software RPIEasy is not restarted automatically, I have this in
crontab to restart it (@reboot sudo python /home/pi/rpieasy/RPIEasy.py &)
is correct ?, also it had only lost the list of devices (reloaded with load
json), I think this update procedure needs a fix.
As for the modification, I brought the timeout to 0.8s (800ms) and let's
see how it behaves, today is bad weather and for the bridge signal it is
usually a critical condition.
I propose another possible improvement ...
Add in the tools the possibility to update the operating system
linux, so that you don't have to access the ssh console (which for those
like me who are a little fasting remains easier and more immediate) what do
you think?
thank you

enesbcs, 2020-03-23:
> I installed update directly from the device (tools / update) before i saved everything (rules and json), the raspberry rebooted but the software RPIEasy is not restarted automatically, I have this in crontab to restart it (@reboot sudo python /home/pi/rpieasy/RPIEasy.py &) is correct ?, 

This method working for me. Although i've seen previously some installations on which the rc.local script not working off the shelf. 
https://github.com/enesbcs/rpieasy/issues/9

Double check if not started another instance of RPIEasy on another port in the same time, it will surely cause data loss.

>also it had only lost the list of devices (reloaded with load json), I think this update procedure needs a fix. 

You are talking about a lot of things in a single issue which is dedicated to ping plugin. "Update" works by replacing *.py files, json files in data directory are not touched, there are nothing to fix here.
If you are playing save/restore/load in between update, it is another story. After "loading" a restart is necessary immediately after it to see devices, you will not see anything without restart.

>I propose another possible improvement ... Add in the tools the possibility to update the operating system linux, so that you don't have to access the ssh console (which for those like me who are a little fasting remains easier and more immediate) what do you think? 

It can be done. https://github.com/enesbcs/rpieasy/issues/136

angelo1967, 2020-03-29:
Hi,
The test it ok , it is working fine.
Thanks!
Angelo

enesbcs, 2020-03-29:
Thank you for your feedback, issue is closed now.

closed 2020-03-29 by enesbcs