enesbcs/rpieasy #136: Adding OS update possibility

enesbcs, 2020-03-23:


enesbcs, 2020-04-10:
Experimental support for OS update added at commit https://github.com/enesbcs/rpieasy/commit/fb77b90db117bccb7c18a9b2423b1a9265045b61

enesbcs, 2020-04-10:
Warning: actual Python library installer 'pip3' requires Python3.5! (For older systems, for example Jessie with python3.4 the latest usable pip is v19.1.1) You can install it manually if you wish.

`sudo pip3 install pip==19.1.1`

angelo1967, 2020-04-10:
> Experimental support for OS update added at commit fb77b90
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "publisher": { "@type": "Organization", "name": "GitHub", "url": "

Hi Alex,
i tested, all ok
for is very good.

Buona Pasqua
Angelo

enesbcs, 2020-04-26:
Closed as implemented.

closed 2020-04-26 by enesbcs