enesbcs/rpieasy #137: deviceID / taskID as variable for MQTT

michapr, 2020-03-25:
Would be nice to get the task number of a device as variable for MQTT reporting.

In this time will be reported (by default %sysname%/#/state):
RPIEasy/[device name]/[parameter]/state/[value]
Because of the device name could be changed and must not be unique would be nice to make it possible to add the ID with (as sample) %TaskID%

Thanks!
Michael

TD-er, 2020-03-25:
Do you mean the number of the task, or the task name?

michapr, 2020-03-25:
of course the number ;)
As I wrote - the *name* must not be unique...

TD-er, 2020-03-25:
Well it is highly recommended to have the name unique, or else rules events and referring to `[taskname#varname]` will be ambiguous.

michapr, 2020-03-25:
yes, right, I know.
For working with MQTT and connected databases etc. would be nice to have an unique ID.

Isn't it available in system so that it could be added easy?

TD-er, 2020-03-25:
I don't know how it is done in R'pi Easy.
In ESPEasy it isn't always available.
It is present in events, but those are not used when you call it from rules.
So in a controller it may be available if you call it from a task, but if you use publish from rules, it is not available.

enesbcs, 2020-03-26:
I agree that unique names are must-have.
In C014 generic MQTT controller - which you may refer for using - the "tasknum" may or may not be forwarded by the sender plugin to the controller, it's up to the plugin. (usually they do) If task number not presented, the MQTT controller identify the plugin by it's taskname.
This feature can be added for testing.

enesbcs, 2020-03-26:
%tskid% variable added at commit https://github.com/enesbcs/rpieasy/commit/4b52d529587d6b1714f7255d8fbff8c7b12aa01b to C014 controller

michapr, 2020-03-28:
Hi,

it is not working for me.

If I use as sample
mp110320/%sysname%/%tskid%/state
Then I will not get the right topics for the system values
I get only "mp110320/RPIEasy/[id]/state" with value for uptime

If I use as sample
mp110320/%sysname%/#/state
I get all - /Load, /RAM, /CPU-Temp, /Uptime

If I use as sample
mp110320/%sysname%/%tskid%/#/state
I get "mp110320/RPIEasy/[id]//state" with value for uptime
(yes, an empty value after ID) - without RAM,Load,CPU

Thanks,
Michael

enesbcs, 2020-03-28:
> I get only "mp110320/RPIEasy/[id]/state" with value for uptime

Please send your tasks.json file for testing.
It is working for me.

I guess you want to use something like this:
`mp110320/%sysname%/%tskid%/%valname%/state`

> If I use as sample
> mp110320/%sysname%/#/state
> I get all - /Load, /RAM, /CPU-Temp, /Uptime

This is the default.

> 
> If I use as sample
> mp110320/%sysname%/%tskid%/#/state
> I get "mp110320/RPIEasy/[id]//state" with value for uptime
> (yes, an empty value after ID) - without RAM,Load,CPU

What do you want to achieve with # ? It is a joker for %tskname%/%valname% . tskname and tskid can not work on the same template so # and tskid also wont work.

michapr, 2020-03-28:
>  tskname and tskid can not work on the same template so # and tskid also wont work.

I didn't know it.

If I use now
mp110320/%sysname%/%tskid%/%valname%/state
it is working.

So if I need to have tskname *and* tskid I shoud add the tskid into the name, as sample "8/system"
(So I made it before)

Thanks!
Michael

enesbcs, 2020-03-28:
Honestly i did not tested %tskid%/%tskname%/%valname% template, it may work to one way.

michapr, 2020-03-29:
> Honestly i did not tested %tskid%/%tskname%/%valname% template, it may work to one way. (for sending,..

Yes, it is working fine.
Maybe to add any info into documentation? ;)

Thanks!
Michael

enesbcs, 2020-03-29:
Volunteers needed to write documentation... plugin headers contains important informations.
Issue is now closed as implemented.

closed 2020-03-29 by enesbcs