enesbcs/rpieasy #138: Beacon Plugin (Presence detection)

Epik77, 2020-04-01:
Hello,
I saw that you have implemented the _P510_ITag.py plugin

I ask you if it was possible to implement a plugin like this:

https://www.domoticz.com/wiki/Presence_detection_(Bluetooth_4.0_Low_energy_Beacon)

Thanks for the continuous development of this software
I remain available to carry out tests

enesbcs, 2020-04-01:
Please be a little more specific. Do you need NUT BLE tag support?

Epik77, 2020-04-02:
hi enesbcs

my request is to be able to detect only the presence of the Address so that I can know the presence of a specific device in the bluetooth network (smartphone, sportwach (mi band), iTag, and any bluetooth device).
If it is also possible to report RSSI in order to calculate the distance.

By implementing this functionality it is possible to extend this application in all home automation processes that require the presence of a device (remove alarm for example).

enesbcs, 2020-04-02:
I see. Then you are suggesting a new plugin, which enters the Bluetooth stack into permanent scanning mode. In theory it can be done, but this plugin will effectively disable all other Bluetooth plugin. As the Bluetooth scan is either working and connecting normally OR it is scanning for available devices, both of them will not work at the same time.

Epik77, 2020-04-02:
Hi,
not a scan but a "ping" check of the address bluethoot.
If the address exist  return 1 on variable (device is on)
if not exist  return 0  on variable (device off)

enesbcs, 2020-04-02:
Current bluepy implementation connect() and isconnected() method is simply not capable to do this reliably. RPIEasy plugins know if device is online if it is responding to a device-specific call. As this is device specific, i am unable to implement this as a generic plugin.
There are no "ping" command for bluetooth. Perhaps Connect() can do that but it is very slow, (even can be 5seconds per device) and it is also blocking Bluetooth from other calls..

The only way i can imagine that is the continous scanning mode.

Epik77, 2020-04-02:
yes you are right i am currently using this script 

https://raw.githubusercontent.com/jmleglise/mylittle-domoticz/master/Presence-detection-beacon/test_beacon.py

together with your project and this generates conflict :-(
I wanted a solution to remove the script and implement it as your project plugin.

enesbcs, 2020-04-02:
Yes, this is a pure HCI implementation, with continous scanning.

I can do something similar with bluepy, but of course other bluetooth plugins will not work together reliably with scanner plugin at the same time. Some BLE peripheral sends back data at random time after the call, and if a scan started in that time window, the data will be lost.

Epik77, 2020-04-02:
can I connect a second bluetooth usb dongle?
This way I can leave it dedicated on this plugin.
obviously the plugins must allow to select the device

enesbcs, 2020-04-02:
For a new scanner plugin, this is possible.

Also P512,P515,P517,P518 can be modified to support dongle selection.

P513 and P516 is hardwired to first dongle in their library.

Epik77, 2020-04-02:
OK thanks
I await your good  news :-)

enesbcs, 2020-04-06:
At commit https://github.com/enesbcs/rpieasy/commit/583b4dba73b98c37676c2c8c001a8821b0fe51df the new BLE scanner plugin added.
Existing BLE plugins prepared for changing dongle HCI0/HCI1, also added a semaphore, so in theory some plugin and the scanner may work on the same dongle. (scanning disconnects existing ble connections anyway)

Epik77, 2020-04-06:
I thank you for the work you have developed.
For a single task, does the P520 plugin work correctly, if I have to add other addresses do I use other tasks?

I purchased an additional dongle, as soon as it arrives I can carry out other tests.

enesbcs, 2020-04-06:
Multiple P520 can work at the same time. The timeout has to be the same, otherwise the results may be surprising. Interval can be different, but on every P520 scan, it will be synchronized with every P520.

enesbcs, 2020-04-14:
Issue closed as addressed but no further feedback provided.

closed 2020-04-14 by enesbcs