enesbcs/rpieasy #139: interval options

rrraf, 2020-04-02:
hi,
thanks again for this nice piece of software. 

it would be usefull to have more options for the intervals (i.e. for mijia devices, mi flower) : to save up battery, I run devices' updates by rules (on clock, taskrun) but the interval option inside each device is max 65k seconds (~18h, less than 24h). options could be : interval (every xx seconds), no auto run, run at xx time (everyday).

regards,
raf

michapr, 2020-04-02:
Hi,
you really think that you save much battery if the interval is > 18hours?
Regardless of whether it really makes sense to poll a sensor only once a day...
If we are speaking about Bluetooth - the quiescent current of the device plays a greater role here, since the advertising process is carried out in any case.

Michael

enesbcs, 2020-04-02:
I agree with michapr that BLE devices will advertise itself eventually regardless if you query it once a day or once an hour.
But with a one-line modification i an modify the BLE plugins to make Interval optional in which way it will only run, if you call it with 'taskrun' from rules.

`self.timeroptional = False`

rrraf, 2020-04-02:
I thought the Bluetooth devices were waiting for handshakes before sending data. I clearly have a battery drop when I have pull data very often (especially on mi flower).

adding that my Bluetooth connections are bad and mostly failing, rpieasy keeps trying until getting full set of data. I'm talking battery drainage, but I'd like to keep the batteries as long as possible.

I'm trying to get a better signal with a dongle and antenna (for temperature mijia, every 5 minutes) because I keep getting errors and long period without receiving any good data. but I really don't care having the mi flower's data more than once a day (but getting them at night is irrelevant).

enesbcs, 2020-04-06:
From commit https://github.com/enesbcs/rpieasy/commit/583b4dba73b98c37676c2c8c001a8821b0fe51df the BLE Mijia plugins interval is now optional, which means that it will not trigger itself is setted to 0, but can be started with taskrun command.
Interval upper limit also raised in webserver.

Existing devices delete&recreate may be necessarry to enforce these changes.

enesbcs, 2020-04-14:
Issue closed as addressed but no further feedback provided.

closed 2020-04-14 by enesbcs