enesbcs/rpieasy #141: miflora read failed = sent 0 temperature

rrraf, 2020-04-14:
hi,
i'm having issues with miflora sensors : i'm receiving temperatures at 0°C whenever an error happens in the reading (and it's quite often, i can't manage to have a good bluetooth reception) 👍

14:57:43 MiFlora error: Miflora read failed
14:57:43 Event: MiFlowerBAR#Temperature=0.0

the value 0 is unfortunately sent to the controller (domoticz in my case). i'm also receiving a lot of 0 battery data (there are all full).

regards,
raf

enesbcs, 2020-04-14:
i can modify this to do not report on error. but how can you find out if the device failed?

enesbcs, 2020-04-14:
Issue fixed at commit https://github.com/enesbcs/rpieasy/commit/4d2d1eeae4dd74214f7ef1050f9a3834dda4c7e9

enesbcs, 2020-04-26:
Closed as no feedback.

closed 2020-04-26 by enesbcs