enesbcs/rpieasy #144: Webserver can not be started, no available port found

dheuts90, 2020-04-22:
I get the error: !Webserver can not be started, no available port found!
How to change the port? Thanks a lot!

Budman1758, 2020-04-22:
Do you have servers running on ports 80 and 8080?  I know that RPIEasy will use those in that order if available.  Not sure if its programmed to search for more or not and I don't believe there is a setting in the config for it.

enesbcs, 2020-04-22:
Port is autoselected. If you are root, and port 80 is free than it is used, if it's not, then port 8080, then finally 8008 is tried.
Please let me now what uses these ports with the following linux command:

`sudo netstat -tulpn | grep LISTEN | grep :80`

enesbcs, 2020-04-22:
Hint: Port 8008 is not used by too many programs, so if it is already taken, it is possible that RPIEasy running in multiple instances at once, which is not a good idea.

dheuts90, 2020-04-28:
Thanks for your responses! Reinstalled it and its working again.

closed 2020-04-28 by dheuts90