enesbcs/rpieasy #145: _P513 BLE Adapter select is missing

michapr, 2020-04-26:
In module is no selection for Adapter - and (in my case) the device will not be polled more.

I have tried to add it in code, but it seems something is missing - can you please add it?

Thanks!
Michael

enesbcs, 2020-04-26:
Not possible. There are no such feature for lywsd02 as Adapter selection.
As i wrote one month ago:

> Also P512,P515,P517,P518 can be modified to support dongle selection.
> P513 and P516 is hardwired to first dongle in their library.
https://github.com/enesbcs/rpieasy/issues/138

You can read about Lywsd02 python module here:
https://github.com/h4/lywsd02

enesbcs, 2020-04-26:
Looks like that Lywsd02() class parameter name changed, and BLEHelper creation needed before connect section. Fixed at https://github.com/enesbcs/rpieasy/commit/3a51188bb6196a592d8ca6fb9de36e228bddb99c

michapr, 2020-04-26:
Great, it is working again!
(I wanted to test with external dongle because of possible connection issues, that's why have seen it. But problem was the lib... ;) )

Have seen another typo in _P518 lib (was not working too):
`    ch = self.BLEPeripheral.getCharacteristics(uuid=self.CGG_DATA)[0]`
should be:
`    ch = self.BLEPeripheral.getCharacteristics(uuid=CGG_DATA)[0]`

Thanks!
Michael

enesbcs, 2020-04-26:
Indeed CGG_DATA is outside of the class. :) Will be added in the next commit!

enesbcs, 2020-04-27:
P518 fixed at commit https://github.com/enesbcs/rpieasy/commit/625ed596d45d3f0bccf45a4d221b66e58c3b85f5

closed 2020-04-27 by enesbcs