enesbcs/rpieasy #148: enhancement: Show last update time

michapr, 2020-04-27:
Hi,
would be great to show below the values the last update time of devices.
So user could see - is the device still responding or is shown an old value.

Thank you for your time!

Best regards
Michael

enesbcs, 2020-04-27:
Originally i've used the battery value to sign this: if this is 0, than connection lost with the device, so the data is old.

michapr, 2020-04-27:
Battery value is not available for all devices in UI. 
And - if you look at the values, you do not know - very old or old or only a connection issue,...

So would be nice to add a line in UI below the values, as sample,...

enesbcs, 2020-04-27:
In theory it is not impossible, _lastdataservetime property stores the last time - per task - when data succesfully read. (measured in milliseconds from rpieasy startup time which may not serves your need)

Do you use any home automation system? For example Domoticz highlight devices that did not update their status since x minutes with red color. I am using this feature to see if a sensor is not responding a long time, as i am not watching all of my RPIEasy UI's all the time, but instead watching for one Domoticz GUI that contains everything.

michapr, 2020-04-27:
ok, I thought to print out the %unixtime% value here when value will be stored.

I'm using different systems (also Domoticz) -in this time all is going via MQTT into a SQL database.
For BLE devices have not decided in this time about the best way - just developing a cheap universal BLE repeater because of the distance / connection issues.

If not possible (or only with hard work) - close the task, it was only an idea to check the device connections while developing ;)

Thanks!
Michael

enesbcs, 2020-04-27:
To print a unique value, like unixtime, all affected plugins has to be modified. (= delete every existing object and recreate it manually)

lastdataservetime is already contained in the object, and can be written in the UI for debugging purposes.

michapr, 2020-05-02:
I have found a workaround - added a dummy device with 4 channels after 4 BLE devices and update each value in rules with %lcltime% after device value change event.

BTW: why _lastdataservetime in device use the uptime value and not the actual timestamp?
Is it needed by design?

Thanks!
Michael

enesbcs, 2020-05-02:
lastdataservetime is meant for the internal scheduler in millisecond, some plugin needs 10, some other 50 call per seconds for GPIO handling or network/uart polling. But startup time is also saved, so something similar can be implemented someday for debug mode.

enesbcs, 2020-05-06:
Last read time added to device page if Tools/Advanced set Web log level setted to "Debug More" at commit https://github.com/enesbcs/rpieasy/issues/148

michapr, 2020-05-07:
Great!!
A little improvement would be nice ;) ... 
The initial value will be set starting up the service -and not after first reading.
It's a minor effect, if much trouble to change it, no problem ;)

Thanks!!
Michael

enesbcs, 2020-05-07:
> Edit: it seems not to work for Dummy task and for OS command task ?
> Here is the time the startup time and will not be changed
I've tested it with Dummy task and manual Run. If interval is 0 than it will not update time yes.
OS command task is an output, which has no plugin_read() function, so lastdataservetime will never update, other output only devices will work similarly. It is meant for sensor reading.

michapr, 2020-05-07:
> I've tested it with Dummy task and manual Run. If interval is 0 than it will not update time yes.

I have tested with update device via rules - will not update time, even it update time =1

Maybe (if not too difficult) to show the update time only for devices where it can work?
New users will better understand it then ;)

Thanks!
Michael

enesbcs, 2020-05-10:
Fixed at commit https://github.com/enesbcs/rpieasy/commit/00636f5229a0cefe24ff873d7b101b847f601633

michapr, 2020-05-19:
Is working fine! 

Thanks!
Michael

haraldtux, 2020-05-23:
Is working fine!
Harald

michapr, 2020-05-23:
I have seen that the update time will not be refreshed more - not sure, but I think it was working before ;) ...

Thanks!

enesbcs, 2020-05-23:
Clock on the bottom of the page is working right now.
If you are insisting the timestamp of every device row, than it was not working automatically - you can hit CTRL-R in you browser, or if anyone have some time to code it in javascript, do it baldly.
I am using javascript as little as possible, as it is effectively eats client cpu time for nothing.

enesbcs, 2020-08-01:
Closed as nobody interested to do it

closed 2020-08-01 by enesbcs