enesbcs/rpieasy #149: plugin  pi servo

Epik77, 2020-05-20:
hi Alexander,

I wanted to know if more servo support can be implemented

https://www.letscontrolit.com/wiki/index.php/Servo

I use Pi Servo with a camera and it would be useful
command it via rpieasy
https://www.instructables.com/id/Raspberry-Pi-Cam-Pan-Tilt-Control-Over-Local-Inter/

Thank's for your support.

enesbcs, 2020-05-20:
In theory it can be implemented with RPIEasy, using this simple library:
https://github.com/Mjrovai/OpenCV-Object-Face-Tracking/blob/master/angleServoCtrl.py

In the meantime, Stepper motor is usable with P204 and 28BYJ-48:
https://github.com/enesbcs/rpieasy/blob/master/_P204_Stepper.py

Epik77, 2020-05-21:
Sorry but P204 is not compatible with SG90
I will wait for your new plugins in the meantime I adapt with the script
and the P511 plugin

Thank's for your support.

enesbcs, 2020-05-21:
Yep, the stepper motor is just an alternative. 
I can add servo command based on pwm capabilities, but keep in mind that Raspberry Pi only has 4 possible hardware PWM pin (GPIO12, GPIO18, GPIO13, GPIO19), which is also used by the onboard sound card circuit of the Pi. So make sure to disable integrated sound, and enable HarwarePWM on the specific pin, before use servo command. Otherwise software pwm will be used, which is not so precise.

enesbcs, 2020-05-23:
servo command added at commit 
https://github.com/enesbcs/rpieasy/commit/69f1156ec80a7765c58912d2dea16c4b03d8e355

Can be used with:

> Servo,<servo nr>,<pin>,<position>

The selected pin has to be setted up for Output at least, before use, and an enabled P001/P029 plugin is also needed as GPIO is a run-time pulled module in RPIEasy. (RPi/OPI/FTDI or NoGpio mode at all)

Epik77, 2020-05-25:
Hi,

I confirm correct operation
Command:  Servo,1,17,150
{
"pin": 17,
"state": 0,
"log": "BCM17 to servo 150 angle",
"mode": "servo",
"plugin": 1
}

Servo,1,17,150
{
"pin": 27,
"state": 0,
"log": "BCM27 to servo 150 angle",
"mode": "servo",
"plugin": 1
}

Thank you very much

enesbcs, 2020-05-25:
Thank you for your feedback, i guess you are used SoftPWM. :) 
In theory the four HardPWM also can be used, but functions slightly differs from the sample applications.
Issue is now closed as implemented, feel free to reopen in case of related problems.

closed 2020-05-25 by enesbcs