enesbcs/rpieasy #150: InfluxDB controller

teepox, 2020-05-21:
Hi,
Thanks for creating something which supports Xiaomi Mijia Temperature & Humidity sensors out of the box!

I'm trying to feed the temperature & humidity data provided by them to InfluxDB, which is not natively supported. Using Telegraf socket listener input with generic UDP controller in rpieasy should be a feasible alternative, but it requires using one of the supported input data formats: https://github.com/influxdata/telegraf/blob/master/docs/DATA_FORMATS_INPUT.md

Implementing the InfluxDB line protocol is probably the easiest way, and it seems I'm very close to getting it working, but I can't find a way to insert a whitespace into the report template of the generic UDP controller. Looking at the documentation of ESP easy, they seem to be working on native InfluxDB support via the HTTP API, and they are using %LF% for inserting the whitespace:
https://www.letscontrolit.com/wiki/index.php?title=EasyProtocols#InfluxDB_HTTP_Api_.28via_Generic_HTTP_Advanced.29

I tested using %LF% on rpieasy, but apparently it is not implemented here. Is there already a way to insert a whitespace into the report template, or could the %LF% way be implemented on rpieasy?

enesbcs, 2020-05-21:
Indeed, the things i've never used in ESPEasy is not implemented in RPIEasy. :)

%LF% implemented in commit https://github.com/enesbcs/rpieasy/commit/311f63b93bffe19ef95134324b4ba960301c8c2f

teepox, 2020-05-23:
Thanks for implementing LF. That got me wondering how the InfluxDB integration is supposed to work on ESP easy, as the InfluxDB line protocol they appear to be using requires a whitespace, not an LF. I would offer a pull with a sysvar for whitespace, but I can't decide what to call that variable :)

enesbcs, 2020-05-23:
%TAB% ? :)

teepox, 2020-05-25:
Tab seems to work too, so let's go with that.

enesbcs, 2020-05-25:
Thanx, merged!

closed 2020-05-26 by teepox