enesbcs/rpieasy #152: local variable 'rssi' referenced before assignment

matriedel, 2020-05-24:
I have installed rpieasy on a RPI 1 without WIFI connected via Ethernet

Build | RPIEasy 2.0.144
Libraries | Python 3.7.3 (default, Dec 20 2019, 18:57:59)[GCC 8.3.0] Linux-4.19.97+-armv6l-with-debian-10.4

When GPIO-4 (Input - Switch Device/Generic GPIO) is triggered I see the following error

22:44:34: Event: Gastrigger#state=0
Plugin SendData Exception:  local variable 'rssi' referenced before assignment ['0', 0, 0, 0]

I have attached two controllers

Generic MQTT and DB Data Storage

I see in via MySQL Workbench that the table has been set up but no trigger event is stored

Do you have any hint ?

matriedel, 2020-05-25:
I think the error comes from getCachedRSSI function in  _C016_DBStore.py. Assigning first  rssi = urssi the problem is solved

enesbcs, 2020-05-25:
Thank you for reporting this. I've never tested C016 with LAN connection only earlier.

enesbcs, 2020-05-25:
Fixed at commit https://github.com/enesbcs/rpieasy/commit/8f5f8f0bd69eb835af1fc5ee3392af2a2bcf5d8d

matriedel, 2020-05-25:
Yes, this is the solution.

closed 2020-05-25 by matriedel