enesbcs/rpieasy #155: New plugin Keypad with pure GPIO

enesbcs, 2020-05-27:
Pad4Pi is a nice library for implementing Keypads (3x4/4x4) RPIEasy.
https://github.com/brettmclean/pad4pi
As Raspberry has 17 GPIO, it can be done without any extender.

enesbcs, 2020-05-28:
pad4pi works fine, but it is returning the characters printed on the buttons as default. I can modify it to return these scancodes instead for further processing: https://www.letscontrolit.com/forum/viewtopic.php?p=21435#p21435

Budman1758, 2020-05-30:
> pad4pi works fine, but it is returning the characters printed on the buttons as default. I can modify it to return these scancodes instead for further processing: 

Thats probably not really necessary. I'm sure the numbers can be processed either way just as easily.

enesbcs, 2020-05-30:
In the end i've reimplemented the whole keypad library.. as pad4pi occasionally skipped the 3rd row button presses somehow. I doubt that it is hardware related problem, as all button worked with direct readout.
Support introduced in commit https://github.com/enesbcs/rpieasy/commit/b93d56ad2a6cd3f700fcc130559bbc652ab2992c Tested with a 3x4 keypad.

closed 2020-06-17 by enesbcs