enesbcs/rpieasy #156: multiple BLE Xiaomi LYWSD03 - BLE adapter is hanging

michapr, 2020-06-13:
Hi,

I have now integrated multiple BLE Xiaomi LYWSD03 devices, they are working fine for a while.

After some time the BLE adapter cannot connect more to these devices - to other Bluetooth devices all is still ok - so BLE-adapter itself is still working.

Nothing help - no restart adapter, no restart software or bluetooth service...
The only solution is a reboot of the Pi - what is not the best solution.
Then all is working again for some hours (up to 18hours).

Already have tried to modify something in code (sleep time, inserting disconnect at one place) - but no real solution found.

Maybe somebody have an idea?

Thanks!
Michael

enesbcs, 2020-06-13:
Maybe the underlaying bluepy python library stucks, that handles BLE requests (there are a background process named "bluepy-helper" which may freeze). In this case there are no real solution - there are 156 opened issues at bluepy page:
https://github.com/IanHarvey/bluepy

Or you can try to remove the LYWSD03 device with the lowest RSSI from the branch and check out if it changes anything.

michapr, 2020-06-14:
For testing all LYWSD03 are really near the Pi - about 1 meter. RSSI seems not to be the problem, because MiFlora devices are far (70-80dBm), and they are working stable - even after the time when LYWSD03 is not responding more.

So the bluepy-helper is not (the only) problem, as far as I understand.

I think there maybe any internal process with opened and not closed connections to the devices or any other blocking process - because only these devices affected...

(BTW: I have of course checked the bluepy page and issues, but didn't found this case but other interesting things ;) ...)

Thanks!
Michael

michapr, 2020-08-06:
Still the same problem, no idea...
The Pi must be rebooted when BLE is hanging. Solved with timer and checking last value update for multiple sensors.
If all sensors will not be updated more Pi will be rebooted.

enesbcs, 2020-08-19:
It is in an early stage, but i am now playing with micropython + BLE. I can not yet say anything about stability, but i feel connection to LYWSD sensors faster with ESP32.

enesbcs, 2020-09-03:
BLE plugins modified to one device use BLE at a time. May help hanging a bit. Commit https://github.com/enesbcs/rpieasy/commit/2d7d900686c6bac56b14681d2e831aa8742c0812
Please backup config before updating

michapr, 2020-09-04:
Something seems to be wrong after update (via web)...
with old (working) config:
```
...
08:43:53: BLE sniffer init ok
08:43:53: BLE sniffer init ok
'list' object has no attribute 'enabled'
08:43:53: Webserver starting at port 80

```
with empty data folder:
```
Creating new pinout
Load devices from file
Load controllers from file
...
08:45:13: BLE sniffer init ok
08:45:13: BLE sniffer init ok
'select.poll' object has no attribute 'enabled'
08:45:13: Webserver starting at port 80
```

- system is hanging, no connection via web.

michapr, 2020-09-04:
> BLE plugins modified to one device use BLE at a time. May help hanging a bit. Commit 2d7d900

Sorry to say - it does not help.
Same as before. I think it is any issue in global BLE lib.

last lines in log:
```
Sep  4 15:31:57 RPIEasy RPIEasy98: 3f:5b:7d:71:7c:38 {'temp': 23.3, 'temp-t': 1599226270.4036047, 'hum': 65.0, 'hum-t': 1599226269.4016993}
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: FlowerCare2#Light=1475.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: LYWSD02-S#Temp=23.3
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: ClearGrassTemp&RH-S#Hum=61.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: FlowerCare1#Fert=1529.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: LYWSD02-S#Hum=65.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: FlowerCare2#Fert=796.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: ClearGrassTemp&RH-S#Batt=6.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: system#Uptime=8109.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: system#RAM=374688.0
Sep  4 15:31:57 RPIEasy RPIEasy98: BLE line not free for P518!
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: system#Load=0.0
Sep  4 15:31:57 RPIEasy RPIEasy98: Event: system#CPU-Temp=36.9
Sep  4 15:31:58 RPIEasy RPIEasy98: BLE line not free for P518!
Sep  4 15:31:58 RPIEasy RPIEasy98: BLE line not free for P518!
Sep  4 15:31:58 RPIEasy RPIEasy98: BLE read error: Failed to connect to peripheral 3f:5b:7d:71:7c:38, addr type: public
Sep  4 15:31:59 RPIEasy RPIEasy98: BLE line not free for P518!
Sep  4 15:31:59 RPIEasy RPIEasy98: BLE connection initiated to 58:2d:34:10:54:50
Sep  4 15:32:02 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:03 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:03 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:04 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:04 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:05 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:05 RPIEasy RPIEasy98: BLE connected to 58:2d:34:10:54:50
Sep  4 15:32:05 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:06 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 15:32:06 RPIEasy RPIEasy98: BLE line not free for P513!
..
Sep  4 16:12:37 RPIEasy RPIEasy98: BLE line not free for P513!
Sep  4 16:12:37 RPIEasy RPIEasy98: BLE line not free for P513!
```
After restarting application all is working again (set hci0 down/up does not help)

enesbcs, 2020-09-04:
I guess that the "58:2d:34:10:54:50" is a LYWSD03?

enesbcs, 2020-09-04:
> ...

michapr, 2020-09-04:
no,... 
in other issue is my backup file ;)

```
 3f:5b:7d:71:7c:38 | public | -58 dBm  | LYWSD02 |   
 a4:c1:38:5e:e1:0f | public | -48 dBm | MHO-C401 | 
 c4:7c:8d:6b:3f:3f | public | -79 dBm |  Flower care |  
 a4:c1:38:f4:84:a9 | public | -90 dBm | LYWSD03MMC |   
 a4:c1:38:6b:f6:b2 | public | -52 dBm | LYWSD03MMC |   
 a4:c1:38:e6:41:f3 | public | -84 dBm | MHO-C401 |  
 00:1a:22:08:bf:d6 | public | -74 dBm | CC-RT-BLE |   
 c4:7c:8d:6a:08:85 | public | -78 dBm | Flower care | 
 a4:c1:38:c1:8c:71 | public | -59 dBm | LYWSD03MMC | 
0 | a4:c1:38:82:9f:32 | public | -75 dBm | True | MHO-C401 |  
```

enesbcs, 2020-09-04:
> with empty data folder:
> 
> ```
> Creating new pinout
> Load devices from file
> Load controllers from file
> ...
> 08:45:13: BLE sniffer init ok
> 08:45:13: BLE sniffer init ok
> 'select.poll' object has no attribute 'enabled'
> 08:45:13: Webserver starting at port 80
> ```
> 
> 
> EDIT: removing the backup folder system is starting - but not working in my case....
> (from where the program take the current settings / tasks if data folder is empty?)

When the program is running it hold configuration in memory. It will save config into data folder from memory on restart. Unless you call a reset setting command. So in short: data folder is not empty in this case.

enesbcs, 2020-09-04:
> no,...
> above is my backup file ;)
> 
> ```
>  3f:5b:7d:71:7c:38 | public | -58 dBm  | LYWSD02 |   
>  a4:c1:38:5e:e1:0f | public | -48 dBm | MHO-C401 | 
>  c4:7c:8d:6b:3f:3f | public | -79 dBm |  Flower care |  
>  a4:c1:38:f4:84:a9 | public | -90 dBm | LYWSD03MMC |   
>  a4:c1:38:6b:f6:b2 | public | -52 dBm | LYWSD03MMC |   
>  a4:c1:38:e6:41:f3 | public | -84 dBm | MHO-C401 |  
>  00:1a:22:08:bf:d6 | public | -74 dBm | CC-RT-BLE |   
>  c4:7c:8d:6a:08:85 | public | -78 dBm | Flower care | 
>  a4:c1:38:c1:8c:71 | public | -59 dBm | LYWSD03MMC | 
> 0 | a4:c1:38:82:9f:32 | public | -75 dBm | True | MHO-C401 |  
> ```

Then what is this from your logs?

Sep  4 15:31:59 RPIEasy RPIEasy98: BLE connection initiated to 58:2d:34:10:54:50

michapr, 2020-09-04:
> Then what is this from your logs?

Sorry.. was missing in actual scan - have not seen. added ... ;)

enesbcs, 2020-09-04:
> > 
> 
> Sorry.. was missing in actual scan - have not seen. added ... ;)

So it is a Cleargrass, using P518?

michapr, 2020-09-04:
> So it is a Cleargrass, using P518?

yes.

enesbcs, 2020-09-04:
> > So it is a Cleargrass, using P518?
> 
> yes.

Ok, than i am looking into P518

michapr, 2020-09-04:
Sure? I thought the P517 is the problem...

>It will save config into data folder from memory on restart. 

I already thought so. Anyway - I made a erase JSON settings from web, restarted app and could not load my old config.
Same as after web update.

I just now made a apt-get upgrade - have seen that new BT files and chip firmware. Maybe something changes ;)

enesbcs, 2020-09-04:
> Sure? I thought the P517 is the problem...

In this case, according your log this was the last device which accessed BLE:

Sep  4 15:32:05 RPIEasy RPIEasy98: BLE connected to 58:2d:34:10:54:50

> I already thought so. Anyway - I made a erase JSON settings from web, restarted app and could not load my old config.
> Same as after web update.

I will try this config, replacing device MAC's with mine.

enesbcs, 2020-09-04:
> EDIT2: only new installation (via git) is working again. my old data.zip is not working more.
> [data5.zip](https://github.com/enesbcs/rpieasy/files/5174552/data5.zip)

Your tasks.json in data5.zip contains invalid JSON structures, it can not be used anymore. Even jsonpickle was unable to understand what is this. The error is after the eq3 plugin in the json.

michapr, 2020-09-04:
>Your tasks.json in data5.zip contains invalid JSON structures

Interesting... I also thought about it. But I could open it with JSON viewer.
I made a normal backup before updating. eq3 was disabled.
Already made all at new, not a big deal. It happen while testing... ;)

enesbcs, 2020-09-04:
Yes it can be opened.
But i can not imagine a scenario when the "_conn" object of _P516 is another plugin like P513... and the "blestatus" of P513 is a P517 object. Something is reallly missing and mixed like in case of buffer overflow. Are you using the latest version of jsonpickle?

michapr, 2020-09-04:
> Are you using the latest version of jsonpickle?

It was installed together with rpieasy... this summer
Now is updated all, do not know what was old version, maybe it was updated today.

michapr, 2020-09-05:
multiple LYWSD03/MHO-C401 are working now fine (since 23hours without problem...)

Will check next 3 days and close then.

THANKS!

enesbcs, 2020-09-05:
I hope this will fix the problem.

michapr, 2020-09-08:
Is working stable if not save new config for many sensors within short time - then restart app will recover the stable work.

Thanks!

closed 2020-09-08 by michapr