enesbcs/rpieasy #157: _P206_PZEM016 problem after upgrade  PZEM init failed at address 1

Epik77, 2020-06-16:
Hi 

I have upgraded my rpieasy but this device not working...

by trying with this script the device works correctly:

# Reading PZEM-004t power sensor (new version v3.0) through Modbus-RTU protocol 
over TTL UART
# Run as:
# python3 pzem_004t.py

# To install dependencies: 
# pip install modbus-tk
# pip install pyserial

import serial
import modbus_tk.defines as cst
from modbus_tk import modbus_rtu

# Connect to the slave
serial = serial.Serial(
                       port='/dev/ttyUSB0',
                       baudrate=9600,
                       bytesize=8,
                       parity='N',
                       stopbits=1,
                       xonxoff=0
                      )
master = modbus_rtu.RtuMaster(serial)
master.set_timeout(3.0)
master.set_verbose(True)

data = master.execute(1, cst.READ_INPUT_REGISTERS, 0, 10)

voltage = data[0] / 10.0 # [V]
current = (data[1] + (data[2] << 16)) / 1000.0 # [A]
power = (data[3] + (data[4] << 16)) / 10.0 # [W]
energy = data[5] + (data[6] << 16) # [Wh]
frequency = data[7] / 10.0 # [Hz]
powerFactor = data[8] / 100.0
alarm = data[9] # 0 = no alarm

print('Voltage [V]: ', voltage)
print('Current [A]: ', current)
print('Power [W]: ', power)
print('Energy [Wh]: ', energy)
print('Frequency [Hz]: ', frequency)
print('Power factor []: ', powerFactor)
print('Alarm : ', alarm)
# Changing alarm W value
#master.execute(1, cst.WRITE_SINGLE_REGISTER, 1, output_value=100)

try:
    master.close()
    if slave.is_open:
        slave.close()
except:
    pass

can you help me?
Thank's

enesbcs, 2020-06-16:
This plugin is working only with the Modbus RS485 PZEM016, you are referring to PZEM004 which is using Modbus Rs232 which is currently not supported.

Epik77, 2020-06-16:
sorry until yesterday before the upgrade worked properly
Has anything changed in serial management?

enesbcs, 2020-06-16:
Interesting, i did not changed anything in the serial code. Could you delete and recreate the P206 plugin for testing?
Do you remember from which RPIEasy version did you upgraded? 
The serial library changed to search ports aggressively if empty list returned by the pyserial lib, but if the array was not empty, there are no change at all.
I've done some changes in plugin.py 3 months ago by adding one variable globally. Also have a minor, cosmetic change in P206 to add com port address to the device list in the webpage.

Epik77, 2020-06-16:
I restored an old version:
Build RPIEasy 2.0.131
Libraries Python 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0] Linux-4.19.97 + -armv6l-with-debian-10.3
Plugins 74
Now it is recognized:
Init hardware ...
Load network settings ...
Load GPIO settings ...
Creating new pinout
Load devices from file
Load controllers from file
Load notifiers from file
19:27:18: Event: System # Boot
19:27:18: Webserver starting at port 8008
19:27:18: Event: Clock # Time = Tue, 19: 27
19:28:00: Event: Clock # Time = Tue, 19: 28
19:29:00: Event: Clock # Time = Tue, 19: 29
19:30: Event: Clock # Time = Tue, 19:30
19:31:00: Event: Clock # Time = Tue, 19: 31
19:32:00: Event: Clock # Time = Tue, 19: 32
19:33:00: Event: Clock # Time = Tue, 19: 33
19:33:06: PZEM initialized at: / dev / ttyUSB0 / 1
19:33:07: Event: Test # Amper = 1.7
19:33:09: PZEM initialized at: / dev / ttyUSB0 / 1
19:33:37: PZEM initialized at: / dev / ttyUSB0 / 1
19:33:38: Event: Test # Amper = 1.7
19:33:40: PZEM initialized at: / dev / ttyUSB0 / 1
19:33:41: Event: Test # Amper = 1.7
19:34:00: Event: Clock # Time = Tue, 19:34
19:34:43: Event: Test # Amper = 1.7
19:34:48: PZEM initialized at: / dev / ttyUSB0 / 1
19:34:48: Event: Test # Volt = 232.4
19:34:49: Event: Test # Amper = 1.7
19:34:56: PZEM initialized at: / dev / ttyUSB0 / 1
19:34:57: Event: Test # Amper = 1.7
19:35:00: Event: Clock # Time = Tue, 19: 35
19:35:11: PZEM initialized at: / dev / ttyUSB0 / 1
19:35:13: Event: Test # Wh = 173137.0
19:35:18: PZEM initialized at: / dev / ttyUSB0 / 1
19:35:20: Event: Test # Volt = 233.0
19:35:20: Event: Test # Amper = 1.7
19:35:27: Event: Test # Amper = 1.7
19:35:27: Event: Test # Wh = 173138.0
19:35:47: Event: Test # Amper = 1.7
19:35:54: Event: Test # Amper = 1.7
19:36:00: Event: Test # Amper = 1.7
19:36:01: Event: Clock # Time = Tue, 19:36

you could create a plugin with script specifications
that I indicated to you?

enesbcs, 2020-06-16:
> you could create a plugin with script specifications
> that I indicated to you?

If it was worked before, there are no reason to reinvent the wheel.
Just have to figure out why init failed in the newer version.

enesbcs, 2020-06-16:
I've added some more checks and debug lines into P206 at commit https://github.com/enesbcs/rpieasy/commit/1a8e69286f0791bb972bf0c124812e6fdd5e2ef2
By the way PZEM016 is working to me: (after i figured out what address i had moved it earlier)
```

Starting RPIEasy 2.0.168
Init hardware...
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
22:46:32: PZEM initialized at: /dev/ttyUSB0 / 1
Load notifiers from file
Loading rules...
22:46:32: Event: System#Boot
22:46:32: Webserver starting at port 8080
22:46:32: Event: Clock#Time=Tue,22:46
22:46:32: Event: Network#Connected
22:46:46: Event: pz1#Volt=235.2
22:46:46: Event: pz1#Amper=0.0
22:46:46: Event: pz1#Watt=0.0
22:46:46: Event: pz1#Wh=13.0
```

Epik77, 2020-06-16:
I'm going crazy :-)
if I start rpieasy without any task and create a new _P206 task
it works properly.
If I start rpieasy with other tasks configured _P206 it cannot initialize

enesbcs, 2020-06-16:
Please send me your tasks.json somehow. I have added several tasks and P206 still works. 
Maybe there are only one of the used plugins have an init bug.

Epik77, 2020-06-16:
I sent you the task.json
at rpieasy@noreply.github.com
It is valid ?

enesbcs, 2020-06-16:
> I sent you the task.json
> at [rpieasy@noreply.github.com](mailto:rpieasy@noreply.github.com)
> It is valid ?

Nope. You can add it by https://pastebin.com/ or send to rpieasy@altmails.com

Epik77, 2020-06-17:
Hi
I sent the task to rpieasy@altmails.com
By updating rpieasy and debugging from console enabled:

From https://github.com/enesbcs/rpieasy
   61dfe5a..1a8e692  master     -> origin/master
Updating 61dfe5a..1a8e692
Fast-forward
 RPIEasy.py       |  2 +-
 _P206_PZEM016.py | 26 ++++++++++++++++++++------
 lib/lib_pzem.py  | 14 ++++++++++++--
 rpieGlobals.py   |  2 +-
 4 files changed, 34 insertions(+), 10 deletions(-)
./RPIEasy.py 
Starting RPIEasy 2.0.168
Init hardware...
07:53:42: Pi 1 Model B+ 40 pins
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
07:53:46: State preserving is disabled
07:53:46: State preserving is disabled
07:53:46: State preserving is disabled
PZEM Exception:  No communication with the instrument (no answer)
07:53:47: !PZEM init failed at address 1
07:53:47: !PZEM init failed at address 1
07:53:47: !PZEM init failed at address 1
07:53:47: !PZEM init failed at address 1
07:53:47: Event: Campanello#State=0
07:53:47: Event registered to pin 18
07:53:47: State preserving is disabled
07:53:47: Event: SegnalatoreAcustico#State=0
07:53:47: !Event can not be added, register backup timer You must setup() the GPIO channel as an input first
07:53:47: Event: Volume#Volume=80
07:53:47: BLE scan init ok
07:53:47: Event: LettoreNFC#State=1
07:53:47: BLE scan init ok
07:53:48: Stopping transmission-daemon (via systemctl): transmission-daemon.service.
07:53:48: OS command executed succesfully
07:53:48: Event: Torrent#State=0
07:53:48: BLE scan init ok
07:53:48: Webserver starting at port 8008

Epik77, 2020-06-17:
I commented on these lines:
lib_pzem.py [33-36]
#   testval = self.dev.read_register(self.VOLT,1,4)
#   if testval == None or testval == False:
#    self.initialized = False
#    self.dev = None

Now the device is successfully initialized
but it mainly provides only the amper data

Starting RPIEasy 2.0.168
Init hardware...
10:16:24: Pi 1 Model B+ 40 pins
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
10:16:26: PZEM initialized at: /dev/ttyUSB0 / 1
10:16:26: PZEM initialized at: /dev/ttyUSB0 / 1
10:16:26: Event: DomoMQTT#Disconnected
10:16:26: MQTT: Try to connect
10:16:26: Set MQTT password
10:16:26: Event: DomoMQTT#Connected
Load notifiers from file
Loading rules...
10:16:26: Event: System#Boot
10:16:26: CMD: lcdcmd,on
10:16:26: !Unknown command: lcdcmd,on
10:16:26: Webserver starting at port 8008
10:16:27: Event: Clock#Time=Wed,10:16
10:16:32: Event: TEst4#Amper=1.5
10:16:32: Event: TEst4#Amper=1.5
10:16:36: Event: TEst4#Amper=1.5
10:16:42: Event: TEst4#Amper=1.5
10:16:44: Event: test#Hz=50.0
10:16:47: Event: TEst4#Amper=1.5
10:16:51: Event: TEst4#Amper=1.5
10:16:53: Event: test#Wh=178078.0
10:16:56: Event: TEst4#Amper=1.5
10:16:57: Event: TEst4#Wh=178078.0
10:17:00: Event: Clock#Time=Wed,10:17
10:17:06: Event: TEst4#Amper=1.5
10:17:21: Event: TEst4#Amper=1.5
10:17:26: Event: TEst4#Amper=1.5
10:17:29: Event: test#Watt=242.9

I need the system to detect amperes and watts equally

enesbcs, 2020-06-17:
> I commented on these lines:
> lib_pzem.py [33-36]
> # testval = self.dev.read_register(self.VOLT,1,4)
> # if testval == None or testval == False:
> # self.initialized = False
> # self.dev = None
> 
> Now the device is successfully initialized

This means that the first read from Modbus device was not succesful, and with disabling testing it can read sometimes.
This line tells the whole story:
"PZEM Exception: No communication with the instrument (no answer)"
Maybe i can add some settings to modify timeout and retry count.

> but it mainly provides only the amper data
If no answer, there were no values, for sure.

enesbcs, 2020-06-17:
Added a new "Increase timeout for slow PZEM004 compatibility" checkbox at commit https://github.com/enesbcs/rpieasy/commit/3d7b64f996e9a9e121fbc3afee2f9b651df1c5b4
This will setup 3 seconds timeout - as in your example - and 3 retry count if no result at read.
I hope it solves the problem. RS485 PZEM016 has no problem with 0.1s timeout, it is fast.

enesbcs, 2020-06-17:
> I sent the task to [rpieasy@altmails.com](mailto:rpieasy@altmails.com)

I've received it, and works OK on my Raspberry PI Zero W. I saw that almost all task slots (47) are used in your config which is amazing, never saw similar before.

Be advised that PZEM004 is a relatively slow device, 1 data read / second is the most what you can get. So if you have 6 tasks which reads the same RS232 device you can not use Interval:1 second in any of your tasks, it will block communication line from the other tasks. (as your Amper device)

> I need the system to detect amperes and watts equally

Solutions:
1. Do not use lower intervals than 10seconds in your Tasks
2. Do not use more tasks for the same device
3. Buy a PZEM016 which is faster

Epik77, 2020-06-17:
I created several tasks because so I can send them to the controller in separate domoticz devices

I applied the patch, the situation is better.
however, the problem remains that consumption watts
it is not aligned with amps.

The script I passed you in the first answer gives me all the updated values in a single query

Loading rules...
17:25:48: Unit alive: 2
17:25:59: PZEM initialized at: /dev/ttyUSB0 / 1
BLE error:  Device disconnected
17:26:01: Event: MiBandLorenzo#Online=0
17:26:01: BLE scan ended
17:26:01: Event: Clock#Time=Wed,17:26
17:26:02: Event: TemperaturaDevil03#Temp=42.2
17:26:03: Event: Pzem#Volt=234.6
17:26:04: Event: CPU#Load=65.8
17:26:04: PZEM initialized at: /dev/ttyUSB0 / 1
17:26:04: Event: Pzem#Amper=4.2
17:26:05: Event: Ampere#Amper=4.1
17:26:06: Event: IphoneEnzoPing#State=0
17:26:09: Event: IphoneMaryPing#State=0
17:26:10: Event: Ampere#Amper=4.1
17:26:14: Event: Ampere#Amper=4.0
17:26:19: Event: Ampere#Amper=3.8
17:26:23: Event: Ampere#Amper=3.8
17:26:27: Event: Ampere#Amper=3.8
17:26:31: PZEM initialized at: /dev/ttyUSB0 / 1
17:26:32: Event: Pzem#Amper=3.9
17:26:34: Event: Ampere#Amper=3.9
17:26:39: Event: Ampere#Amper=4.0
17:26:43: Event: Ampere#Amper=3.7
17:26:45: Event: Pzem#Amper=3.7
17:26:47: PZEM initialized at: /dev/ttyUSB0 / 1
17:26:49: Event: Pzem#Wh=181810.0
17:26:49: Event: Ampere#Amper=3.8
17:26:53: Event: Ampere#Amper=3.8
17:26:57: Event: IphoneLorenzoPing#State=0
17:26:57: Event: LettoreNFC#State=1
17:26:57: Parsed condition: if 1 == 0 False
17:26:57: Parsed condition: if 1 == 1 True
17:26:57: CMD: gpio,13,1
17:26:57: BCM13 set to 1
17:26:57: Event: NFC#State=1
17:26:57: Event: Ampere#Amper=4.0
17:27:00: PZEM initialized at: /dev/ttyUSB0 / 1
17:27:01: Event: Clock#Time=Wed,17:27
17:27:03: Event: TemperaturaDevil03#Temp=42.8
17:27:06: Event: CPU#Load=39.6
17:27:06: Event: FattorePotenza#CosF=1.0
17:27:07: BLE scan started
17:27:07: Event: Ampere#Amper=3.8
17:27:08: PZEM initialized at: /dev/ttyUSB0 / 1
17:27:10: Event: IphoneEnzoPing#State=0
17:27:18: Unit alive: 2
17:27:37: PZEM initialized at: /dev/ttyUSB0 / 1
17:27:42: PZEM initialized at: /dev/ttyUSB0 / 1
17:27:48: Unit alive: 2
BLE error:  Device disconnected
17:28:07: BLE scan ended
17:28:07: Event: Clock#Time=Wed,17:28
17:28:08: BLE connection initiated to a4:c1:38:23:2a:bd
17:28:08: Event: TemperaturaDevil03#Temp=42.8
17:28:09: Event: Pzem#Amper=3.6
17:28:10: Event: CPU#Load=63.8
17:28:10: Event: IphoneRosyPing#State=0
17:28:10: Event: Ampere#Amper=3.6
17:28:12: Event: Consumo#Wh=181830.0
17:28:15: Event: IphoneMaryPing#State=0
[... 175 lines cut ...]
17:38:06: Event: Ampere#Amper=4.1
17:38:12: !BLE connection failed 4c:65:a8:dc:01:54
17:38:13: BLE connection initiated to 4c:65:a8:dc:01:54
17:38:13: Event: Ampere#Amper=4.2
17:38:18: BLE connected to 4c:65:a8:dc:01:54
17:38:19: Event: Ampere#Amper=4.4
17:38:21: Event: Temperatura_disimpegno#Temperature=25.3
17:38:21: Event: Temperatura_disimpegno#Humidity=62.0
17:38:21: Event: Temperatura_disimpegno#Battery=69.0
17:38:25: Event: IphoneMaryPing#State=0
17:38:26: BLE scan started
17:38:26: Event: Ampere#Amper=4.2
17:38:51: Unit alive: 2
17:39:22: Unit alive: 2
BLE error:  Device disconnected
17:39:26: BLE scan ended
17:39:26: Event: Clock#Time=Wed,17:39
17:39:28: Event: TemperaturaDevil03#Temp=42.8
17:39:29: Event: CPU#Load=45.9
17:39:30: Event: Ampere#Amper=4.3
17:39:31: Event: IphoneEnzoPing#State=0
17:39:36: Event: Ampere#Amper=4.2
17:39:40: !DHT read failed, using cached value!
17:39:42: Event: Ampere#Amper=4.1
17:39:54: Event: Ampere#Amper=4.1

enesbcs, 2020-06-17:
I see. Honestly this plugin was never ment to be used with PZEM004, it is made for PZEM016 and it works well with it. 

I've opened a new issue for a new plugin, which may be added someday, if i see a cheap pzem004 for testing.
https://github.com/enesbcs/rpieasy/issues/158

closed 2020-06-17 by enesbcs