enesbcs/rpieasy #158: New device: PZEM004

enesbcs, 2020-06-17:


enesbcs, 2020-06-18:
PZEM-004T-10A ordered from Ali.

enesbcs, 2020-07-18:
Caching method added for PZEM004 type to P206 at commit https://github.com/enesbcs/rpieasy/commit/9d1aaddacc090eb79671bb60428a760cb7e685b7

Epik77, 2020-07-20:
Hi Alexander,

I confirm the correct functioning of the pzem-004
I wanted to reset the counter but the pzemreset command doesn't work
08:01:07: Unknown command: pzemreset, 1
am I wrong?
thank you very much for the support

enesbcs, 2020-07-20:
Interesting, it looks like the serial port needs to be closed after every transaction to work stable. 
The _performCommand(66,'') alias "resetenergy" function fails to run otherwise. / I feel some Star Wars hint here :) /
Fixed in commit https://github.com/enesbcs/rpieasy/commit/0335dbca4773ef83311c74f64ca51a1f9811adeb

TD-er, 2020-07-20:
Maybe you can also force to flush the buffers of the serial port?
Or does it need some specific logic pin state which you also set by closing the port?

enesbcs, 2020-07-21:
Honestly it is the minimalmodbus library which handles the serial at all, like that:

minimalmodbus.Instrument(self.port,self.address,close_port_after_each_call=True)

In theory there are a "clear_buffers_before_each_transaction" attribute which may be used alternately but "close_port_after_each_call" works fine when i had tested it.

enesbcs, 2020-08-01:
Closed as fixed

closed 2020-08-01 by enesbcs

Epik77, 2021-07-30:
Hello
I re-open this issue after one year because after the python modbus upgrade the device does not work correctly.
The pzemreset command no longer works:
Unknown command: pzemreset, 1
The serial is not read correctly rpieasy reports the last value read "correctly" on the display and on the controller.

thanks for the support.