enesbcs/rpieasy #159: RPi 4 Unknown GPIO

avassdal, 2020-07-06:
Hello

I don't know if this is a known issue or not, but in RPI Easy i get this error message when i try to access Pinouts&Ports on RPi 4: 

**`This hardware has unknown GPIO.`**

enesbcs, 2020-07-06:
Your RPI4 model is not in the RPIEasy database, than sure
Please write your 'cat /proc/cpuinfo' (Linux console) command output.

avassdal, 2020-07-06:
Here you are:

`pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03111
Serial		: 10000000fe973a8e
Model		: Raspberry Pi 4 Model B Rev 1.1`

enesbcs, 2020-07-06:
BCM2711 support added at commit https://github.com/enesbcs/rpieasy/commit/3bdb5d4df40c8411977347f23f123bbc7b46ec81

avassdal, 2020-07-07:
Thank you! Will test this later today.

enesbcs, 2020-07-26:
Closed as no feedback provided

closed 2020-07-26 by enesbcs