enesbcs/rpieasy #160: Not listening to serial commands.

avassdal, 2020-07-06:
Hello

I'm having issues with RPI Easy not listening to serial commands from USB Serial. I can check that the commands are actually being received if i screen the serial port, but nothing nothing seems to be happening in the logs.

enesbcs, 2020-07-06:
You are right. Neither serial logging, nor serial command interpreter was implemented in RPIEasy since the first version at 2018, as i do not need it.

As it requires several steps from the user to init serial, and installation of requirement libraries, (and also needs cpu power to poll serial port in the background continously) it will never be contained in the core, it can be at most an external plugin.
If you really need it, please open a new issue for a serial command interpreter plugin.

avassdal, 2020-07-06:
I was planning on doing this to get values from an Arduino Leonardo via USB Serial. What is the best way to do this then? i2c?

enesbcs, 2020-07-06:
1. There are an existing P201 plugin for simple data receiving and sending. Received data can be postprocessed by Rules at RPIEasy side.

2. P011 ProMini plugin can be used for simple I2C communication. Sample Slave sketch can be found here:
https://github.com/enesbcs/ESPEasySlaves/blob/master/MiniProExtender/MiniProExtender.ino

enesbcs, 2020-08-01:
Closed as no further feedbacks received

closed 2020-08-01 by enesbcs