enesbcs/rpieasy #162: TM1637 4 Digit display does not stay off if blinking clock setting used.

Budman1758, 2020-07-21:
As stated in the title the the display does not stay off if the clock with blinking colon setting is used.  Using the "7doff" command turns the display off but it comes back on within a second.  It works correctly if the "non blinking" clock settings are used.

I would assume this is a bug.

And as I write this.... The non blinking clock display setting also does not stay off . It comes back on when the minuet updates. In other words it turns back on every 60 seconds.

I would assume this is also a bug.  :)

enesbcs, 2020-07-21:
You are right off is just setted the brightness to 0 until now, and the loop wrote values regardless of the display state. From commit https://github.com/enesbcs/rpieasy/commit/8491dfabc6a72a3c756dff3994e6ece3fb93ecae the background refreshing of the display stopped if 7doff executed.

Budman1758, 2020-07-21:
That seems to have fixed it!!  Good work!

Just for the hell of it I also tried with the plugin set to "manual" .  When the display is off it does turn on when sending a command to display a number or other value.  I guess its open for debate as to weather this is a bug or not.  I guess I could see both sides of the argument.  

One one hand if manually writing to it I could imagine you would want it to display whatever you are writing to it.

On the other hand I could see you wanting to keep it off and only back on when wanted by sending the on command. 

Maybe an option to chose?

Thoughts?

enesbcs, 2020-07-22:
Thanks!
Its a very good question, as if you are writing directly to the screen, you may want it to display..

Maybe for compatibility it should work similarly like in ESPEasy? I checked 7DGT source code and as i see the write commands executed whether display is enabled or not. Although ESPEasy controls display low level which may enable writing with 0 brightness vs RPIEasy uses a higher level library which may changes automatically brightness upon writing to display.

I can implement a checkbox if needed to control this behaviour. Which behaviour is needed in your use case now?

Budman1758, 2020-07-22:
> Which behaviour is needed in your use case now?

To be honest it really doesn't matter to much to me.  I just thought would be good to see what others thought.  If I had to choose I would probable want the option to either leave it off or turn it on when manually writing to it. But in my case at the moment it's not a critical thing. I am just toying with several options for a carputer I'm building for my 64 Chevy truck project.

enesbcs, 2020-09-21:
Closed as no more feedbacks received.

closed 2020-09-21 by enesbcs