enesbcs/rpieasy #165: optimize _P517_BLELYWSD03 plugin (battery)

michapr, 2020-09-03:
The provided battery level from related UUID is not the best. It will show full or empty only.

Better would be to use the given battery voltage value and create an own level.
Using the following procedure (same as in OMQTTGW and other sources):

```
in def handleNotification(self, cHandle, data):

     temp = float(received[1] * 256 + received[0]) / 100
     hum = received[2]
     bat1= float(received[4] * 256 + received[3]) / 1000
     misc.addLog(rpieGlobals.LOG_LEVEL_DEBUG,"voltage: "+str(bat1))
     misc.addLog(rpieGlobals.LOG_LEVEL_DEBUG,"bat: " + str(((3.1-bat1)*100)))
```

result in plugin code (first 2 lines):

```
Sep  3 17:18:13 RPIEasy RPIEasy98: voltage: 2.438
Sep  3 17:18:13 RPIEasy RPIEasy98: bat: 66.19999999999999
Sep  3 17:18:13 RPIEasy RPIEasy98: Event: LYWSD03-1#Temperature=21.9
Sep  3 17:18:13 RPIEasy RPIEasy98: Event: LYWSD03-1#Humidity=60.0
Sep  3 17:18:13 RPIEasy RPIEasy98: Event: LYWSD03-1#Battery=100.0
```

Maybe you could include it in plugin? ;)

Thanks!
Michael

enesbcs, 2020-09-03:
Yes, of course. In the meantime I've modified BLE plugins a bit to not interfere each other when BLE is occupied. LYWSD03 is the one which holds the line over the others according my tests.

enesbcs, 2020-09-03:
LYWSD03 battery handling modified at commit bahttps://github.com/enesbcs/rpieasy/commit/2d7d900686c6bac56b14681d2e831aa8742c0812
Please backup config before updating

michapr, 2020-09-04:
Battery reporting is working fine, thanks!

closed 2020-09-04 by michapr