enesbcs/rpieasy #166: minor issue - if backupfile contain"("

michapr, 2020-09-04:
if created backup file contain a "(" like data(5).zip it cannot be imported:
upload JSON seems not to work more: 
```
Creating new pinout
Load devices from file
Load controllers from file
Load notifiers from file
Loading rules...
10:27:40: Event: System#Boot
10:27:40: Webserver starting at port 80
10:27:40: Event: Clock#Time=Fri,10:27
10:27:40: Event: Network#Connected
10:28:00: Event: Clock#Time=Fri,10:28
/bin/sh: 1: Syntax error: "(" unexpected
10:29:01: Event: Clock#Time=Fri,10:29
```

Not critical, but good to know ;)

enesbcs, 2020-09-04:
Your data(5).zip contains invalid json files otherwise.

enesbcs, 2020-09-04:
I see, '' will be added around file name in next commit

enesbcs, 2020-09-04:
Addressed at commit https://github.com/enesbcs/rpieasy/commit/b7fa8d5757a578db087a5561be8797767e05ee4b

closed 2020-09-08 by michapr