enesbcs/rpieasy #167: BLE connection issues after updates

teepox, 2020-09-10:
I updated my raspberry-pi along with rpieasy yesterday. Went from 2.0.1xx to 2.0.248, and on the OS side from something available in late July to current raspbian. After this, the Xiaomi Mijia meters worked ok for a few hours, but then stopped. Nothing in the log, which makes troubleshooting a bit hard. Currently I can get the meters working ok for a few minutes, but then the log just shows clock-events. Seems like something happens on BT side which makes the system to just give up. Previously I saw gaps in data and lack of retries for a while after several BLE connection failures, but now those do not happen. The gap last night was about 8 hours, so it doesn't seem to be recovering on it's own like it has in the past.

enesbcs, 2020-09-10:
From latest updates if a BLE plugin fails to free BLE semaphor, the other sensors will never get  free BLE again until RPIEasy restart.
Please attach debug logs.

teepox, 2020-09-10:
Yeah, it's saying:
22:00:47 BLE line not free for P512! [4]
22:00:47 BLE line not free for P512! [4]
22:00:47 BLE line not free for P512! [4]

enesbcs, 2020-09-10:
> 22:00:47 BLE line not free for P512! [4]

What is your Task5 device type? What is the last five line above BLE line not free?

teepox, 2020-09-11:
I have 6 tasks configured, all of them Xiaomi Mijia Temperature&Humidity sensors with 2 minute interval. The sensors can be slow to respond, so under normal operation I get the "BLE line not free for P512! [X]" messages from time to time for all sensors. It's hard to catch the last lines before the hang, as the "BLE line not free" repeats several times per second. Thus the log just scrolls so fast the history is no longer there when I notice it. Should probably store it in a syslog or something... Considering the config, I wouldn't expect to find anything drastic, though.

enesbcs, 2020-09-11:
I see.
If intervals differs with at least 10 seconds to each other, it will help a lot to avoid race for the BLE. Anyway, the global semaphor should help, now i've implemented a 5 minute timeout for holding the line (failsafe).
https://github.com/enesbcs/rpieasy/commit/b98bc9a4fd9aa3b99979034145c56bbe6847fc46

teepox, 2020-09-11:
Ok, updated. Let's see how it runs.
It seems that under normal operation the timestamps for data are spread out quite evenly. Errors seem to happen only when a meter is slow to connect for some reason. To help with pacing, the interval counter could start only after the connection is terminated, but it feels like it's already doing that, or is there a queue system in place?

michapr, 2020-09-11:
> I have 6 tasks configured, all of them Xiaomi Mijia Temperature&Humidity sensors with 2 minute interval.

I think it is really too short... (if LYWSD03 ?)
I have same sensors (3 pc.) - all of them were connected every 5 minutes. After 9-10 weeks I had to change the battery.
It seems that the connection time cost many battery.
Have reconfigured after battery change to 10 minutes and will see now what happen.

>Errors seem to happen only when a meter is slow to connect for some reason.

What I have seen is that sensor will be connected, but the UUID "will not be found" - so the connection seems not to be stable.
In such a case the BLE adapter can hang in result.

Anyway I think in result of current research that the best would be to use a sensor without needed connection, but use the advertising data from sensors. It will save battery and is very safe.

enesbcs, 2020-09-11:
> To help with pacing, the interval counter could start only after the connection is terminated, but it feels like it's already doing that, or is there a queue system in place?

Interval is global in RPIEasy for all devices.
BLE queue is implemented in recent versions to avoid crosstalk.

teepox, 2020-09-11:
Yeah, listening for BLE beacons would probably be the best. Sensors doing that (e.g. Ruuvi Tag) are a bit on the pricy side, though.

My sensors are LYWSDCGQ, which seem to have good battery life even with frequent polling. My oldest sensor has been running since 21st of May, so more than three months, and it's at 79% now (assuming the numbers are accurate).

enesbcs, 2020-09-11:
> Yeah, listening for BLE beacons would probably be the best. Sensors doing that (e.g. Ruuvi Tag) are a bit on the pricy side, though.
>
> My sensors are LYWSDCGQ, which seem to have good battery life [...]

LYWSD02, LYWSDCGQ, LYWSDCGG1 and MiFlora are supported by P527 Passive sniffer.

michapr, 2020-09-12:
> My sensors are LYWSDCGQ

I really would suggest to use the new P527 sniffer plugin - here never will be any device specific BLE connection problem.
While sensor is advertising all data will be received.

michapr, 2020-09-13:
I have now ater some time also again such a "hanging" issue:
```
Sep 13 12:10:35 RPIEasy RPIEasy98: BLE read error: Failed to connect to peripheral 3f:5b:7d:71:7c:38, addr type: public
Sep 13 12:10:38 RPIEasy RPIEasy98: BLE connected 3f:5b:7d:71:7c:38
Sep 13 12:10:41 RPIEasy RPIEasy98: BLE line not free for P517! [5]
Sep 13 12:10:41 RPIEasy RPIEasy98: BLE line not free for P515! [5]
Sep 13 12:10:41 RPIEasy RPIEasy98: BLE line not free for P517! [5]
Sep 13 12:10:41 RPIEasy RPIEasy98: BLE line not free for P515! [5]
Sep 13 12:10:42 RPIEasy RPIEasy98: BLE line not free for P517! [5]
Sep 13 12:10:42 RPIEasy RPIEasy98: BLE line not free for P515! [5]
Sep 13 12:10:42 RPIEasy RPIEasy98: BLE line not free for P517! []
Sep 13 12:10:42 RPIEasy RPIEasy98: BLE connection initiated to a4:c1:38:c1:8c:71
Sep 13 12:10:42 RPIEasy RPIEasy98: BLE read error: Failed to connect to peripheral 3f:5b:7d:71:7c:38, addr type: public
Sep 13 12:10:42 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:43 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:43 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:44 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:44 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:45 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:45 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:10:46 RPIEasy RPIEasy98: BLE line not free for P513! [6]
Sep 13 12:10:46 RPIEasy RPIEasy98: BLE line not free for P515! [6]
...
Sep 13 12:11:22 RPIEasy RPIEasy98: BLE line not free for P513! [6]
Sep 13 12:11:22 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:11:23 RPIEasy RPIEasy98: BLE line not free for P513! [6]
Sep 13 12:11:23 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:11:23 RPIEasy RPIEasy98: BLE line not free for P513! [6]
Sep 13 12:11:23 RPIEasy RPIEasy98: BLE line not free for P515! [6]
Sep 13 12:11:24 RPIEasy RPIEasy98: BLE connection failed a4:c1:38:c1:8c:71
Sep 13 12:11:24 RPIEasy RPIEasy98: BLE line not free for P513! []
Sep 13 12:11:24 RPIEasy RPIEasy98: BLE connected 3f:5b:7d:71:7c:38
Sep 13 12:11:24 RPIEasy RPIEasy98: BLE line not free for P515! [5]
Sep 13 12:11:24 RPIEasy RPIEasy98: BLE line not free for P515! [5]
...
Sep 13 16:27:04 RPIEasy RPIEasy98: BLE line not free for P515! [5]
```
After restarting application all was ok again.

task 6- LYWSD02 Clock&Hygrometer (:38)
task 7- LYWSD03/MHOC401  (:71)

enesbcs, 2020-09-13:
Did you upgraded to v2.0.255? There are a 5 minute timeout in this version.

teepox, 2020-09-13:
I'll test the P527 at some point, but for now the timeout is working fine. 2,5 days without issues.

michapr, 2020-09-13:
> Did you upgraded to v2.0.255?

No, sorry... forgot about this.
(never touch a running system ;) ....)
Will do and tell you.

Epik77, 2020-09-15:
Sorry,
 after last update  the process bluepy-helper  is freezed

root      7377  0.0  0.4   3728  2192 ?        SN   02:52   0:00 /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper 0

also blocking the processes of other devices :(

killing the process 7377 restores correct operation.

and new processes are performed 

root     29185  0.0  0.4   3756  2396 ?        SN   Sep14   0:00 /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper 0
root     29386  0.0  0.4   3756  2420 ?        SN   Sep14   0:00 /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper 0

thank's for you support :-)

michapr, 2020-09-15:
Can you provide some debug log? Would be helpful ;)

Epik77, 2020-09-15:
sorry i don't have a debug
the trial remained open from 2:52 AM this night and easyrpi remained to wait.
If it happens again I will make a strace of the process.

michapr, 2020-09-15:
I thought about the syslog configured under tools -> advanced. ;)

Epik77, 2020-09-15:
I don't have a syslogserver and the device is mounted read-only
as soon as it comes back I try to provide you with more information

enesbcs, 2020-09-15:
> Sorry,
> after last update the process bluepy-helper is freezed

Do you use only P512 BLE plugins or other BLE related plugins are also used on the same RPI?

Did you updated bluepy module with Tools->System Updates->PIP update? Maybe you are using an old version of that.

michapr, 2020-09-17:
Same problem again - seems with LYWSD02, but not sure.
Have attached the log, maybe you can see here anything helpful.
[rpieasy.log.gz](https://github.com/enesbcs/rpieasy/files/5237016/rpieasy.log.gz)

The one LYWSD03 have a permanent BLE connection after this, I see it on display (BLE symbol)
In this time a restart of application didn't help - no more BLE connection was possible.. I had to reboot.

Thanks!
Michael

enesbcs, 2020-09-17:
> The one LYWSD03 have a permanent BLE connection after this, I see it on display (BLE symbol)

I guess than your LYWSD03 freezed the BLE stack in the connection phase, it is handled by bluepy, i can not help with it.

> In this time a restart of application didn't help - no more BLE connection was possible.. I had to reboot.

Yep, if restart of RPIEasy did not help, than possibly the bluepy helper was stucked. Its process can be killed or the reboot help as you say.

Do you have a spare ESP32 for testing?

michapr, 2020-09-17:
>possibly the bluepy helper was stucked

I think so too.

> Do you have a spare ESP32 for testing?

Yes, but what to test? ;)

enesbcs, 2020-09-17:
I am still working on micropython code for ESP32. Although the LYWSD03 is constantly killing the ESP32 NimBLE implementation, not just the RPI bluepy stack.. :D

```
21:25:27: Event: ly3#Temperature=27.5
21:25:27: Event: ly3#Humidity=48.0
assertion "rc == 0" failed: file "/home/micropython/esp-idf-v3/components/nimble/nimble/nimble/host/src/ble_hs.c", line 576, function: ble_hs_event_start_stage2
abort() was called at PC 0x40176d73 on core 1
ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
Backtrace: 0x4009434b:0x3ffff640 0x400946a1:0x3ffff660 0x40176d73:0x3ffff680 0x40167747:0x3ffff6b0 0x4016e79e:0x3ffff6d0 0x400fa19a:0x3ffff6f0 0x4009571e:0x3ffff710
Rebooting...
```

Epik77, 2020-09-18:
Hi,
I have update python  and  all pip packages  but the problem occurs
root     22375  0.0  0.4   3728  2276 ?        SN   06:20   0:00 /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper 0
root     28474  0.0  0.4   3756  2204 ?        SN   Sep15   0:00 /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper 0
root     28601  0.0  0.4   3756  2204 ?        SN   Sep15   0:00 /usr/local/lib/python3.7/dist-packages/bluepy/bluepy-helper 0

#strace -p 22375
strace: Process 22375 attached
write(1, "rsp=$scan\36addr=b0284AA6FC5F3\36typ"..., 127

06:19:44 Event: Energia#Watt=286.7
06:19:44 Event: Pzem#Volt=234.1
06:19:44 Event: Pzem#Amper=1.4
06:19:44 Event: Pzem#Watt=287.1
06:19:44 Event: Pzem#Wh=1153149.0
06:19:44 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:19:48 Event: Ampere#Amper=1.4
06:19:48 Event: Energia#Watt=287.1
06:19:48 Event: Pzem#Volt=234.1
06:19:48 Event: Pzem#Amper=1.4
06:19:48 Event: Pzem#Watt=287.6
06:19:48 Event: Pzem#Wh=1153149.0
06:19:48 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:19:52 Event: Ampere#Amper=1.4
06:19:52 Event: Energia#Watt=287.6
06:19:52 Event: Pzem#Volt=234.0
06:19:52 Event: Pzem#Amper=1.4
06:19:52 Event: Pzem#Watt=287.4
06:19:52 Event: Pzem#Wh=1153149.0
06:19:53 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:19:56 Event: Ampere#Amper=1.4
06:19:56 Event: Energia#Watt=287.4
06:19:56 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:20:00 Event: Pzem#Volt=234.0
06:20:00 Event: Pzem#Amper=1.4
06:20:00 Event: Pzem#Watt=289.8
06:20:00 Event: Pzem#Wh=1153150.0
06:20:00 Event: Clock#Time=Fri,06:20
06:20:01 Event: Ampere#Amper=1.4
06:20:01 Event: Energia#Watt=289.8
06:20:01 Event: PresenzaMary#Status=0
06:20:02 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:20:03 Event: PresenzaLorenzo#Status=0
06:20:07 Event: Pzem#Volt=236.8
06:20:10 Event: Pzem#Amper=1.4
06:20:11 Event: Pzem#Watt=290.6
06:20:11 Event: Pzem#Wh=1153151.0
06:20:12 Event: Ampere#Amper=1.4
06:20:12 Event: Energia#Watt=290.6
06:20:12 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:20:18 Event: IphoneMaryPing#State=0
06:20:19 Event: Ampere#Amper=1.4
06:20:19 Event: Energia#Watt=290.6
06:20:20 Event: Pzem#Volt=237.0
06:20:20 Event: Pzem#Amper=1.4
06:20:20 Event: Pzem#Watt=290.4
06:20:20 Event: Pzem#Wh=1153152.0
06:20:20 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:20:23 Event: Ampere#Amper=1.4
06:20:23 Event: Energia#Watt=290.4
06:20:23 Event: PresenzaLorenzo#Status=1
06:20:24 Event: Pzem#Volt=236.8
06:20:24 Event: Pzem#Amper=1.4
06:20:24 Event: Pzem#Watt=290.0
06:20:24 Event: Pzem#Wh=1153152.0
06:20:24 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:21:25 Event: Clock#Time=Fri,06:21
06:21:26 Event: Ampere#Amper=1.4
06:21:27 Event: Temperatura_Bagno#Temperature=24.2
06:21:27 Event: Temperatura_Bagno#Humidity=75.0
06:21:27 Event: Energia#Watt=290.0
06:21:27 Event: Consumo#Wh=1153152.0
06:21:27 Event: Voltaggio#Volt=236.8
06:21:27 Event: TemperaturaDevil03#Temp=40.6
06:21:27 Event: Pzem#Volt=236.7
06:21:27 Event: Pzem#Amper=1.4
06:21:27 Event: Pzem#Watt=288.9
06:21:27 Event: Pzem#Wh=1153157.0
06:21:28 LCD write error! Cursor position (1, 16) invalid on a 2x16 LCD.
06:21:28 Event: CPU#Load=47.1

Epik77, 2020-09-18:
sorry
i used this workaround to fix the problem
killall --older-than 1h bluepy-helper
in the crontab

enesbcs, 2020-09-19:
I exprienced similar thing after 1 hour, I had to remove my BLE stick from USB and reconnect to be able to connect any BLE device. :(

enesbcs, 2020-09-27:
> Yes, but what to test? ;)

@michapr 
https://drive.google.com/file/d/1rCrc3NJsRqt55J-empC-nAvvIoH8UR_d/view?fbclid=IwAR1Mzep37tUD5Fajm80lud7cqMlMl1g3vJEyyQGluM95Zg7F0HMrP56DnmI

michapr, 2020-09-30:
@enesbcs 
Have no free ESP32 with PSRAM available, will take some days...

enesbcs, 2020-10-02:
Honestly it may run without PSRAM, but on some occasions it throws buffer overflow error on wifi/bt reconnect..

cyraxxx, 2020-10-02:
Hi, first of all, thank you for your work, it is a very nice software :)

I just installed RPIEasy for first time so cannot tell if my issue is related to update or not. But definitely connects with BLE so here I am. Tried to add 4 sensors (LYWSD02) to a RPI4 using the integrated BT (connecting to router by cable, not wifi). After the first one, I had to disable that for adding the second and restart RPIEasy. Now I added all of them like this (disabling the others), but  now after restart, it connects to the first one and ignores the others. My log is full of messages like this:

20:16:20: BLE line not free for P513! [0]
20:16:20: BLE line not free for P513! [0]
20:16:20: BLE line not free for P513! [0]
20:16:21: BLE line not free for P513! [0]
20:16:21: BLE line not free for P513! [0]
20:16:21: BLE line not free for P513! [0]
20:16:21: BLE line not free for P513! [0]
20:16:21: BLE line not free for P513! [0]
20:16:21: BLE line not free for P513! [0]
20:16:22: BLE line not free for P513! [0]
20:16:22: BLE line not free for P513! [0]
20:16:22: BLE line not free for P513! [0]
20:16:22: BLE line not free for P513! [0]
20:16:22: BLE line not free for P513! [0]

It looks like the same problem like @michapr. But as I read you added a 5 minutes timeout, it is working:

20:12:46 Pi 4 Model B 40 pins
20:12:47 Event: System#Boot
20:12:47 Webserver starting at port 8008
20:12:47 BLE connected e7:2e:01:42:----
20:12:48 Sync LWSD02 time
20:17:47 BLE connected e7:2e:01:71:----
20:17:48 Sync LWSD02 time

So maybe isn't is possible to add a variable which can manipulate the time of variable somehow? I am happily here to test :D

enesbcs, 2020-10-03:
300 seconds is a fail safe for BLE devices which stucks.. it is not a good idea to modify it globally. Maybe it will be a new setting for the LYWSD02 plugin in the next version.

michapr, 2020-10-03:
>https://drive.google.com/file/d/1rCrc3NJsRqt55J-empC-nAvvIoH8UR_d/view?fbclid=IwAR1Mzep37tUD5Fajm80lud7cqMlMl1g3vJEyyQGluM95Zg7F0HMrP56DnmI

@enesbcs A really mixed issue here - maybe open a new one for this? :)
Testing with ESP with PSRAM - scanning plugin is working fine.
What do you think - would it be possible to pause in future the scanning plugin while other BLE device could be read?
A generic MQTT would be nice while testing ;) (MQTT is not working - I think you know)

Thanks!
Michael

enesbcs, 2020-10-05:
> @enesbcs A really mixed issue here - maybe open a new one for this? :)
> Testing with ESP with PSRAM - scanning plugin is working fine.
> What do you think - would it be possible to pause in future the scanning plugin while other BLE device could be read?

There are a random gap between two scans. As scanning is a blocking function, which is done by the micropython nimble stack, i can not stop that. The same semaphore system works in mpyEasy which i am using at RPIEasy.

> A generic MQTT would be nice while testing ;) (MQTT is not working - I think you know)

Domoticz MQTT worked when i last checked. As i am a Domoticz user, Domoticz always comes first. :)

enesbcs, 2020-10-05:
At least BLE devices can coexist now:
[image]

enesbcs, 2020-10-05:
mpyeasy-alpha-2
https://drive.google.com/file/d/1VvOExB7aJIa99GnCKBt1hR7JnW17MjG6/view?usp=sharing

michapr, 2020-10-07:
Scanner is working fine, MQTT is now working also (there was an error before any missing definition).
Thanks for global MQTT ;)
BTW: would be great to have the JSON publishing option also for generic MQTT even for RPIEasy  ;)

enesbcs, 2020-10-08:
> Scanner is working fine, MQTT is now working also (there was an error before any missing definition).

Yes, umqtt.simple was missing from the compiled binary. I thought it is included in micropython as default, now added manually.

> BTW: would be great to have the JSON publishing option also for generic MQTT even for RPIEasy ;)

Please add some examples to the issue how it has to look like https://github.com/enesbcs/rpieasy/issues/175

michapr, 2020-10-08:
mypyeasy: I have tried FlowerCare ("MiFlora-R") read together with scanning module - and it is working without problems (in this time).
I see no error messages in log.

```
11:23:40.948 -> 09:23:40: Event: FlowerCare1#Temp=22.2
11:23:41.041 -> 09:23:40: Event: FlowerCare1#Moist=61.0
11:23:41.186 -> 09:23:40: Event: FlowerCare1#Fert=2262.0
11:23:41.277 -> 09:23:40: Event: FlowerCare1#Light=395.0
11:23:42.689 -> {'temp-t': 655464125, 'moist': 45, 'temp': 22.3, 'moist-t': 655464188, 'fertil-t': 655464034, 'fertil': 599, 'light': 2336, 'light-t': 655463974}
11:23:42.781 -> 09:23:42: Event: FlowerCare2#Temp=22.3
11:23:42.878 -> 09:23:42: Event: FlowerCare2#Moist=45.0
11:23:42.975 -> 09:23:42: Event: FlowerCare2#Fert=599.0
11:23:43.110 -> 09:23:42: Event: FlowerCare2#Light=2336.0
11:23:43.444 -> {'hum-t': 655464188, 'hum': 48.6, 'batt': 84, 'batt-t': 655464080, 'temp-t': 655463924, 'temp': 22.9}
11:23:43.532 -> 09:23:43: Event: ClearGrassTemp&RH-S#Temp=22.9
11:23:43.674 -> 09:23:43: Event: ClearGrassTemp&RH-S#Hum=48.6
11:23:43.770 -> 09:23:43: Event: ClearGrassTemp&RH-S#Batt=84.0
11:23:43.968 -> {'temp-t': 655464184, 'temp': 23.2, 'hum': 54.0, 'hum-t': 655464189}
11:23:44.060 -> 09:23:43: Event: LYWSD02-S#Temp=23.2
11:23:44.201 -> 09:23:43: Event: LYWSD02-S#Hum=54.0scan start
11:23:44.201 -> 10 sec
11:23:44.201 -> 
11:23:54.198 -> scan pause
11:24:00.586 -> 09:24:00: Event: Clock#Time=Thu,09:24
11:24:02.549 -> dl0 {'t': 4098240, 'f': 4048272, 'u': 49968}
11:24:02.934 -> dl1 {'t': 4098240, 'f': 4048272, 'u': 49968}
11:24:03.671 -> dl2 {'t': 4098240, 'f': 4047232, 'u': 51008}
11:24:03.955 -> d3 {'t': 4098240, 'f': 4047280, 'u': 50960}
11:24:28.257 -> dl0 {'t': 4098240, 'f': 4048208, 'u': 50032}
11:24:28.630 -> dl1 {'t': 4098240, 'f': 4048288, 'u': 49952}
11:24:29.377 -> dl2 {'t': 4098240, 'f': 4047248, 'u': 50992}
11:24:29.609 -> d3 {'t': 4098240, 'f': 4047296, 'u': 50944}
11:24:43.659 -> 09:24:43: Event: MiFlora-R#Temperature=22.2
11:24:43.753 -> 09:24:43: Event: MiFlora-R#Moisture=63.0
11:24:43.892 -> 09:24:43: Event: MiFlora-R#Conductivity=2253.0
11:24:44.031 -> 09:24:43: Event: MiFlora-R#Battery=94.0
11:24:44.454 -> scan start
11:24:44.454 -> 12 sec
11:24:46.936 -> 09:24:46: Event: esp-system#Uptime=507.0
11:24:47.497 -> 09:24:47: Event: esp-system#RAM=4049167.6
11:24:47.684 -> 09:24:47: Event: esp-system#wifirssi=-40.0
11:24:47.778 -> 09:24:47: Event: esp-system#CPU-Temp=46.7
11:24:49.081 -> {'temp-t': 655464229, 'moist': 61, 'light-t': 655464122, 'light': 395, 'moist-t': 655464287, 'temp': 22.3, 'fertil': 2262, 'fertil-t': 655464181}
11:24:49.348 -> 09:24:48: Event: FlowerCare1#Temp=22.3
...
```

michapr, 2020-10-08:
Just have seen a reboot (before inserting the MiFlora element - so this may not be the reason) - only scanner was active with 4 devices + esp-system.
```
11:16:03.762 -> 09:16:02: Event: Clock#Time=Thu,09:16
11:16:12.518 -> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
11:16:12.518 -> Core 0 register dump:
11:16:12.518 -> PC      : 0x400f37c9  PS      : 0x00060d34  A0      : 0x800f390e  A1      : 0x3ffdcc60  
11:16:12.518 -> A2      : 0x3f81bb40  A3      : 0x3f817b60  A4      : 0x00000002  A5      : 0x00267da2  
11:16:12.518 -> A6      : 0x3f81bb18  A7      : 0x3fa826a0  A8      : 0x800f37c4  A9      : 0x3ffdcc40  
11:16:12.518 -> A10     : 0xffffffff  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
11:16:12.518 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
11:16:12.518 -> EXCVADDR: 0x00000000  LBEG    : 0x40098aac  LEND    : 0x40098ab7  LCOUNT  : 0x00000000  
11:16:12.518 -> 
11:16:12.518 -> ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
11:16:12.518 -> 
11:16:12.518 -> Backtrace: 0x400f37c9:0x3ffdcc60 0x400f390b:0x3ffdcca0 0x400e3ec5:0x3ffdccd0 0x400dfcb9:0x3ffdccf0 0x400dfdb5:0x3ffdcd10 0x400e0a48:0x3ffdcd40 0x400e0b4d:0x3ffdcdc0 0x400ee244:0x3ffdcde0 0x400e4058:0x3ffdce80 0x400dfcb9:0x3ffdceb0 0x400dfdf5:0x3ffdced0 0x400eddef:0x3ffdcef0 0x400e4058:0x3ffdcf90 0x400dfcb9:0x3ffdcfc0 0x400e2520:0x3ffdcfe0 0x400e253d:0x3ffdd030 0x400dfcb9:0x3ffdd050 0x400ece24:0x3ffdd070 0x400fbc91:0x3ffdd100 0x40093852:0x3ffdd120
11:16:12.518 -> 
11:16:12.518 -> Core 1 register dump:
11:16:12.518 -> PC      : 0x401d192e  PS      : 0x00060934  A0      : 0x80117e72  A1      : 0x3ffbc430  
11:16:12.518 -> A2      : 0x3ffc2d8d  A3      : 0x00060920  A4      : 0x80093a4b  A5      : 0x3ffbc360  
11:16:12.518 -> A6      : 0x00000003  A7      : 0x00060023  A8      : 0x80118964  A9      : 0x3ffbc420  
11:16:12.518 -> A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x3ffbb1e0  
11:16:12.518 -> A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
11:16:12.518 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
11:16:12.518 -> 
11:16:12.518 -> ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
11:16:12.518 -> 
11:16:12.518 -> Backtrace: 0x401d192e:0x3ffbc430 0x40117e6f:0x3ffbc450 0x40094c25:0x3ffbc470 0x40093852:0x3ffbc4a0
11:16:12.518 -> 
11:16:12.518 -> Rebooting...
```

enesbcs, 2020-10-08:
> Core  0 panic'ed (Interrupt wdt timeout on CPU0)

It simply means that the watchdog works and rebooted the non-responsive cpu.

> EDIT: Maybe the new scan start is coming up too fast (old scan is not finished?)

It is not possible, as there are only one background scan process existing as in RPIEasy.

michapr, 2020-10-10:
I have some such reboots in log - it isn't normal, right?

```
22:19:17.015 -> scan start
22:19:17.015 -> 16 sec
22:19:30.075 -> abort() was called at PC 0x4008718f on core 0
22:19:30.075 -> 
22:19:30.075 -> ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000
22:19:30.075 -> 
22:19:30.075 -> Backtrace: 0x40090fef:0x3ffdc6a0 0x40091345:0x3ffdc6c0 0x4008718f:0x3ffdc6e0 0x400872a5:0x3ffdc720 0x4017b75d:0x3ffdc740 0x401769f5:0x3ffdca00 0x4017697d:0x3ffdca50 0x400f3743:0x3ffdca80 0x400f390b:0x3ffdcac0 0x400e3ec5:0x3ffdcaf0 0x400dfcb9:0x3ffdcb10 0x400dfdb5:0x3ffdcb30 0x400e0a48:0x3ffdcb60 0x400e0b4d:0x3ffdcbe0 0x400ee244:0x3ffdcc00 0x400e4058:0x3ffdcca0 0x400dfcb9:0x3ffdccd0 0x400e25a1:0x3ffdccf0 0x400dfcb9:0x3ffdcd40 0x400edd69:0x3ffdcd60 0x400e4058:0x3ffdce00 0x400dfcb9:0x3ffdce30 0x400dfdf5:0x3ffdce50 0x400eddef:0x3ffdce70 0x400e4058:0x3ffdcf10 0x400dfcb9:0x3ffdcf40 0x400e2520:0x3ffdcf60 0x400e253d:0x3ffdcfb0 0x400dfcb9:0x3ffdcfd0 0x400ece24:0x3ffdcff0 0x400fbc91:0x3ffdd080 0x40093852:0x3ffdd0a0
22:19:30.075 -> 
22:19:30.075 -> Rebooting...
```

I have seen that the BLE scan is not working if the sniffer module is active.

>There are a random gap between two scans.

Can it be possible to activate here the requested BLE scan and maybe the request of other BLE devices?
I know that it seems to be possible in Arduino NimBLE implementation (OpenMQTT GW scan with detecting values and then connect to some devices)

Thanks!

enesbcs, 2020-10-10:
> I have some such reboots in log - it isn't normal, right?

I am not sure that the ESP32 core will be stable in the near future. I am waiting for it since 2 years.

> 22:19:17.015 -> scan start
> 22:19:17.015 -> 16 sec
> 22:19:30.075 -> abort() was called at PC 0x4008718f on core 0

However i've never seen abort() in passive scanning on my setup.

> I have seen that the BLE scan is not working if the sniffer module is active.

Yes, as multiple scanning can not work at the same time. It wont change.

> Can it be possible to activate here the requested BLE scan and maybe the request of other BLE devices?

?

> OpenMQTT GW scan with detecting values and then connect to some devices

EspEasy, RPIEasy and mpyEasy has a time controlled main loop. Every device has an interval after it will be asked for values, if the BLE line is free, it will start its operation, otherwise it will wait and pass the control to the next device. I have no intentions to rewrite the core.
The BLE sniffing thread runs at the background for x sec, than pause for y sec, in the pause section, other BLE devices may use the BLE.

TD-er, 2020-10-10:
> I am not sure that the ESP32 core will be stable in the near future. I am waiting for it since 2 years.

Not tested BLE on the ESP32, but the rest is quite stable lately.
I'm running nodes here with LAN and WiFi, which were up for over 25 days... but had a power outage this week (also lost Sonoff's running over 155 days :( )

enesbcs, 2020-10-10:
> Not tested BLE on the ESP32, but the rest is quite stable lately.

As ESP32 has BLE capabilities, maybe not too much demand that 4 years after it's release date, to be usable.
My first encounter with ESP32 and Arduino is a two line sketch to connect and disconnect a BLE device, and it is ended with buffer overflow error on disconnect (connect was sussecful). Disappointing.

Now i am playing with micropython, it seems that BLE works - at least if BLE devices are 10cm far from the ESP32 i have no connection problems, neither reboots for 1-2 hours while i am tested it. But michapr reports random reboots which has to be triggered by the nimble stack, and i have no clue why.

> I'm running nodes here with LAN and WiFi, which were up for over 25 days... 

You are maybe right, simple GPIO operations seems to work fine on the ESP32. Although i have one ESP32 which occasionally disconnects from Wifi and when it is trying to reconnect it triggers a reboot. :( Maybe an undervoltage problem causes the wifi disconnect, but on reconnect a stack overflow occurs.. hilarious. I am using latest ESPIDF v3 with micropython 1.13

michapr, 2020-10-10:
>and it is ended with buffer overflow error ...
Yes, I can confirm that. My first steps last year were the same. There are some issues in BLE lib of ESP, something you can patch, then it will be better.
I made a BLE MQTT Gateway for some devices far from the main gateway. It was working with watchdog and reboots, not critical for BLE gateway. I then switched NimBLE for Arduino - all was much better.
Now I switched to OpenMQTTGateway (with some add-ons from my side) - it make the same as I made before, with much more options.
And I must say - it is stable. I have in this time one reboot within 4 weeks on ESP32 TTGO (with LCD ouput for debugging).
Using Arduino with latest ESP32 library and NimBLE lib.

enesbcs, 2020-10-17:
mPyEasy code is now public, waiting for beta testers.
https://github.com/enesbcs/mpyeasy

enesbcs, 2020-12-30:
Closed as no further informations in the last 2 months.

closed 2020-12-30 by enesbcs