enesbcs/rpieasy #168: PWM command argument 3 sets 'duration' on ESPeasy but 'frequency' on RPIEasy

kylehase, 2020-09-13:
Per [ESP Easy documentation](https://www.letscontrolit.com/wiki/index.php/GPIO), the 3rd argument for PWM command sets duration to fade in/out.
`PWM,<GPIO>,<state>,<duration>`

The same argument on RPIeasy instead changes the PWM frequency. 
For example, running: `PWM,21,1,100`
results in 
```
{
"log": "BCM21 PWM 1% 100Hz",
"plugin": 1,
"pin": 21,
"mode": "pwm",
"state": 1
}
```

System Info:
Build | RPIEasy 2.0.248
Libraries | Python 3.7.3 (default, Jul 25 2020, 13:03:44)[GCC 8.3.0] Linux-5.4.51+-armv6l-with-debian-10.4
Type: | RPI-Linux-apt (supported)
OS: | linux 5.4.51+
OS full name: | Raspbian GNU/Linux 10 (buster)
CPU model: | ARM1176
CPU speed: | 1000.0000 Mhz
CPU count: | 1
CPU architecture: | armv6l
Hardware: | Pi Zero W 512MB

enesbcs, 2020-09-13:
Yes PWM is working that way from the beginning in RPIEasy, as the kernel has much more options for PWM frequencies. (at least for hardware pwm)

Interesting, i have never seen this duration parameter before in ESPEasy. Currently i have no idea how to implement it into RPIEasy code, as millisec precision timing is not very well on any SBC with normal Linux kernel. Maybe with separated threads (per pwm lines) it can be accomplished.

enesbcs, 2020-12-15:
PWM 3rd param as fade added at commit https://github.com/enesbcs/rpieasy/commit/3862088e8ba26ae2a8e0c3ab8882644f8a59f001

It has blocking nature as in ESPEasy.

enesbcs, 2020-12-30:
Closed as implemented.

closed 2020-12-30 by enesbcs