enesbcs/rpieasy #174: Modified BLE device LYWSD03MMC with customized FW

michapr, 2020-10-07:
Hi,

there is a customized firmware for the Temp/Hum sensor LYWSD03MMC  available.
Firmware can be updated via web using the BLE connection!

Here it is possible to get all data from advertising package without connect, what will save battery and work more stable in distance on limit.
https://github.com/atc1441/ATC_MiThermometer

I have tested and it is working fine.
OpenMQTTGateway have already added it - and I think it could be added here too in BLESniff.py ;)

Format is there described.

for the code this means - 
`if cdata[0] == 0x1A18  { ...`

Some sample data:
```
"servicedata":"a4c138c18c7100e831240a710c" -> "tem":23.2,"hum":49,"batt":36,"volt":2.673
"servicedata":"a4c138c18c7100e732240a7204" -> "tempc":23.1,"hum":50,"batt":36,"volt":2.674}
```

Thanks!
Michael

enesbcs, 2020-10-08:
It is very interesting. I guess LYWSD03 is still connectable in the standard way after upgrading?

michapr, 2020-10-08:
Yes, it is. But because of new ID it will not be recognized as the "original" - so there will not be an automatic connection again after flash new firmware.
And - there are no needs more to connect to the device, only for firmware update ;)

(BTW: original firmware is available, so you could go back if you like)

enesbcs, 2020-10-08:
I see now. Device is still connectable, but Service and Battery UUID's are all changed, so the only way to get data from it is the advertising packets.

michapr, 2020-10-08:
`so the only way to get data from it is the advertising packets.`

Yes and no...
UUID 0x181A give you at 0x2A1F the temperature and at 0x2A6F the humidity.
Battery level is available at 0x180F under 0x2A19

Anyway it is much better to get it from advertising package ;)

enesbcs, 2020-10-08:
ATC advertising packet support added at commit https://github.com/enesbcs/rpieasy/commit/8ab3511737ed1f390b061cb13bc1870e2a38b0d1

michapr, 2020-10-08:
Something seems to be wrong...

blescan:
```
    Device (new): a4:c1:38:c1:8c:71 (public), -60 dBm
        16b Service Data: <1a18a4c138c18c7100ec31270a87f8>
        Complete Local Name: 'ATC_C18C71'

```
```
Oct  8 21:20:47 RPIEasy RPIEasy98: A4:C1:38:C1:8C:71 {}
Oct  8 21:20:47 RPIEasy RPIEasy98: Event: ATC_C18C71#Temp=0.0
Oct  8 21:20:47 RPIEasy RPIEasy98: Event: ATC_C18C71#Hum=0.0
Oct  8 21:20:47 RPIEasy RPIEasy98: Event: ATC_C18C71#Batt=255.0
Oct  8 21:20:47 RPIEasy RPIEasy98: Event: ATC_C18C71#rssi=-1.0
Oct  8 21:21:00 RPIEasy RPIEasy98: Event: Clock#Time=Thu,21:21

```

enesbcs, 2020-10-09:
Yep too much changes in a row... fixed in commit https://github.com/enesbcs/rpieasy/commit/4bf7b15999023d21a8488fbac3c71181945d4777

```
7:36:52: a4:c1:38:c3:ef:b0 {'temp': 24.1, 'temp-t': 1602221806.6226397, 'hum': 66, 'hum-t': 1602221806.6226425, 'batt': 48, 'batt-t': 1602221806.622645}
07:36:52: Event: lyatc#temp=24.1
07:36:52: Event: lyatc#hum=66.0
07:36:52: Event: lyatc#batt=48.0
07:36:52: Event: lyatc#rssi=-64.0

```

michapr, 2020-10-09:
Hmm... still not working for me...
Have disabled all other devices and this one is first task now, but no changes.
```
10:00:54: A4:C1:38:C1:8C:71 {}
10:00:54: Event: ATC_C18C71-S#Temp=0.0
10:00:54: Event: ATC_C18C71-S#Hum=0.0
10:00:54: Event: ATC_C18C71-S#Batt=255.0
10:00:54: Event: ATC_C18C71-S#rssi=-100.0
```
Data will be received - I have checked debugging a bit...
```
09:58:54: AdvDecoder: a4:c1:38:c1:8c:71/-48/{'temp': 23.1, 'hum': 51, 'batt': 39}
09:58:54: AdvDecoder: c4:7c:8d:6a:08:85/-66/{'fertil': 2265}
09:58:54: AdvDecoder: c4:7c:8d:6b:32:9b/-82/{'moist': 61}
09:58:55: AdvDecoder: c4:7c:8d:6b:3f:3f/-65/{'moist': 42}
09:58:55: AdvDecoder: 3f:5b:7d:71:7c:38/-41/{'temp': 23.0}
```
But not sure in this time how to check more - you can do it faster ;)

michapr, 2020-10-09:
Found it out....
`if str(Settings.Tasks[x].address) == str(addr):`

was in my case:
`if A4:C1:38:C1:8C:71 == a4:c1:38:c1:8c:71` ... :)

So mac must be converted before to lowercase...

enesbcs, 2020-10-09:
Ehm, yes bluepy accepts and uses only lower case mac addresses. :) Scan results also showed in lower case.

enesbcs, 2020-10-09:
> But not sure in this time how to check more - you can do it faster ;)

Except when i am working. :D

enesbcs, 2020-10-09:
Updated at commit https://github.com/enesbcs/rpieasy/commit/8074e031549a7c81b903901d8f5f6a8de5eb34e1

michapr, 2020-10-09:
It is working fine now, thanks!

closed 2020-10-09 by michapr