enesbcs/rpieasy #175: JSON data in Generic MQTT controller

enesbcs, 2020-10-08:


michapr, 2020-10-08:
Hi,
here are some samples (from OpenMQTTGateway results):
`{"model":"MHO_C401","id":"A4:C1:38:5E:E1:0F","tem":22.09,"hum":52,"volt":2.953,"batt":85.3}`
`{"model":"LYWSD03MMC","id":"A4:C1:38:6B:F6:B2","tem":22.42,"hum":56,"volt":2.782,"batt":68.2}`

for flower care the values will be reported value by value - so maybe this JSON is right (like from OpenMQTTGateway):
`{"id":"C4:7C:8D:6A:08:85","name":"Flower care","rssi":-80,"distance":9.521558,"model":"HHCCJCY01HHCC","tem":21.9}`
`{"id":"C4:7C:8D:6B:3F:3F","name":"Flower care","rssi":-81,"distance":10.46739,"model":"HHCCJCY01HHCC","lux":311}`
`{"id":"C4:7C:8D:6B:3F:3F","name":"Flower care","rssi":-86,"distance":16.54559,"model":"HHCCJCY01HHCC","fer":621}`

But maybe a full JSON (with all values) could be better for some reason, not sure.

enesbcs, 2020-10-24:
"Use JSON payload" option added to C014 controller.
Backup first, as this commit contains several core related modifications also.
https://github.com/enesbcs/rpieasy/commit/2366fe1d9f9789fa523731510d407296f25b4191

michapr, 2020-10-25:
receive (as sample):
`{"taskname":"FlowerCare1","Temp":22.2,"Moist":67.0,"Light":351.0,"Fert":2391.0,"rssi":-71}`

Would be nice to get additional the MAC address as unique identifier for external systems ;)
(maybe as "port", so that it could be used also for other, not BLE devices, as parameter?)

enesbcs, 2020-10-25:
Port added at commit https://github.com/enesbcs/rpieasy/commit/4a9be108fa09058e1f4ea618029062284164b0d2

michapr, 2020-10-25:
`{"taskname":"FlowerCare2","Temp":20.9,"Moist":74.0,"Light":152.0,"Fert":0.0,"rssi":-72,"port":"c4:7c:8d:6b:3f:3f"}`

`{"taskname":"FlowerCare2","Temp":20.8,"Moist":74.0,"Light":136.0,"Fert":882.0,"rssi":-71,"port":"c4:7c:8d:6b:3f:3f"}`

The only "issue" - I think it would be better NOT to report missing values (like Fert":0.0) - it will be not useful in any database/graph ;)
Same I can see in standard MQTT report.

Thanks,
Michael

enesbcs, 2020-10-25:
Controller is for generic plugins and "missing values" not applicable.
In case of Xiaomi sniffer plugin they will come, when they will come.

michapr, 2020-10-27:
Yes, understand, that it is for generic plugins.
I think it would be better not to report no existing values to MQTT server - also the JSON must not contain all parameters every time. So if valid battery value was not send by the device - then not report it to MQTT server. 
I think this option is valid also for all other plugins.

For CGG1 as sample I receive this kind of data for battery (sniffer plugin)
[image]

If restarting (or if mpyEasy is restarting itself) we can get also "0" values for other parameters.

enesbcs, 2020-10-27:
> I think it would be better not to report no existing values to MQTT server - also the JSON must not contain all parameters every time.

There are no such things like "no existing values". Every value is existing, just not yet populated by the sniffer, but the Controller has zero knowledge about that. I guess that some kind of algorithm can be fabricated for Xiaomi sniffer plugin to do not publish until all values has larger than zero timestamps, as self._attribs cleared at each init().

It has nothing to do with the Generic MQTT controller, as task values has no timestamps.

> So if valid battery value was not send by the device - then not report it to MQTT server.
> I think this option is valid also for all other plugins.

If battery value is -1 or 255 than it is not forwarded by the controller as it is invalid. 0 is valid battery value.

michapr, 2020-10-28:
Is there any difference in code between mpyEasy and RPIEasy? As I can see (checking latest values) this effect (0 / 255 / [value]) I can see only in mpyEasy build....
(see picture)

enesbcs, 2020-10-28:
> Is there any difference in code between mpyEasy and RPIEasy? 

Yes. micropython and python3 is similar but i have to redone a lot of code in the core, as they are far from compatible.

enesbcs, 2020-11-16:
Issue is closed now as implemented. Please report mpyeasy issues at mpyeasy.

closed 2020-11-16 by enesbcs