enesbcs/rpieasy #178: Device:	Display - LCD2004 I2C

Epik77, 2020-10-16:
Hello,
after several hours of correct operation, the display shows dirty characters
https://snipboard.io/svMUq8.jpg
the device is seen correctly by the system:
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

I2C chip type: PCF8574

on this display there is a rule that prints the values of the pzem device every 5 seconds
if I restart the rpieasy service or update the Task Settings, the display resumes its correct functioning.

do you have any suggestions?

thank's

TD-er, 2020-10-16:
Have you tried lowering the I2C clock frequency?
See Hardware tab.

Epik77, 2020-10-17:
Hello,
with this script the speed of the i2c is detected at 100000

`#!/bin/bash
var="$(xxd /sys/class/i2c-adapter/i2c-1/of_node/clock-frequency | awk -F': ' '{print $2}')"
var=${var//[[:blank:].\}]/}
printf "%d\n" 0x$var`

should I try to increase it to 400000?
My hardware is a raspberry B +

thank's

TD-er, 2020-10-17:
If you get scrambled data, I would expect it to be caused by corrupted data.
So increasing the I2C clock does not make sense.
I guess adding (or lowering) the pull up resistors on both I2C data and clock lines may be more appropriate.
Typical values for such resistors are 4k7 ... 10k
But on higher speeds (and/or longer cables) you may want to lower those to something like 2k2.

Epik77, 2020-10-17:
Hello,
the connection cable between the raspberry and the display is 100cm long
there is no way to modify a frequency at the software level without inserting resistors?

TD-er, 2020-10-17:
You are mixing up 2 different issues.

100 cm cable is (officially) too long for I2C, but I guess it could be used if you know what knobs to turn.

A long cable may pick up noise as it is effectively an antenna.
So you may want to lower (or even add) pull-up resistors.
Those are resistors between the 3v3 and the both I2C lines (so 2 resistors).
They can be mounted at the LCD if that's more convenient.

The max usable I2C frequency may be limited by the following factors:
- Capacitance (like a capacitor) of the wires + connected devices.
- Duration of noise spikes
- Value of used pull-up resistors.

So if you use longer wires, the capacitance of the wires increases and thus the time needed for a pulse to change between low and high state. => lower the clock frequency
If you lower the pull-up resistors, you may lower the time needed for a pulse change, so you could keep the current clock frequency.

enesbcs, 2020-10-17:
> the connection cable between the raspberry and the display is 100cm long

100cm long? I've tried to use I2C temperature sensors over long cable, but after several failures i gave it up, finally replaced with DHT22 which works fine even on a 5m cable.

> after several hours of correct operation, the display shows dirty characters
> if I restart the rpieasy service or update the Task Settings, the display resumes its correct functioning.

Did you try the CLEAR or the ON/OFF commands if they can help when it occurs? If not, i can implement a REINIT command, that you could call periodically from rules, which will correct errors.
I guess that some internal memory error occurs on the chip.

I'm using RPLCD library to display text on LCD2004. Last RPLCD version on pypi released: May 3, 2020 i guess you are using the latest version?

I've found a compat_mode in the RPLCD issue tracker, i can add an option for it in RPIEasy:
https://github.com/dbrgn/RPLCD/issues/102

Epik77, 2020-10-17:
Hi

> Did you try the CLEAR or the ON/OFF commands if they can help when it occurs?
Yes  i tried but it doesn't work

>  If not, i can implement a REINIT command, that you could call periodically from rules, which will correct errors.
it could be a solution that solves me definitely the problem.
I already use the rules to print the display putting a REINIT command in a timer could be a great workaround.

thank's

enesbcs, 2020-10-17:
As compat_mode setting is not available for i2c instances of RPLCD, i will implement REINIT command.

enesbcs, 2020-10-17:
"lcdcmd,reinit" command added at commit https://github.com/enesbcs/rpieasy/commit/d194ad70e05cb209070d55137376b87fd51c0c1a

Epik77, 2020-10-19:
Hi,
I confirm the correct functioning of the command
   lcdcmd, reinit
24 hours have passed and the display no longer shows printing errors

thank's

closed 2020-10-19 by Epik77