enesbcs/rpieasy #179: extra I2C ports (BME280)

Marcmk1, 2020-10-18:
Hi I want to connect multiple samen BME80 devices on I2C port.
I support only 2 BME280 devices on the same bus  because of the adressing.

Is it possible to assign an 2th I2C channel?

to over come this issue I created manual an second I2C manual in the /boot/config.txt file for gpio 23 and 24.
dtoverlay=i2c-gpio,bus=4,i2c_gpio_delay_us=1,i2c_gpio_sda=23,i2c_gpio_scl=24

With an Python script I can read the outputs of this port/sensor.
I'm stuck trying to publish the sensor data to the mqtt broker.

enesbcs, 2020-10-19:
Second I2C is on GPIO0/GPIO1
According to my knowledge second I2C on Raspberry PI is reserved for external I2C EEPROM, which is checked on startup by the kernel.

I guess you are trying to use some sort of software-I2C, which may be an interesting addition but requires complete rewrite of the RPIEasy GPIO subsystem.

TD-er, 2020-10-19:
You could also consider implementing support for an I2C extender like the one added to ESPEasy recently.
This allows to switch I2C ports on the extender before the I2C communication with the sensor is performed.

N.B. this also can be used as a level switcher and also may be useful for isolating problematic I2C devices that really need a low clock frequency or may be attached to long cables and thus need a low clock speed due to increased capacitance of the cables.

enesbcs, 2020-10-19:
It can also be done as Linux has kernel support for I2C extenders in i2c-mux kernel driver:
https://www.kernel.org/doc/Documentation/devicetree/bindings/i2c/i2c-mux-pca954x.txt

But simple software i2c seems more easier and cheaper solution, as Raspberry Pi has a lot of  GPIO.

enesbcs, 2020-10-24:
Be advised as from today build 2.0.298 the handling of EVERY I2C plugin is a bit different, to support multiple I2C lines in the future.
Backup before upgrade, in case of errors either try to recreate I2C devices in RPIEasy and restart, or fallback to previous release:
https://github.com/enesbcs/rpieasy/releases/tag/2.0.292

Marcmk1, 2020-10-25:
Great!!
Works like a charme..

added in the /boot/config.txt file these lines:

dtoverlay=i2c-gpio,bus=7,i2c_gpio_delay_us=1,i2c_gpio_sda=6,i2c_gpio_scl=13
dtoverlay=i2c-gpio,bus=6,i2c_gpio_delay_us=1,i2c_gpio_sda=19,i2c_gpio_scl=26
dtoverlay=i2c-gpio,bus=5,i2c_gpio_delay_us=1,i2c_gpio_sda=20,i2c_gpio_scl=21

After that I can asign it in the Devices page
[image]

enesbcs, 2020-10-25:
Super! I've plans to add some selector boxes to enable software I2C on the hardware pages with the GUI, but may take some time.

enesbcs, 2021-01-01:
Selector boxes added to hardware page at commit https://github.com/enesbcs/rpieasy/commit/056fcbbcbfa617dea1e0747184c23c010a241361

closed 2021-01-11 by enesbcs