enesbcs/rpieasy #180: LWSD02 time sync problme

cyraxxx, 2020-10-25:
Hi,

I tried the modified version of RPIEasy which can handle BLE devices better and it is working like a charm. But now I need to sync the time because of winter time. I remember when I first installed RPIEasy it synced the Chinese time to my local time but now it simply don't work. I have 4 LWSD02 devices, disabled 4, enabled 1, tried to remove battery, restart RPIEasy, nothing helped. RPIEasy tries to sync (11:48:41: Sync LWSD02 time) but it is not working while temp and humidity datas are still arriving. Do you have any idea what to check or try?

enesbcs, 2020-10-25:
LYWSD02 library is caching _tz_offset parameter and it is saved at the first creation of the plugin as i see. I guess i can implement to reset timezone always when pressing Submit.

enesbcs, 2020-10-25:
Timezone fix added at commit https://github.com/enesbcs/rpieasy/commit/4a9be108fa09058e1f4ea618029062284164b0d2

Please install missing pytz dependency to let the plugin automaintain Lywsd02 timezone parameter. ( Plugin&controller dependencies )

cyraxxx, 2020-10-25:
Perfect, working, thank you!

michapr, 2020-10-25:
hmm... pytz was already installed. but time will not be synchronized after update if pressing submit.
In log I cannot see anything interesting...
```
19:38:03: Event: LYWSD02-R#Temperature=23.3
19:38:09: Event: LYWSD02-R#Humidity=52.0
19:38:09: Event: LYWSD02-R#Battery=47.0
19:38:09: Event: Clock#Time=Sun,19:38
```
Any idea what can be wrong?

enesbcs, 2020-10-25:
Strange, its working on my LYWSD02. 

Try to:
- move the Lywsd02 closer to RPI unit
- do full apt update on Linux (at least on tzdata package)
- upgrade lywsd02 pip package if possible
- reboot RPI

If nothing helps, try to delete, then recreate Lywsd02 Task.

michapr, 2020-10-26:
All is working as expected....
I have not seen that other RPi was blocking the BLE connection. 
There is another RPIEasy version running with scanning plugin and every 4 hours is checking the battery of LYWSD02 connecting to the device.
This connection was not closed until a reboot of the RPi...
After reboot the actual version on first Pi was working as expected.

Thanks!
Michael

enesbcs, 2020-10-26:
Thank you for feedbacks, issue is now closed as solved.

closed 2020-10-26 by enesbcs