enesbcs/rpieasy #182: Pulse counter

curgan78, 2020-11-07:
Szevasz!

Áramlásmérőt építettem a fűtésvisszatérőbe. Pulse Counter-rel olvasom ki, de nem találok formula megadási lehetőséget. Az álamlásmérő specifikációja szerint Q*6-8=számolt érték, ahol Q=L/min ESPeasyben benne van, de itt nem találok ilyen lehetőséget.

Köszi
Gábor

enesbcs, 2020-11-07:
Hm, jogos, a formulaoption paraméter kimaradt a pluginból valamiért és eddig nem hiányzott senkinek. :)
Pótolva. Elképzelhető, hogy újra létre kell hozni a taszkot, hogy megjelenjen.

curgan78, 2020-11-07:
Nagyon köszi agyors reagálást!

Köszönettel:

 Gábor

Alexander Nagy <notifications@github.com> ezt írta (időpont: 2020. nov. 7.,
Szo, 16:32):

enesbcs, 2020-11-09:
Issue solved

closed 2020-11-09 by enesbcs