enesbcs/rpieasy #183: issue in _P201_GenSerial.py using string method

bouracho, 2020-11-22:
Hi, 
thanks for your work.
I used rpieasy to control my Neato Robot with an USB connexion on a pi zero w.
I have to write string in this serial connexion.
I found an issue in your plugin_write() function.
In fact, you have to had a carriage return at the end of the string to simulate the "enter" command. I can validate that with screen.
I'm not familiary with python so I don't know if my code is good but it works.
I just changed 
**line 251 :** sbuf = str(text)
by mymsg =  str(text)+"\n"
**line 254 :** self.serdev.write(sbuf)
by self.serdev.write(mymsg)

I hope you will correct it in next release (with better code ;-) )

regards

ben

enesbcs, 2020-11-22:
Interesting, but i never planned to add carriage return to serialwrite as hex payload is supported, it can be easily done like 0x5445535410
Anyway in the next release serialwriteln command will be included for this purpose.

enesbcs, 2020-11-22:
serialwriteln added at commit https://github.com/enesbcs/rpieasy/commit/39e12c625e972ed59ff5da921e85f59714eaae01

bouracho, 2020-11-23:
update ok, work as expected
thanks

enesbcs, 2020-11-24:
Thanks for feedback, issue is closed now.

closed 2020-11-24 by enesbcs