enesbcs/rpieasy #184: improvement in lib/lib_serial.py

bouracho, 2020-11-22:
hi,
I use the serial plugin (201) to manage my neato robot throw USB.
Each time I have to diconnect/connect usb to start the cleaning.
The usb tty change every time  so I used udev to fix it with tty symlink
Unfortunetly, this tty it's not enumerate by the plugin.
I found a solution with the function serial.tools.list_ports.comports() that you use in line 133 of this lib file.
You just have to add "1" in parameter to allow listing of serial symlinks.
It works like a charm for me.
Could you add this feature ?
Thanks a lot.

Ben

enesbcs, 2020-11-22:
I can add this, although on my testPC it does not matter what i add, only /dev/ttyS0 and /dev/ttyUSB0 listed with this function.

bouracho, 2020-11-22:
My USB serial tty on "pi zero w" can be /dev/ttyACM0, /dev/ttyACM1, /dev/ttyACM...etc...etc..
So I create this udev rule, /etc/udev/rules.d/99-usbserial.rules : 
ACTION=="add",ENV{ID_BUS}=="usb",ENV{ID_SERIAL}=="Linux_2.6.33.7_with_fsl-usb2-udc_Neato_Robotics_USB_v2",SYMLINK+="ttyNeato"

Without the 1 in parameter, I see :
/dev/AMA0
/dev/ACM0

with this parameter:
/dev/AMA0
/dev/ACM0
/dev/ttyNeato

So I can add a device with a symlink tty which can work every time

enesbcs, 2020-11-22:
Added at commit https://github.com/enesbcs/rpieasy/commit/39e12c625e972ed59ff5da921e85f59714eaae01

bouracho, 2020-11-23:
update ok, work as expected
thanks

enesbcs, 2020-11-24:
Thanks for feedback, issue is closed now.

closed 2020-11-24 by enesbcs