enesbcs/rpieasy #185: Need advice on pinout for new hardware design

happytm, 2020-11-23:
Hi Alex,

I am in process of designing new PCB for home automation using Raspberry Pi Zero.I need your help in identifying proper pins for following hardware:

- I2C Bus for temperature, humidity, pressure, light and air quality sensors.
- 18 led NeoPixel strip.
- IR receiver.
- IR transmitter.
- RCWL-0516 – Doppler Radar Sensor for presence detection.
- HC-SR04 ultrasonic range finder
- Interrupt pin for APDS-9960
- I am not sure what you use for ADC but I woluld like to have couple of ADC channels to measure battery power and  an extra 
  channel for future.
- If possible Lora Radio prefferably using SX1262 chip EBYTE Module ( http://www.ebyte.com/en/new-view-info.aspx?id=332).
-MP3 Player for system announcements and alarms.

Please help me select proper pins.

Thanks.

happytm, 2020-11-23:
I found this comment at https://meshtastic.discourse.group/t/tbeam-with-sx1262/972/25?u=happytm for SX1262 LORA using chip with ESP32.

This is what I have in configuration.h but you could define nearly any pin (there are some limitations).

#define SX1262_CS 5
#define SX1262_DIO1 14
#define SX1262_RESET 13
#define SX1262_BUSY 33
#define SX1262_RXEN 12
//#define SX1262_TXEN
#define SX1262_E22
Standard SPI definitions are:

// -----------------------------------------------------------------------------
// LoRa SPI
// -----------------------------------------------------------------------------

#define SCK_GPIO 5
#define MISO_GPIO 19
#define MOSI_GPIO 27
#define NSS_GPIO 18
I use DIO2 to control the TX_en pin of the E22 module which is assumed by the code if you don’t define a control GPIO with:

#define SX1262_TXEN xx

TD-er, 2020-11-23:
Just a warning about this one:

- RCWL-0516 – Doppler Radar Sensor for presence detection.

This one is extremely sensitive to near by conductive materials (copper/iron... you get the idea)
So I would not try to include that in the PCB unless you have done some tests on it.

By the way, it is possible to tap off a pin of the chip and get analog reading from it :)

happytm, 2020-11-24:
@TD-er  Thank you for your warning abour radar sensor. I have followed and discussed same issue at https://github.com/jdesbonnet/RCWL-0516/issues/2. The solution seems to be to use higher voltage (like 10 to 12 v) using boost converter. I am also going to provide header only on PCB so I can experiment with locating the module at suitable place.

I will be waiting for Alex to chime in for other suitable pins I can use.

Thanks.

happytm, 2020-11-24:
@enesbcs I forgot to tell you that because of space and layout limitation I can only use physical pin number 1 to 28 of RPI ZERO.

Thanks.

enesbcs, 2020-11-24:
Hi!

>     * I2C Bus for temperature, humidity, pressure, light and air quality sensors.

RPI default hardware I2C1 is on GPIO2 and GPIO3. (this is the only usable HW I2C on RPI)
But as i found out recently, there are dtoverlays which enables software I2C on _any GPIO_. (it is slower a bit, but very comfortable) Currently it is not implemented in RPIEasy hardware settings, but work in progress.. if manually added, it can be tested with (most) I2C device. https://github.com/enesbcs/rpieasy/issues/179

>     * I am not sure what you use for ADC but I would like to have couple of ADC channels to measure battery power and  an extra

I am using 4 channel ADS1015 devices for analog inputs, which is a standard I2C device, I2C pins as before. INA219 also a good choice if voltage measurement is the main goal. (also I2C) 
MCP3008 ADC also supported, which is SPI0 is at GPIO10,GPIO9,GPIO11,GPIO8.
SPI1 is after pin28 so you wont use it i guess.
 
>     * 18 led NeoPixel strip.

One pin is enough to drive a lots of neopixel leds, but you have to deal with level-shifting (RPI GPIO can only handle 3v3) and a separated power supply for leds. (5v usually)
This pin must be a hardware PWM capable one. RPI has four of it, and two of them can be active at the same time, either Channel0 or Channel1.
Channel0: GPIO12 OR GPIO18
Channel1: GPIO13 OR GPIO19
GPIO18 is quite popular (for ex: pyLora and IR also using it), so choose wisely when planning wiring.

>     * IR receiver.
>     * IR transmitter.

Did you try these on RPI earlier? IR-TX works mostly, but IR Receiver part is not very reliable according my tests. :( Its based on LIRC which has a kernel driver that deals with IR decoding, it is the best and fastest solution for a Linux based microcomputer. But i had to admit, that the ESP8266 is better in time-sensitive tasks as IR decoding or RF decoding.

gpio-ir receiver and gpio-ir-tx sender defaults itself to GPIO18 but can be changed to any free pin.
On the other hand pwm-ir-tx sender is more reliable but can be used on either GPIO18 or GPIO12.

>     * RCWL-0516 – Doppler Radar Sensor for presence detection.
>     * HC-SR04 ultrasonic range finder
>     * Interrupt pin for APDS-9960

Any free pin will be good for these.

>     * If possible Lora Radio prefferably using SX1262 chip EBYTE Module - http://www.ebyte.com/en/new-view-info.aspx?id=332.

SX1262 is not supported by RPIEasy, and i do not find any python related libraries.
Currently the only python LoRa library i found and tested is pyLoRa which works with SX127x familiy and RFM95x chips, or HPD13A. Pinout is hardwired in the lib, and can be seen on the project page and on C020 controller settings.
https://pypi.org/project/pyLoRa/

>       -MP3 Player for system announcements and alarms.

I am using USB sound card with Speakers and PyGame plugins for playing wav/mp3 files in RPIEasy. Do you want to use some kind of serial MP3 players? I guess i can add a plugin for this kind of device in case it is needed, Serial1 is GPIO14 and GPIO15 on RPI.

enesbcs, 2020-11-24:
> @TD-er Thank you for your warning abour radar sensor. I have followed and discussed same issue at [jdesbonnet/RCWL-0516#2](https://github.com/jdesbonnet/RCWL-0516/issues/2). The solution seems to be to use higher voltage (like 10 to 12 v) using boost converter. 

It is very interesting as RCWL has an onboard step down converter which converts immediately down to 3v3. I am curious how it can be more stable with higher voltage? I have tried to add capacitors between vin and gnd, also tried to place aluminium foil behind the rcwl, it is a bit better now.

happytm, 2020-11-24:
Hi Alex,

Thank you for your detailed response. Now most of the pin usage is clear to me . 

For LORA I will bring out 10 pins on header s so in future if SX1262 is supported by RPIEasy I will have access to those pins. I will use any pin out of 4 other than in 18 which can be used for LORA. 

I am planning to use MCP3208 for ADC with 4 channel used for mains current measuring described here https://github.com/boblemaire/IoTaWatt sparing 3 of ADC Channels for other uses including measuring battery voltage.

MP3 player is not as important decision right now so I can use USB just as you do and have serial pins on header for future.

Thanks.

enesbcs, 2020-11-24:
> Thank you for your detailed response. Now most of the pin usage is clear to me .

No problem.

> For LORA I will bring out 10 pins on header s so in future if SX1262 is supported by RPIEasy I will have access to those pins. I will use any pin out of 4 other than in 18 which can be used for LORA.

We are talking about SPI connected SX1262 like this?
https://www.aliexpress.com/item/32919771141.html?spm=a2g0o.cart.0.0.17943c00atIOJb&mp=1
I've just found a micropython lib for sx126x https://github.com/ehong-tl/micropySX126X which may be used for CPython, but needs some rewrite and more testing.

happytm, 2020-11-24:
Yes it is similar for function but I was going to use it's smaller and less powerful brother linked here:

https://www.aliexpress.com/item/33015323065.html?spm=a2g0o.productlist.0.0.74fcf7cbZ64x22&algo_pvid=4d3aa85b-c372-42a0-892f-40e5f3ccec1a&algo_expid=4d3aa85b-c372-42a0-892f-40e5f3ccec1a-1&btsid=0bb0623b16062613430472234e2daf&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

This one put out 22dBm compared to 30 dBm output for one you link.They claim range of upto 15kms.

I like Ebyte modules as they are more powerful then other lora modules and easy to hand solder.

Thanks.

happytm, 2020-11-25:
Just found another variant of SX1262 chip combined with STM32 ultra low power MCU. they are very compact at 15mmx15mm.It will be nice if it communicate with RPI Zero either via serial or USB. Links for product below:

https://docs.rakwireless.com/Product-Categories/WisDuo/RAK4270-Module/Datasheet/#description
https://store.rakwireless.com/products/rak4270-lpwan-module?variant=36313068142750

It is arduino compatible  as shown here: https://github.com/ElectronicCats/BastWAN.

Thanks.

happytm, 2020-11-25:
I found this arduino library https://github.com/PiSupply/rak811-arduino  for serial passthrough example between arduino Nano and STM32L used on RAK811 which has SX1276 Lora chip on it . The library should work on RAK4270 too as it is just communication between 2 MCUs which are both STM32L. I also noticed that RAK811 have 6 ADC channels.

Thanks.

TD-er, 2020-11-25:
What are you trying to achieve with LoRa?
Do you want to connect to a LoRaWAN network, or is it peer-2-peer communication?

And stated range of LoRa modules are not bogus, but they do depend on mainly 2 factors (transmission power isn't one of them):
- Antenna height of sender and/or receiver
- Antenna + cabling

Meaning if you have both units placed on the ground with an antenna in the enclosure, I can guarantee you won't make 10 km, regardless of the transmission power (within a certain range of course)

At the Delsbo Electric event we had a gateway mounted at roughly 3 meter and the sending unit was at +/- 40 cm from the ground.
With line-of-sight you could cover roughly 2 km.
Transmission power was 14 dBm (20 mW).
The gateway had a very good antenna measuring roughly 1m.
The sender had a very decent antenna measuring roughly 17 cm and was mounted outside the enclosure.

If you plan on p2p communication, those EByte units probably work just fine.
If you plan on connecting to a LoRaWAN gateway, then don't use those as there is no library for them to implement the LoRaWAN stack.

enesbcs, 2020-11-25:
> I found this arduino library https://github.com/PiSupply/rak811-arduino for serial passthrough example between arduino Nano and STM32L used on RAK811 which has 

Interesting, but serial can only be the last resort. SPI is much faster and reliable for these things i think.

> Yes it is similar for function but I was going to use it's smaller and less powerful brother linked here:
> This one put out 22dBm compared to 30 dBm output for one you link.They claim range of upto 15kms.

The one i linked is cheaper and has 2.54mm pinout spacing. 1.27 spacing modules (like the one you linked) is a bit small to me for soldering. I will order the bigger one for testing. :)

> If you plan on p2p communication, those EByte units probably work just fine.
> If you plan on connecting to a LoRaWAN gateway, then don't use those as there is no library for them to implement the LoRaWAN stack.

TD-er is right, these modules will not be very good for LoRaWAN, but it is not a big deal, as i do not plan to support any LoRaWAN functionality in the near future, only P2P.

happytm, 2020-11-25:
@TD-er I am planning to use it to collect data from remote sensors in a farm. It will be used to collect data locally so I am not worried about lack of support for LoRaWAN network. I understand range is very complicated subject but everything being equal new generation SX1262 specs suggest it will outperform those old generation Lora chips in receive sensitivity , transmit power and power consumption. I have read about lot of field test report from professional RF people to make above statement.

@enesbcs The serial passthrough is only used for data exchange between built in MCU on RAK811 and any other board (in our 
                  case RPI). The heavy work of long range data exchange is done by RAK811.

Yes the bigger one is cheaper right now but because we are linking 2 different vendors. You are linking directly to manufacturer of the modules and they do not have smaller module in stock right now. The pinout is definitely a plus point for the bigger one besides extra power.

Thanks.

enesbcs, 2020-11-25:
> @enesbcs The serial passthrough is only used for data exchange between built in MCU on RAK811 and any other board (in our
> case RPI). The heavy work of long range data exchange is done by RAK811. RAK811 looks appealing as it gives us 6
> 12 bit ADC channels which raspberry lacks.

I see. But RPI serial communication is still the last communication form i will choose ever.
There are no heavy work in case the SPI device, as the LoRa handles everything, it just pulls IRQ GPIO line to sign RPI that a message arrived... and a simple spi read will gather information. 
Honestly, polling a slow UART line continously with an RPI is more resource-intensive, than waiting for an IRQ from the SPI LoRa device, and reading only once. Not impossible, but unpractical. In the end i have to say, that reading and translating the UART line is more heavier than IRQ driven SPI.

Yes, ADC is a good point, but an ADS1015/MCP3008 for 2 USD will also do this job.
I've found RAK811 at aliexpress from 15-20 USD and i have to say that the 5 USD SPI SX1262 is more sympathetic to me.

happytm, 2020-11-25:
Your points well taken. As for ADC I think instead of using any ADC chips I will use either ESP32 or NRF52 or STM32L chip since I am designing my board from scratch. These MCU cost around $2.50 but they will give lot of extra features. I am planning to use 28 pin connector on board to connect RPI Zero W.

For Lora I found another arduino library  which claims to work with ESP32, ESP8266 and NRF52 boards and compatible with lorawan. There are some discussion around Ebyte modules which author used for his testing in issues section. Below are links:

https://github.com/beegee-tokyo/SX126x-Arduino
https://github.com/beegee-tokyo/SX126x-Arduino/issues/12
https://github.com/beegee-tokyo/SX126x-Arduino/issues/18

Thanks.

enesbcs, 2020-11-26:
Thanks these libraries may be handy for implementing SX126x, but i hope that the micropython sx126x lib will work after changing code to work with RPI spidev lib.

happytm, 2020-11-29:
@enesbcs For energy monitor I have decided to use
https://github.com/Khaalidi/HLW8032/blob/master/Docs/DS_HLW8032_EN_Rev1.5.pdf .I believe this chip is used on Wolfblitz automated sockets with ESP8266. It is a serial output device. There is arduino library here https://github.com/Khaalidi/HLW8032.

Do you see any issue using it with RPI?

Thanks.

TD-er, 2020-11-29:
The one implemented in ESPEasy (thus for the various power measurement ESP boards out there) is using the HLW8012
Let's hope the one you link is better than the HLW8012 as that one isn't really accurate and also quite unstable for loads with a low power factor.  (and also for low energy consumption appliances)

happytm, 2020-11-29:
The sales literature of the company claims: 

- It is better stability and more accurate.
- No calibration required. 
- Requires only single pin RX.

There are not many libraries on github. Some people report using it  on Tasmota & Espurna github repos. There is a vendor who sells it here https://www.electrodragon.com/product/energy-meter-hlw8032-breakout-board/ but they do not have arduino library.

Thanks.

TD-er, 2020-11-30:
If I were you, I would go for the "isolated" version.
Working with mains is bound to go wrong at some point.

happytm, 2020-11-30:
That's what I am thinking too.

Thanks.

TD-er, 2020-11-30:
Only disadvantage of using the "isolated" one is that it is using an inductor for measuring current.

Measuring using an inductor has the following drawbacks compared to measuring using a shunt:
- Phase (and thus power factor) is harder to measure correct. => expect some larger error on devices with low power factor.
- Larger error on low currents as it may pick up noise from elsewhere.

And only if the measured currents are larger then can be handled by the ferrite core, you will get significant errors.
But I guess those are still preferrable to running such currents over a shunt :)

happytm, 2020-11-30:
@TD-er   What you think about this system. 
https://www.youtube.com/watch?v=8KXLDy0pp4I&t=2091s

Thanks.

enesbcs, 2020-11-30:
> @enesbcs For energy monitor I have decided to use
> https://github.com/Khaalidi/HLW8032/blob/master/Docs/DS_HLW8032_EN_Rev1.5.pdf .I believe this chip is used on Wolfblitz automated sockets with ESP8266. It is a serial output device. There is arduino library here https://github.com/Khaalidi/HLW8032.

I bought a HLW8012 earlier for testing purposes, but high freq outputs can not be read reliably in a multi-tasking OS controlled RPI, so i've stopped dealing with that. 
HLW8032 with UART communication has to be easier to implement than HLW8012. It is possible.

And what about RS485 PZEM016? I think it is a similar device and already implemented.

happytm, 2020-11-30:
@enesbcs thank you for your hint. I have lot of reading to do now.

happytm, 2020-12-09:
ST just announced this:

https://st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-wireless-mcus/stm32wl-series.html?ecmp=tt19127_gl_social_dec2020

They claim first micro with built in LoRa. May be this is your starting point.Their developement board are sold out at the moment.

Thank you.

enesbcs, 2020-12-09:
I am not really understand what you mean about "starting point"?

This board is interesting, but as i said there are absolutely no need for other MCU for a Raspberry to do LoRa. There are simply no reason. SPI is the best approach.

I guess they aim developers whom wants to develop their firmware to STM32. When they will have at least 4MB flash as ESP32 boards has, i will be interested, but 256kByte flash.. they lost me.

I have added Lora direct support for ESP32 at mPyEasy, i am moving towards this direction. Honestly even 4MB is low a bit. If OTA is needed 8MB or 16MB boards will be needed in the near future.

happytm, 2020-12-09:
It is not just LoRa but all other benefits which comes with it like ADC's , Touch interface & simplicity of All-in-One hardware so you don't have to worry about buying expensive Lora modules. That said it will be only worth looking into if chip price is around $5.

Thanks.

happytm, 2020-12-13:
I came across this amazing video and I am definitely going to add hardware to my design to experiment with it. It uses ESP32 I2S bus for audio input and output.

https://www.youtube.com/watch?v=re-dSV_a0tM

Thanks.

enesbcs, 2020-12-13:
Interesting, but as i see these projects depends on Cloud based voice recognition.

ESP Skainet runs on ESP32-LyraT and ESP32-Korvo boards and if i understand well, they are working offline, and able to decode 100 voice command after voice activation:
https://www.hackster.io/news/esp32-korvo-dev-board-launches-alongside-english-language-esp-skainet-voice-assistant-platform-dcf20688a82a

happytm, 2020-12-14:
The products you linked use ADC so there are lot of noise issue. The I2S protocol for microphone is better for microphone voice quality. There is another example for Raspberry PI Zero which does not require cloud connection. Link is below:

https://github.com/HarringayMakerSpace/voice-controlled-switch

Thanks.

enesbcs, 2020-12-14:
> The products you linked use ADC so there are lot of noise issue. The I2S protocol for microphone is better for microphone voice quality. 

Perhaps, but Skainet AEC & AGC may filter noise. I've ordered LyraT for testing. :) It looks very nice.
https://github.com/espressif/esp-skainet 
https://www.cnx-software.com/2020/04/27/esp32-korvo-ai-development-board-leverages-esp-skainet-voice-assistant/

85% detection rate seems to be still amazing to me.
https://www.espressif.com/en/news/ESP-Skainet_Released

>There is another example for Raspberry PI Zero which does not require cloud connection. Link is below:

Yep, but Speech Recognition is already integrated into RPIEasy by PocketSphinx for linux based microcomputers.
https://github.com/enesbcs/rpieasy/blob/master/_P506_PocketSphinx.py

happytm, 2020-12-14:
I did not know about PocketSphinx until now. Looks like I2S is very complicated.

Below is link for reference where he has tested few audio chips:

https://github.com/miketeachman/micropython-esp32-i2s-examples

Thanks.

enesbcs, 2020-12-14:
> I did not know about PocketSphinx until now. 

I too just stumbled upon ReSpeaker boards by accident while browsing the net, and ReSpeaker uses PocketSphinx for offline voice recognition.
https://spin.atomicobject.com/2016/09/19/respeaker/
 
> Below is link for reference where he has tested few audio chips:
> 
> https://github.com/miketeachman/micropython-esp32-i2s-examples

Yep, and it is only raw data. :)

happytm, 2020-12-15:
This recent blog post of Lady Ada have lot of technical details about machine learning.

https://www.raspberrypi.org/blog/adafruit-guest-post-machine-learning-add-ons-for-raspberry-pi/

Thanks.

happytm, 2020-12-15:
I found open source hardware design based on respeaker here:

https://github.com/Makerfabs/Raspi-Voice-Interaction-Hat

Looks like it uses 3 pins of RPI ( 35, 38 & 40) for audio processing and 5V and GND to power whole board.

Thanks.

Thanks.

closed 2021-03-04 by enesbcs