enesbcs/rpieasy #187: _P201-_GenSerial.py test connection to serial and reconnect if it'is down

bouracho, 2020-11-23:
Hi Alex,
thanks for the 2 patches.
Another thing I want to discuss with you.
In this plugin you never test the connection once is established.
If i unplug/plug my USB device there is no warning or error with the plugin and the command "serialwrite" does'nt work anymore but nothing said that.
The only thing I found is to re-save the device that open a new connection bu it's not dynamic.
To resolve I just added self.connect() before write() at line 266. I'm sure it's not the best way but it works. I can unplug/plug USB device and the plugin always reconnect.
I hope you will find a better solution.
Regards
ben

enesbcs, 2020-11-24:
Earlier i have used a "self.connect()" when read error occured in the background process. You can enable it by removing comment sign from line 236. But i have removed it for some reason, it must crash something badly. :)

I think i can add a try except before write, then connect if write failed and repeat write operation for your use-case it may be enough.

enesbcs, 2020-12-15:
From commit https://github.com/enesbcs/rpieasy/commit/3862088e8ba26ae2a8e0c3ab8882644f8a59f001
plugin will automatically try to reconnect on failed write() and rewrite.

bouracho, 2020-12-15:
Thanks a lot, i will try asap.

enesbcs, 2021-01-23:
Closed as no more feedback received. Feel free to reopen in case of further related problems.

closed 2021-01-23 by enesbcs