enesbcs/rpieasy #188: Run OS Command plugin.  Bug or ??

Budman1758, 2020-12-02:
Hello.
I am using the RunOS "device" to send D-Bus commands to a program running on the same machine. The program is [Navit ](https://github.com/navit-gps/navit/issues/1059)and the only way (apparently) to control certain things manually is using the D-Bus service. Currently the only thing I am trying to switch is the day/night mode of the program display. I am using the commands according to this page on [Navit's Wiki](https://wiki.navit-project.org/index.php/Dbus#Common_actions).

These are the commands I am trying to use in the RunOS device.

dbus-send --print-reply --session --dest=org.navit_project.navit /org/navit_project/navit/default_navit org.navit_project.navit.navit.set_layout string:"Car-dark"

dbus-send --print-reply --session --dest=org.navit_project.navit /org/navit_project/navit/default_navit org.navit_project.navit.navit.set_layout string:"Car"

These commands work as they are supposed to from a terminal session but nothing at all happens when using them in the RunOS plugin.  The web log page shows that the command was run successfully but the action does not happen.

I don't know if this is a bug in the RunOS command or if there needs to be something else used. 
Let me know what I can do to help troubleshoot this.  Thanks.

enesbcs, 2020-12-02:
Dbus can not be accessed by a process, which is started by another user. RPIEasy runs at root user as this is the only way to access every hardware component without obstacles. D-Bus session is opened by the user who logged in at the welcome screen, which is not root.
You can try to run RPIEasy by the same user which do you use to log in to the GUI. Disable default autorun inside RPIEasy and consult your Desktop Environment manual about how to start automatically an application after login.

Budman1758, 2020-12-02:
I log into the computer as "Pi" Would I not have to run RPIEasy as that user?

enesbcs, 2020-12-03:
> I log into the computer as "Pi" Would I not have to run RPIEasy as that user?
> Or, run D-Bus as user Pi?

If you are logging in automatically as Pi, your Dbus system is initialized as Pi. You should try to start RPIEasy as Pi. (but do not forget to stop the RPIEasy integrated startup as root, as two RPIEasy instance at once will results strange errors)

You need Method 2 from this:
https://learn.sparkfun.com/tutorials/how-to-run-a-raspberry-pi-program-on-startup/all

Budman1758, 2020-12-03:
Program does not seem to want to run as PI user. Cannot get the AutoStart to work and if I try to start the program from a terminal it seems to hang at "loading rules...

```
Starting RPIEasy 2.0.330
Init hardware...
Load network settings...
Load GPIO settings...
Load devices from file
Load controllers from file
Load notifiers from file
Loading rules...
```
This is the contents of the .desktop file....

```
[Desktop Entry]
Type=Application
Name=RPIEasy
Exec=/usr/bin/python3 /home/pi/rpieasy/RPIEasy.py
```

 Ideas?

Budman1758, 2020-12-04:
OK.  Got this working.  One thing that's rather curious though. When running as user PI the program will not run on port 80. While running as root I unchecked all but port 80 in the advanced settings page. After doing that it will not run as PI. Runs fine on port 8080 or 8008 but not 80.  Not a big deal but seems odd to me. There are no other programs using port 80. Console shows "permission denied" when attempting to run as PI.

TD-er, 2020-12-04:
Only processes with root level rights can bind to ports below 1024.

TD-er, 2020-12-04:
If you really would like to access it via port 80, or 443, you can also install Apache and make a vhost config in Apache with very basic `proxypass` statements to map HTTP(S) traffic to a specific port (which can be local or anywhere on your network)
Just make sure to add also the reverse proxypass.

Example:
```
<VirtualHost *:80>
    DocumentRoot "/www/example1"
    ServerName rpieasy.mydomain.nl

    ProxyPass "/"  "http://192.168.10.97:8080/"
    ProxyPassReverse "/"  "http://192.168.10.97:8080/"

    # Other directives here
    CustomLog /var/log/apache/rpieasy.log combined
</VirtualHost>
```

enesbcs, 2020-12-05:
> 8080 or 8008 but not 80. Not a big deal but seems odd to me. There are no other programs using port 80. Console shows "permission denied" when attempting to run as PI.

Only root has right to do elevated things, such as opening a port below 1024.
There are a lot of system calls which will not run properly without root permissions, such as modifing system configuration, pin configuration, alsa volume change, hardware PWM, Wifi scan, BLE, etc..

TD-er, 2020-12-05:
I have not yet looked into Rpi Easy that well, but I guess it does make sense to have all those things that need low level access to run as a service.
A service is typically run by a user with elevated right, so therefore it makes sense to have R'pi Easy also run as a service.

enesbcs, 2020-12-05:
> A service is typically run by a user with elevated right, so therefore it makes sense to have R'pi Easy also run as a service.

Except when someone desperately needs to reach Dbus session, which is not accessible by other users, non even root for security reasons. ;)

TD-er, 2020-12-05:
Can't you run some 2nd service then to access that session and hand over the data?
Essentially voiding the security :)

enesbcs, 2020-12-06:
> Can't you run some 2nd service then to access that session and hand over the data?
> Essentially voiding the security :)

I've done it earlier for the Jami plugin (jamibridge), as Jami also using DBus API... but this bridge is not rock solid, i am thinking seriously about dropping Jami Dbus as it also supports REST API, however has almost zero documentation.

closed 2021-01-02 by enesbcs