enesbcs/rpieasy #189: Error: MQTT: Invalid data received!

Epik77, 2020-12-14:
Hi,

I wanted to tell you that when I start the rpieasy application, the domoticz service reports

2020-12-14 12:31:33.949 Error: MQTT: Invalid data received!
2020-12-14 12:31:34.050 Error: MQTT: Invalid data received! 

this happens in Mode: MQTT

These two errors occur only during the service start-up phase and are then no longer encountered
with other mqtt clients (espeasy, tasmota, etc ..) I have no problems.

thanks for the support

enesbcs, 2020-12-14:
RPIEasy uses paho-mqtt which has no reliable way to ensure if socket is disconnected or not. So RPIEasy tries to send "PING" package periodically through MQTT and if it is unsuccesful, it forces a reconnect.
I can add an option to the controller to do not test connection, but in that case it will never try to reconnect on case of failure.

Epik77, 2020-12-14:
yes if you can put it as an optional field recommended on wifi connection

I currently use a cable connection I think I have no MQTT disconnections

thank's

enesbcs, 2020-12-15:
Ping option added to C002 at commit https://github.com/enesbcs/rpieasy/commit/3862088e8ba26ae2a8e0c3ab8882644f8a59f001

Epik77, 2020-12-16:
Ok thank's
I confirm the problem has been solved

closed 2020-12-16 by Epik77