enesbcs/rpieasy #190: Is support possible for LCD display?

Budman1758, 2020-12-28:
I'm curious if there could possibly be support for this display.
[https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/2.4inch-lcd-module.htm](url)

I'd be glad to donate one for testing purposes.  :>)

TD-er, 2020-12-28:
Given it is color, I guess it will be very challenging for the ESP8266 to keep track of all the memory of a single page.
Maybe uGL lib  (or any other lib supporting those) can be used for these?

enesbcs, 2020-12-28:
Interesting idea. Raspberry Pi has some kernel modules which supports SPI display, altough 320x240 is very low resolution for XFCE. 
I guess something similar can be made for RPIEasy like the P036 FrameOLED plugin, using luma.lcd library.
https://pypi.org/project/luma.lcd/

Budman1758, 2020-12-28:
I was thinking of something like the framed OLED plugin.  Not full blown graphics like video or anything like that.  Maybe something like colors or a picture as a background for lines of output like the framed OLED. Or possibly a different graphic per "page" of output.  Something along those lines.....

enesbcs, 2021-02-16:
Support added at commit https://github.com/enesbcs/rpieasy/commit/2aa0cc6cff8fa0599e9d080038db844477b1e564

Currently it is very similar as I2C OLED Framed plugin, but with support of these commands: (if plugin interval setted to 0)

#| command | details | description |
#|-----|-----|-----|
#| tft | `TFT,[tft_subcommand],....` | Draw line, rect, circle, triangle and text |
#| tftcmd | `TFTCMD,[tftcmd_subcommand]` | Control the screen (on, off, clear,..) |
#
#TFT Subcommands:
#| TFT Subcommands | details | description |
#|-----|-----|-----|
#| txt | txt,[text] | Write simple text (use last position, color and size) |
#| txp | txp,[X],[Y] | Set text position (move the cursor) |
#| txc | txc,[foreColor],[backgroundColor] | Set text color (background is transparent if not provided |
#| txs | txs,[SIZE] | Set text size |
#| txtfull | txtfull,[row],[col],[size=1],[foreColor=white],[backColor=black],[text] | Write text with all options |
#| l | l,[x1],[y1],[2],[y2],[color] | Draw a simple line |
#| r | r,[x],[y],[width],[height],[color] | Draw a rectangle |
#| rf | rf,[x],[y],[width],[height],[bordercolor],[innercolor] | Draw a filled rectangle |
#| c | c,[x],[y],[radius],[color] | Draw a circle |
#| cf | cf,[x],[y],[radius],[bordercolor],[innercolor] | Draw a filled circle |
#| t | t,[x1],[y1],[x2],[y2],[x3],[y3],[color]| Draw a triangle |
#| tf | tf,[x1],[y1],[x2],[y2],[x3],[y3],[bordercolor],[innercolor] | Draw a filled triangle |
#| px | px,[x],[y],[color] | Print a single pixel |
#| img | img,[X],[Y],[filename] | Draw image file at position |
#
#TFTCMD Subcommands:
#| TFT Subcommands | details | description |
#|-----|-----|-----|
#| on | on | Display ON |
#| off | off | Display OFF |
#| clear | clear,[color] | Clear display |
#| rot | rot,[value] | Rotate display (value from 0 to 3 inclusive) |

An image can be written onto screen with "tft,img,0,0,imgfilenamewithpath" command.

Other plugin parameters can be added such as background image if needed.
Perhaps one icon per line also possible. But these settings will make scrolling a lot slower.

Budman1758, 2021-02-18:
Got this set up for testing. Basic functions seem to be OK. 

I'm a bit fuzzy on the commands for text and background colors. Can't seem to get that working. Can you provide an example command to write text with a color with a colored background? Do the color commands "persist"?  In other words do you "set" a text color and then write the text with a separate command? 

The write image command seems to work well.

Is it possible to add a text and background color setting in the plug-in settings page? That would make things a lot easier for standard output use cases. A background image setting would be pretty awesome too. I'm thinking along the lines of a color gradient or similar look for my use case. Maybe a starfield image or something. I don't know how difficult that is. Just me dreaming...:>)

enesbcs, 2021-02-18:
> I'm a bit fuzzy on the commands for text and background colors. Can't seem to get that working. Can you provide an example command to write text with a color with a colored background? Do the color [...]

Honestly text background is not currently implemented, as text background in PIL is non-existent. It can be added by clearing its area with a background color filled rectangle before write.
txc command color persists for txt commands.
 
> Is it possible to add a text and background color setting in the plug-in settings page? That would make things a lot easier for standard output use cases. A background image setting would be pretty [...]

Not difficult bit might blink a little. Will see.

Summary, the next features will be on focus:
- text background handling
- plugin setting for background and foreground color
- plugin setting for background image

enesbcs, 2021-02-19:
background support added at commit https://github.com/enesbcs/rpieasy/commit/ab730f411a0b5a895957c33fd65a194c3b95f956

Text background is always transparent, when a background image is selected.

Budman1758, 2021-02-19:
Working pretty good!!  Nice work!!

enesbcs, 2021-02-20:
Thanks! The display can only show 16bit colors but actually RGB JPEG images looks decently. Background image is auto-resized to display dimensions.

In theory this plugin will also work with ST7735, i've got one somewhere, will test.

enesbcs, 2021-03-08:
Closed as implemented.
Feel free to open a new issue in case of error.

closed 2021-03-08 by enesbcs