enesbcs/rpieasy #191: New controller: SX1262 SPI LoRa

enesbcs, 2020-12-30:


enesbcs, 2021-01-09:
I am testing now the SX1262 with this library, https://github.com/ehong-tl/micropySX126X
but i've got only ERR_WRONG_MODEM answers from that. It may take a while to figure out how to use it in the first place.

enesbcs, 2021-01-11:
I think i've finally identified the source of my problem. According to Semtech SX1261/2 datasheet, Status byte returned at the beginning of the buffer. (2x in normal reads, 3x at buffer read and 4 times with register read functions!)
So i've modified the code and shifted result buffers for Raspberry Pi and finally the Sx1262 receiver on the RPI received this:

READ BUFFER: [255, 1, 1, 0, 36, 111, 40, 37, 27, 108, 21, 39, 33, 3, 7, 109, 112, 121, 69, 97, 115, 121, 0]
from an ESP32+Sx1262 sender running mPyEasy. :)
Which is in fact a standard alive P2P message! :)

However it is just the beginning of my journey, now i am able to implement sx1262 driver into LoraP2P controller.

Budman1758, 2021-01-12:
Can you tell me what LoRa module you are using?  Might like to try this out.

enesbcs, 2021-01-12:
I'd bought an EBYTE E22-900M30S SPI connected chip:  (serially connected is not supported)
https://www.aliexpress.com/item/32919771141.html?spm=a2g0o.cart.0.0.17943c00atIOJb&mp=1

After i succesfully implement the code into C020 controller, it will drive any SPI connected SX1262 chip.

(At this time the C20 controller supports the more common SX1276/HPD13A/RFM95W chips )

On the other side (mPyEasy) i am using simply a LILYGO TTGO ESP32 SX1276 LoRa 868 / 915MHz Lora Development Board without display:
https://www.aliexpress.com/item/4001226226435.html?spm=a2g0o.productlist.0.0.2cfe1f72suQcdj&algo_pvid=109042f6-ac2b-4fbe-b8b4-c5bcd3e1227a&algo_expid=109042f6-ac2b-4fbe-b8b4-c5bcd3e1227a-11&btsid=0b0a555d16104718474431694e8bee&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

enesbcs, 2021-01-19:
For the records the test installation settings on RPIEasy side:
[image]

and on the mPyEasy side:
[image]

Its now working, RPIEasy codebase will be updated soon. However it is a little experimental, as the micropython library had to be altered quite a bit. But sending/receiving works according my tests.

enesbcs, 2021-01-19:
Support added at commit https://github.com/enesbcs/rpieasy/commit/55cbfeb27967f335398e82e0f730a5d29d2ebbdb

enesbcs, 2021-01-27:
Closed as implemented

closed 2021-01-27 by enesbcs